zephyrproject-rtos/zephyr #60155: mimxrt1024_evk: configure GPIO_SD_B1_05 cause system failure

jeppenodgaard, 2023-07-07:
**Describe the bug**
If GPIO_SD_B1_05 is configured using `gpio_pin_configure_dt` the system hangs.
This has not always been an issue.

The issue is present on:
 - main (5f0bba7ced041c791561fc49c4af12566f8a8f83)
 - v3.4-branch (356c8cbe63ae01b3ab438382639d25bb418a0213)

The issue is **not** present on:
 - v3.1-branch (0c70f3bad725c2389264eaea0bc4591fcebea1bc)

I have not tested v3.2 and v3.3.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit `mimxrt1024_evk.dts` node `gpio_keys`: `gpio5 0` -> `gpio3 25`.
2. Build button sample: `west build -p auto -b mimxrt1024_evk zephyr/samples/basic/button/`
3. Flash: `west flash`
4. Attach: `west attach`
5. Print backtrace: `bt`

**Expected behavior**
Button sample work and prints:
```
Set up button at GPIO_3 pin 25
Set up LED at GPIO_1 pin 24
Press the button
```

**Impact**
GPIO_SD_B1_05 is unusable as GPIO.

**Logs and console output**
`west attach` then `bt` gives the following output.
```
Program received signal SIGTRAP, Trace/breakpoint trap.
0xe8401f80 in ?? ()
(gdb) bt
#0  0xe8401f80 in ?? ()
#1  <signal handler called>
#2  0xe8401f80 in ?? ()
#3  <signal handler called>
#4  z_impl_gpio_pin_interrupt_configure (flags=96468992, pin=25 '\031', port=<optimized out>)
    at /zephyrproject/zephyr/include/zephyr/drivers/gpio.h:658
#5  gpio_pin_interrupt_configure (flags=96468992, pin=25 '\031', port=<optimized out>)
    at /zephyrproject/build/zephyr/include/generated/syscalls/gpio.h:35
#6  gpio_pin_interrupt_configure_dt (spec=0x60005948 <button>, flags=96468992)
    at /zephyrproject/zephyr/include/zephyr/drivers/gpio.h:691
#7  main () at /zephyrproject/zephyr/samples/basic/button/src/main.c:58
(gdb) 
```
**Additional context**
The MIMXRT1024-EVK schematic has a note about `GPIO_SD_B1_05` which says:

> FlexSPI_DQS PIN need floating for QSPI Flash RW @133MHz

I am not sure this means exactly or could be related to the issue.

DerekSnell, 2023-07-19:
Hi @jeppenodgaard ,
Sorry for the delay, I just found this issue.

The note you found in the schematic does explain this issue, and as the FlexSPI is configured, the `GPIO_SD_B1_05` signal is reserved for the QSPI interface for the integrated flash.  To maximize the QSPI clock frequency, the FlexSPI bitfield `MCR0[RXCLKSRC]` is set to 0b01.  The RX clock is used to latch the QSPI data in during the read, and this bitfield controls the source and timing of that signal.  The option 0b01 routes an internal read strobe signal through the FlexSPI DQS pad of the pin, which improves the timing by accounting for the pin pad delay.  But when configured like this, this pin must be reserved for the FlexSPI DQS signal.

One option is to configure the FlexSPI `RXCLKSRC` to 0, which does not route the read strobe signal through the DQS pad.  Then you can use that pin for other functions.  However, from the RT1024 datasheet, this setting will reduce the limit of the QSPI clock frequency from 133MHz to 60MHz (in SDR mode).  That could have a significant impact on the flash performance.  So it may be better to find another pin for your needs.

Best regards

jeppenodgaard, 2023-07-28:
Hi @DerekSnell.

Thanks for the thorough explanation!

Do you know if Zephyr 3.1 run at a slower flash speed?

DerekSnell, 2023-08-01:
Hi @jeppenodgaard ,
Yes, it appears the startup code for the RT1024 in Zephyr v3.1 may have ran at a slower QSPI clock frequency, and did not use the `FLEXSPI_A_DQS` pad for the FlexSPI timing.

First, some background on how the RT1024 device boots.  After reset, the ROM bootloader executes.  When configured to boot from FlexSPI, the boot ROM configures the FlexSPI interface and pins as a 1-bit SPI interface.  It then reads the Flash Configuration Block (FCB) from the QSPI.  The FCB tells the ROM how to configure the FlexSPI for XIP.  The boot ROM then reconfigures the FlexSPI and pins for QSPI mode and other settings in the FCB, and then starts executing the application from QSPI.

The RT1024 Reference Manual has this note about the `FLEXSPI_A_DQS` pin:
> The FLEXSPI_A_DQS pad is not supported by ROM natively in RT1024, which restricts the maximum FLASH frequency to 60MHz. If the user application requires a high-performance eXecute-in-Place(for example, executing code on the FLASH at 133MHz), the FLEXSPI_A_DQS function can be enabled by the following steps:
• configure the FLEXSPI_A_DQS pad in the startup code;
• set FLEXSPI_MCR0[RXCLKSRC] bitfield to 1;
• switch the FLEXSPI clock root to a high-frequency clock.

Now let's look how this is done in the latest Zephyr startup code for this SOC (same as Zephyr v3.4).  Zephyr calls [z_arm_platform_init()](https://github.com/zephyrproject-rtos/zephyr/blob/518bbc130327bfa0463a2b075d7f88b1b233990a/soc/arm/nxp_imx/rt/soc_rt10xx.c#L333C6-L333C26) very early in the startup code, to initialize the hardware.  A recent change in Zephyr added `z_arm_platform_init()` calling [SystemInit()](https://github.com/zephyrproject-rtos/zephyr/blob/518bbc130327bfa0463a2b075d7f88b1b233990a/soc/arm/nxp_imx/rt/soc_rt10xx.c#L344C2-L344C14).

`SystemInit()` is a common API in the NXP MCUXpresso SDK that Zephyr uses as the HAL.  It provides the same purpose as Zephyr's `z_arm_platform_init()` to do the initial hardware initialization needed to boot the app.  The Zephyr port for these boards now call `SystemInit()` to re-use this startup code already in the SDK.  For the RT1024 when XIPing from QSPI, `SystemInit()` [reconfigures the FlexSPI](https://github.com/zephyrproject-rtos/hal_nxp/blob/32de41804dd8c5242be58604682583f2e8889da5/mcux/mcux-sdk/devices/MIMXRT1024/system_MIMXRT1024.c#L88) as described in that Reference Manual note.  That code configures pin `GPIO_SD_B1_05` as `FLEXSPI_A_DQS`.  It configures `RXCLKSRC` for mode 0b01 for `kFLEXSPIReadSampleClk_LoopbackFromDqsPad`.  And it configures the FlexSPI clock for the QSPI SDR maximum 133MHz.

Reviewing the code from Zephyr v3.1, [z_arm_platform_init()](https://github.com/zephyrproject-rtos/zephyr/blob/2ddd73feafd3316af2c547c34d6969bea637d5c6/soc/arm/nxp_imx/rt/soc_rt10xx.c#L306C6-L306C26) did not call `SystemInit()`, so the FlexSPI and pins were left in the state after the boot ROM configured them.

I hope that helps

jeppenodgaard, 2023-08-01:
Thanks @DerekSnell !

That makes a lot of sense.
I have no further questions 😊

I will close the issue since it is not a bug. I might open a PR pointing this out somehow as it was not something I was able to find without help.