zephyrproject-rtos/zephyr #60167: `tests/application_development/code_relocation` linker warning for nucleo_h7a3zi_q

manuargue, 2023-07-07:
**Describe the bug**
`tests/application_development/code_relocation/tests.application_development.code_relocation` emits a linker warning for `nucleo_h7a3zi_q`, causing a CI failure when this test is built.

**To Reproduce**
`west build -p always -b nucleo_h7a3zi_q tests/application_development/code_relocation -T tests.application_development.code_relocation`

**Expected behavior**
No linker warning is emitted so that CI doesn't detect this as a failure.

**Impact**
Blocks https://github.com/zephyrproject-rtos/zephyr/pull/60166, and may cause unrelated CI build failures in other PR's.

**Logs and console output**
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /__w/zephyr/zephyr/twister-out/nucleo_h7a3zi_q/tests/application_development/code_relocation/tests.application_development.code_relocation/zephyr/zephyr_pre0.map 
: && ccache /opt/toolchains/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/__w/zephyr/zephyr/twister-out/nucleo_h7a3zi_q/tests/application_development/code_relocation/tests.application_development.code_relocation/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/toolchains/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -L/__w/zephyr/zephyr/twister-out/nucleo_h7a3zi_q/tests/application_development/code_relocation/tests.application_development.code_relocation/zephyr  -lgcc  zephyr/libcode_relocation_source_lib.a  test_lib/libtest_lib.a  zephyr/arch/common/libisr_tables.a  -Wl,--fatal-warnings  -mcpu=cortex-m7  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cd /__w/zephyr/zephyr/twister-out/nucleo_h7a3zi_q/tests/application_development/code_relocation/tests.application_development.code_relocation/zephyr && /usr/bin/cmake -E true
/opt/toolchains/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd:zephyr/linker_zephyr_pre0.cmd:11: warning: redeclaration of memory region `SRAM2'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Linux, Windows
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA 5f0bba7ced041c791561fc49c4af12566f8a8f83

**Additional context**
Caught in https://github.com/zephyrproject-rtos/zephyr/pull/60166/checks?check_run_id=14868457593 when fixing the same test but for another board. The issue is unrelated and likely to happen with any board that defines a `SRAM2` memory region.

manuargue, 2023-07-07:
Hi @erwango @ABOSTM @FRASTM, tagging you as codeowners of /boards/arm/nucleo*/

erwango, 2023-07-10:
See my comment in https://github.com/zephyrproject-rtos/zephyr/pull/60166#pullrequestreview-1521462343

closed 2023-07-14 by carlescufi