zephyrproject-rtos/zephyr #60193: SAME51 & SAME54 CACHE_MANAGEMENT Error when CAN-Bus is used

DomiHive, 2023-07-10:
I am currently developing a board with the ATSAME51J19A as soc and I am facing a problem with the activation of the can bus. I have added can0 and can1 to the soc in the .dts file. The same error occures when compiled for the atsame54_xpro with the babbling sample. Now I get the following warning/error: 
```
warning: CACHE_MANAGEMENT (defined at arch/Kconfig:860) has direct dependencies DCACHE || ICACHE with value n, but is currently being y-selected by the following symbols:
 - CAN_SAM (defined at drivers/can/Kconfig.sam:5), with value y, direct dependencies DT_HAS_ATMEL_SAM_CAN_ENABLED && CAN (value: y), and select condition DT_HAS_ATMEL_SAM_CAN_ENABLED && CAN (value: y)
Parsing /home/hive/hive_zephyr/zephyr/samples/drivers/can/babbling/Kconfig
Loaded configuration '/home/hive/hive_zephyr/zephyr/boards/arm/atsame54_xpro/atsame54_xpro_defconfig'
Merged configuration '/home/hive/hive_zephyr/zephyr/samples/drivers/can/babbling/prj.conf'

error: Aborting due to Kconfig warnings
```

**Here are some additional information to this case**
 - I am using a self developed board with the same51j19a as SOC
 - This error also happens when I add CAN-Bus to the atsame54_xpro and try to compile the CAN-Bus babbling sample
 - I added the CAN as follows in the .dts file:
```
   	soc {
		can0: can@42000000 {
			status = "okay";
			compatible = "atmel,sam-can";
			reg = <0x42000000 0x100>;
			reg-names = "m_can";
			interrupts = <36 0>;
			interrupt-names = "LINE_0";
			clocks = <&gclk 7>, <&mclk 0x10 8>;
			clock-names = "GCLK", "MCLK";
			divider = <12>;
			bosch,mram-cfg = <0x0 28 8 3 3 0 1 1>;
			sjw = <1>;
			sample-point = <875>;
			sjw-data = <1>;
			sample-point-data = <875>;
		};

		can1: can@42000400 {
			status = "okay";
			compatible = "atmel,sam-can";
			reg = <0x42000400 0x100>;
			reg-names = "m_can";
			interrupts = <38 0>;
			interrupt-names = "LINE_0";
			clocks = <&gclk 7>, <&mclk 0x10 9>;
			clock-names = "GCLK", "MCLK";
			divider = <12>;
			bosch,mram-cfg = <0x0 28 8 3 3 0 1 1>;
			sjw = <1>;
			sample-point = <875>;
			sjw-data = <1>;
			sample-point-data = <875>;
		};
	};

&can0 {
	status = "okay";

	pinctrl-0 = <&can0_default>;
	pinctrl-names = "default";

	bus-speed = <125000>;
	bus-speed-data = <1000000>;

	can-transceiver {
		max-bitrate = <5000000>;
	};
};

&can1 {
	status = "okay";

	pinctrl-0 = <&can1_default>;
	pinctrl-names = "default";

	bus-speed = <125000>;
	bus-speed-data = <1000000>;

	can-transceiver {
		max-bitrate = <5000000>;
	};
};
```

 - I also added CAN-Bus as follows in the .dtsi-file:
```
	can0_default: can0_default {
		group1 {
			pinmux = <PA23I_CAN0_RX>,
				 <PA22I_CAN0_TX>;
		};
	};

	can1_default: can1_default {
		group1 {
			pinmux = <PB15H_CAN1_RX>,
				 <PB14H_CAN1_TX>;
		};
	};
```

**Steps to reproduce the behavior**
1. add CAN-Bus to the SOC as described above (tested with the SAME51J19 and the SAME54 of the atsame54_xpro board)
2. west build -b atsame54_xpro zephyr/samples/drivers/can/babbling
3. See error

**Expected behavior**
I expected the project to build with maybe a CAN-Bus error or something, but not a CACHE_MANAGEMENT Error

**Impact**
I am developing a new board. If this dosen't work it is a complete showstopper. I can't continue to implement other functionalities if the CAN-Bus is not working.

**Logs and console output**
```
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/hive/hive_zephyr/zephyr/samples/drivers/can/babbling
-- CMake version: 3.26.4
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter 
-- Cache files will be written to: /home/hive/.cache/zephyr
-- Zephyr version: 3.4.99 (/home/hive/hive_zephyr/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: atsame54_xpro
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/home/hive/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (/home/hive/zephyr-sdk-0.16.1)
-- Found Dtc: /home/hive/zephyr-sdk-0.16.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/hive/hive_zephyr/zephyr/boards/arm/atsame54_xpro/atsame54_xpro.dts
-- Generated zephyr.dts: /home/hive/hive_zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/hive/hive_zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/hive/hive_zephyr/build/zephyr/dts.cmake

warning: CACHE_MANAGEMENT (defined at arch/Kconfig:860) has direct dependencies DCACHE || ICACHE with value n, but is currently being y-selected by the following symbols:
 - CAN_SAM (defined at drivers/can/Kconfig.sam:5), with value y, direct dependencies DT_HAS_ATMEL_SAM_CAN_ENABLED && CAN (value: y), and select condition DT_HAS_ATMEL_SAM_CAN_ENABLED && CAN (value: y)
Parsing /home/hive/hive_zephyr/zephyr/samples/drivers/can/babbling/Kconfig
Loaded configuration '/home/hive/hive_zephyr/zephyr/boards/arm/atsame54_xpro/atsame54_xpro_defconfig'
Merged configuration '/home/hive/hive_zephyr/zephyr/samples/drivers/can/babbling/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /home/hive/hive_zephyr/zephyr/cmake/modules/kconfig.cmake:343 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/hive/hive_zephyr/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
  /home/hive/hive_zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/hive/hive_zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/hive/hive_zephyr/build -GNinja -DBOARD=atsame54_xpro -S/home/hive/hive_zephyr/zephyr/samples/drivers/can/babbling
```

**Environment (please complete the following information):**
 - OS: Linus
 - Toolchain: Zephyr SDK 0.16.1
 - Zephyr 3.4.0

str4t0m, 2023-07-10:
I have removed the bug lable as CAN is not yet supported on this SoC.

str4t0m, 2023-07-10:
I am not familiar with these SAME5x Socs and  neither about the difference to the currently only SoC (SAME70) that has been tested with the  `atmel,sam-can` driver. But a few notes:
- Having a quick look into the zephyr repo it looks like the same5x Socs are part of the sam0 product line, so most likely you'll have better chances with the `atmel,sam0-can`  driver.
- Also, you need to adopt the dts parameters to match your Soc, the interrupt line for example definitely does not match the Soc and seems to be just copied from an in-tree Soc.
- Furthermore, enabling a driver for a different series often requires the one or the other small tweak to the driver as well. So even if your device tree parameters are defined correctly not everything might work right away.

henrikbrixandersen, 2023-07-28:
Related PR: https://github.com/zephyrproject-rtos/zephyr/pull/60425