zephyrproject-rtos/zephyr #60196: FRDM_K64F: tests/drivers/dma/scatter_gather test fails.

gbarkadiusz, 2023-07-10:
**Describe the bug**
 `tests/drivers/dma/scatter_gather/` fails.

Platform: FRDM_K64f
After bisect sesion.
```
8eeb3554b0afa055e196484d2b8498f1faeecf65 is the first bad commit
commit 8eeb3554b0afa055e196484d2b8498f1faeecf65
Author: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
Date:   Thu May 11 06:54:47 2023 -0500

    tests: dma: Increase data transferred in scatter_gather test
    
    Increase the amount of data to be transferred and the
    number of blocks to transfer.
    
    Signed-off-by: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>

 tests/drivers/dma/scatter_gather/src/test_dma_sg.c | 25 +++++++++++-----------
 1 file changed, 13 insertions(+), 12 deletions(-)
```

**To Reproduce**
Steps to reproduce the behavior:
 `./scripts/twister --device-testing --hardware-map ./map.yml -T ./tests/drivers/dma/scatter_gather/ -v`
Output:
```
INFO    - 1/1 frdm_k64f                 tests/drivers/dma/scatter_gather/drivers.dma.scatter_gather  FAILED Failed (device 2.475s)
ERROR   - see: /home/zephyr/zephyrproject/zephyr/twister-out/frdm_k64f/tests/drivers/dma/scatter_gather/drivers.dma.scatter_gather/handler.log
```
Handler.log contains
```
*** Booting Zephyr OS build zephyr-v3.3.0-4545-g8eeb3554b0af ***
Running TESTSUITE dma_m2m_sg
===================================================================
START - test_dma_m2m_sg
DMA memory to memory transfer started
Preparing DMA Controller
dma block 0 block_size 8192, source addr 20008060, dest addr 20000060
set next block pointer to 0x2000a6a4
dma block 1 block_size 8192, source addr 20008060, dest addr 20002060
set next block pointer to 0x2000a6c4
dma block 2 block_size 8192, source addr 20008060, dest addr 20004060
set next block pointer to 0x2000a6e4
dma block 3 block_size 8192, source addr 20008060, dest addr 20006060
Configuring the scatter-gather transfer on channel 0
E: please config DMA_TCD_QUEUE_SIZE as 4
ERROR: transfer config (0)
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/dma/scatter_gather/src/test_dma_sg.c:160: dma_m2m_sg_test_dma_m2m_sg: (test_sg() == TC_PASS) is false
FAIL - test_dma_m2m_sg in 0.068 seconds
===================================================================
TESTSUITE dma_m2m_sg failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [dma_m2m_sg]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.068 seconds
- FAIL - [dma_m2m_sg.test_dma_m2m_sg] duration = 0.068 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 04db2067ac5d10604b1c561c779bfb17
PROJECT EXECUTION FAILED
```

**Environment :**
 - OS: Linux
 - Toolchain Zephyr_SDK 0.16.1

hakehuang, 2023-07-12:
mimxrt10xx series has the same issue

mmahadevan108, 2023-07-24:
@gbarkadiusz , a fix has been pushed. Can you please re-test.

gbarkadiusz, 2023-07-25:
It has been fixed. So I'm going to close this issue.

closed 2023-07-25 by gbarkadiusz