zephyrproject-rtos/zephyr #60212: ATT channel not connected

fredrikdanebjer, 2023-07-10:
Description
Sometimes (I guesstimate 20% of the time) when reconnecting to a bonded device over Bluetooth, the connection fails with error output:
<err> bt_att: ATT channel not connected

Equipment
Peripheral Platform: nRF5340 running zephyr/tests/bluetooth/shell/
Central Platform: Android Pixel7/7 Pro/7a, running Android Beta 14.3

Reproduce:
1. Compile zephyr/tests/bluetooth/shell/ and flash

2. Run the following on the shell app
bt init
bt clear all
bt bondable on
bap init
bt advertise on

3. On android, pair and connect with the peripheral

4. Make a power cycle on the shell and run the following commands
bt init
bap init

5. Wait for the Android to reconnect and output:
<err> bt_att: ATT channel not connected
If it doesn't, then repeat all steps until it does.

Expected behavior
The expected buggy behavior is that the reconnect fails ~20% of the time, with error output:
<err> bt_att: ATT channel not connected

The desired behavior is that it never fails.

Impact
Critical

**Logs and console output**
Failure log:
```
*** Booting Zephyr OS build zephyr-v3.3.0-5820-g14ef3907bc77 ***
[00:00:00.252,197] <dbg> bt_ots: bt_gatt_ots_l2cap_init: Initialized OTS L2CAP
Type "help" for supported commands.Before any Bluetooth commands you must `bt init` to initialize the stack.
uart:~$ bt init
Bluetooth initialized
Settings Loaded
[00:00:02.931,243] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:02.931,274] <inf> fs_nvs: alloc wra: 0, e28
[00:00:02.931,304] <inf> fs_nvs: data wra: 0, 3e8
[00:00:02.945,343] <dbg> bt_att: bt_eatt_init: 
uart:~$ bap init
[00:00:02.945,648] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:02.952,026] <inf> bt_hci_core: Identity: E9:41:68:C0:42:0D (random)
[00:00:02.952,056] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:02.952,087] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
[00:00:04.728,057] <dbg> bt_pacs: add_bonded_addr_to_client_list: Added 74:74:46:AE:25:66 (public) to bonded list

[00:00:04.728,088] <dbg> bt_pacs: bt_pacs_cap_register: cap 0x20001958 dir sink codec_cap id 0x06 codec_cap cid 0x0000 codec_cap vid 0x0000
[00:00:04.728,240] <dbg> bt_pacs: add_bonded_addr_to_client_list: Added 74:74:46:AE:25:66 (public) to bonded list

[00:00:04.728,271] <dbg> bt_pacs: bt_pacs_cap_register: cap 0x20001950 dir source codec_cap id 0x06 codec_cap cid 0x0000 codec_cap vid 0x0000
uart:~$ bt advertise on
Advertising started
Connected: 74:74:46:AE:25:66 (public)
[00:00:10.681,152] <dbg> bt_att: bt_att_accept: conn 0x2000a050 handle 0
[00:00:10.681,213] <dbg> bt_att: att_chan_attach: att 0x2002dbe8 chan 0x2002d988 flags 0
[00:00:10.681,213] <dbg> bt_att: bt_att_connected: chan 0x2002d990 cid 0x0004
[00:00:10.683,563] <dbg> bt_att: bt_att_req_alloc: req 0x2003336c
[00:00:10.683,593] <dbg> bt_att: bt_att_req_send: conn 0x2000a050 req 0x2003336c
[00:00:10.683,624] <dbg> bt_att: bt_att_chan_req_send: req 0x2003336c
[... 203 lines cut ...]
[00:00:13.191,375] <dbg> bt_att: att_write_req: handle 0x0020
[00:00:13.191,406] <dbg> bt_att: write_cb: handle 0x0020 offset 0
[00:00:13.191,436] <dbg> bt_att: chan_send: code 0x13
[00:00:13.221,435] <dbg> bt_att: bt_att_sent: chan 0x2002d508
[00:00:13.221,435] <dbg> bt_att: chan_rsp_sent: chan 0x2002d508
[00:00:13.251,312] <dbg> bt_att: bt_att_recv: Received ATT chan 0x2002d508 code 0x12 len 4
[00:00:13.251,342] <dbg> bt_att: att_write_req: handle 0x0023
[00:00:13.251,403] <dbg> bt_att: write_cb: handle 0x0023 offset 0
[00:00:13.251,434] <dbg> bt_att: chan_send: code 0x13
[00:00:13.281,433] <dbg> bt_att: bt_att_sent: chan 0x2002d508
[00:00:13.281,463] <dbg> bt_att: chan_rsp_sent: chan 0x2002d508
[00:00:13.311,309] <dbg> bt_att: bt_att_recv: Received ATT chan 0x2002d508 code 0x12 len 4
[00:00:13.311,340] <dbg> bt_att: att_write_req: handle 0x0017
[00:00:13.311,370] <dbg> bt_att: write_cb: handle 0x0017 offset 0
[00:00:13.311,401] <dbg> bt_att: chan_send: code 0x13
[00:00:13.341,461] <dbg> bt_att: bt_att_sent: chan 0x2002d508
[00:00:13.341,461] <dbg> bt_att: chan_rsp_sent: chan 0x2002d508
LE conn param updated: int 0x0018 lat 0 to 42
uart:~$
```

Environment
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.0 (also seen in 0.15.0)
 - Commit SHA: f3cafe75593596e09893fdb9377b57538d4031ea

carlescufi, 2023-07-10:
@fredrikdanebjer can you capture a sniffer trace with a spare Nordic DK?

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_sniffer_ble%2FUG%2Fsniffer_ble%2Fintro.html&cp=11_5

jori-nordic, 2023-07-11:
@fredrikdanebjer yes ellisys is even better

fredrikdanebjer, 2023-07-11:
@jori-nordic @carlescufi please find Ellisys Trace attached to the issue

jori-nordic, 2023-07-17:
@fredrikdanebjer think I found it:
could you try with 
[image]

What I believe is happening is `result` is "Some connections refused – not enough resources available" (as can be seen from the trace. Then _none_ of the callbacks end up being called, and thus none of the ATT channels exist.

When it works, it works because the channels are initiated from the zephyr device, this is a different code path. 

You can verify that, on looking at the logs: the working case has no `bt_eatt_accept`, only `att_chan_attach`. 
And similarly, you can see that the working setup has multiple `bt_att_connected` whilst the other one only has the fixed (U)ATT one on connection establishment.

PR #60866 (2023-07-31): Bluetooth: l2cap: change `connected` cb call condition for ECRED
This was recently refactored (in #58440). But it introduced a bug in which
some channels were connected but not the whole list asked for by the peer.

In that case, `result` will not be `SUCCESS` but we still want to call the
`connected` callback as the peer will consider those channels to be
connected when we send the response.

The symptom is that EATT channels are being instantiated, but not
considered connected (ie. usable by the stack).

Also introduce a test that has asymmetric channel resources (5 on central
vs 2 on periph [... cut]