zephyrproject-rtos/zephyr #60213: Anytree update breaks ram_report

kica-z, 2023-07-10:
**Describe the bug**
Zephyr does not specify which version of anytree is needed within it's `requirements-extras.txt` file this can lead to unexpected behaviour when an update of the package occurs. More precisely the problem seems to be that anytree released a new version `2.9.0` on 25. June which seems to break the ram_report target. Error message:
```
Traceback (most recent call last):
  File "/opt/zephyrproject/zephyr/scripts/footprint/size_report", line 803, in <module>
    main()
  File "/opt/zephyrproject/zephyr/scripts/footprint/size_report", line 790, in main
    root = generate_any_tree(symbol_dict, symsize, common_path_prefix)
  File "/opt/zephyrproject/zephyr/scripts/footprint/size_report", line 560, in generate_any_tree
    root = TreeNode('Root', "root")
  File "/opt/zephyrproject/zephyr/scripts/footprint/size_report", line 534, in __init__
    self.size = size
AttributeError: can't set attribute 'size'
```

The problem is mitigated when specifying version `2.8.0` in the  `requirements-extras.txt` manually. 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b qemu_x86
2. ninja -C build ram_report
5. See error

**Expected behavior**
Ram report should be printed to the shell

**Impact**
No big impact since a workaround was found by manually defining the anytree version.

JPHutchins, 2023-07-12:
@kica-z Verifying this bug and the fix.  To whom it concerns, any interesting to pyproject.toml modules in order to comply with PIP 518 and leave these kinds of issues in the dust?  🚀

VincentFontanellaLinkio, 2023-07-24:
@kica-z 
I have the same problem and we have found a workaround : 
We had a previous version for anytree into the file zephyr/scripts/requirements-extras.txt

`# used by twister for --test-tree option
anytree==2.6.0`

PR #60774 (2023-08-16): scripts: size_report: Fix ram_report and rom_report with anytree 2.9.0
The anytree package has introduced a breaking change in version 2.9.0 by adding a 'size' property to the NodeMixin class. Since the TreeNode class in size_report derives from NodeMixin and defines an attribute with the same name, an AttributeError is raised when generating reports. With this change, the attributes of the TreeNode class are prefixed with an underscore to resolve the name collision and to prevent future name collisions.

Fixes #60213.
commit 6a20f4b0 (2023-08-16): scripts: size_report: Fix ram_report and rom_report with anytree 2.9.0

The anytree package has introduced a breaking change in version 2.9.0 by
adding a 'size' property to the NodeMixin class. Since the TreeNode class
in size_report derives from NodeMixin and defines an attribute with the
same name, an AttributeError is raised when generating reports.
With this change, the attributes of the TreeNode class are prefixed with
an underscore to resolve the name collision and to prevent future name
collisions.

Fixes #60213.
closed 2023-08-16 by nashif