zephyrproject-rtos/zephyr #60217: S32Z270DC2 defined SRAM region too small for almost all workloads

benmordaunt, 2023-07-10:
The S32Z270DC2 DeviceTree includes its chosen SRAM definition node from `dts/arm/nxp/nxp_s32z27x_r52.dtsi`, which presents as:
```
sram0: memory@31780000 {
			compatible = "mmio-sram";
			reg = <0x31780000 DT_SIZE_M(1)>;
		};
```
... offering 1MiB of application storage for an image programmed directly into RAM.
At least on the S32Z270DC2 evaluation board, it is known that a significantly larger region exists at `0x32100000` with size `DT_SIZE_M(7)`, though without direct access to a datasheet myself, it is difficult to elaborate on this.

Can someone with more insight look into updating this board's SRAM definitions?

Thanks.

benmordaunt, 2023-07-10:
On further investigation, it seems that these regions (as specified in the current Zephyr tree), are actually designated as "fast data" regions and not "code" regions, for which there are actually 2 7MiB regions for each RTU.

manuargue, 2023-07-14:
Hi @benmordaunt, thanks for logging this issue. I've removed the bug label and tag this as a feature request to support CRAM beside DRAM, currently supported. There's no restriction on how these RAM banks can be used, but a recommended use. I agree it would be better to separate data and code to use the optimized RAM banks for each.
Our team will look into this and feedback you.

manuargue, 2025-02-09:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/81112

closed 2025-02-09 by manuargue