zephyrproject-rtos/zephyr #60235: Zerphyr module uoscore-uedhoc sample `samples/zephyr_oscore/server` refers to tinycbor which no longer exists

nordicjm, 2023-07-11:
**Describe the bug**
This file: https://github.com/zephyrproject-rtos/uoscore-uedhoc/blob/zephyr/samples/zephyr_oscore/server/CMakeLists.txt refers to tinycbor in zephyr, a module which no longer exists. It also uses the old `$ENV{ZEPHYR_BASE}` variable which should be replaced with `${ZEPHYR_BASE}`

**Expected behavior**
Issues to be addressed

**Impact**
Probably NIL since no one has reported this issue before so assuming no-one uses this codebase and/or sample

nordicjm, 2023-07-11:
@StefanHri since you seem to maintain this module

carlescufi, 2023-07-12:
@nordicjm I just discussed this with @rlubos and apparently the samples included  in the module are not used at all by Zephyr, only the library itself is. So perhaps this issue should be moved to the upstream project?

rlubos, 2023-07-25:
I've opened #60789 which updates the module to the latest revision. Tinycbor is no longer in use.

closed 2023-08-02 by carlescufi