zephyrproject-rtos/zephyr #60239: Bluetooth: Host: `bond_deleted` not called if bond is overwritten when `CONFIG_BT_KEYS_OVERWRITE_OLDEST=y`

Thalley, 2023-07-11:
**Is your enhancement proposal related to a problem? Please describe.**
When `CONFIG_BT_KEYS_OVERWRITE_OLDEST=y` then bonding information may be overwritten based on "age". In that case, the old bonds are implicitly deleted, and a new one is stored, but the `bt_conn_auth_info_cb.bond_deleted` does not seem to be called in this case. 

This means that an application effectively have implement an identical handling of overwriting oldest bonds, as it cannot rely on `bond_deleted`. 

**Describe the solution you'd like**
The `bond_deleted` should be called if a bond is implicitly overwritten, or some other mechanism should provide that information to the application. 

**Describe alternatives you've considered**
N/A

**Additional context**
See e.g. https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/csip_set_member.c#L568C39-L586 where it has been necessary to implement the age counter again

alwa-nordic, 2023-07-11:
I think this should be marked as a bug.

Thalley, 2023-07-11:
> I think this should be marked as a bug.

Was unsure if it was intentional or not, thus why I didn't create it as a bug. I'd be OK to mark it as such though

alwa-nordic, 2023-08-03:
I am not able to reproduce. `bt_conn_auth_info_cb::bond_deleted` gets invoked as expected. See my bsim test at https://github.com/alwa-nordic/zephyr/commit/dc8e371a29712a7dd0b2a29b9cd4c0f327100ba7

Thalley, 2023-08-14:
@alwa-nordic thanks - I went through the code again and saw that `bt_keys_get_addr` may call `bt_unpair` (which may then call the `bond_deleted`) when `CONFIG_BT_KEYS_OVERWRITE_OLDEST=y`. 

I think you are right that this is not an issue, and will close the issue due to that. Thanks for testing!

closed 2023-08-14 by Thalley