zephyrproject-rtos/zephyr #60243: Settings does not support external backends not in handle order

daee-demant, 2023-07-11:
**Symptom**
On an implementation using an external settings backend, some subscriptions may end up incorrectly loaded.

**Cause**
The default Zephyr backend saves CCC settings using the ccc_set function in gatt.c. This saves the settings by iterating over the service table in handle order using the subroutine bt_gatt_foreach_attr. This adds an additional unspecified feature to the stored data that it is in handle order.

When the settings are loaded back using ccc_load, these values are read back with an assumption that the data is stored in this handle order. A subscription out of handle order will be dismissed with the debug message "Unable to restore CCC: handle [] cannot be found." 

The issue here is that Zephyr supports reading from an external backend that may or may not be in handle order, yet does not process out-of-handle order correctly.

**Issue seen**
We operate an external backend and we found that a lot of our subscriptions vanished on a reboot. This came from the fact that as our storing did not have a notion of handle order, a number of subscription services came in unsupported order and therefore were not loaded.

**Impact**
We lose subscription settings. A lot of phones will attempt to compensate, but it is not a nice way to treat our central.

**Proposals**
We would like to request that when an external backend is read into Zephyr, a function is implemented that sorts the incoming data in handle order.

We are unable to estimate the exact consequences of removing the handle check in ccc_load, and we are therefore not keen on suggesting doing this.

It would be possible to not do anything whatsoever, but in this case, we would suggest an explicit requirement somewhere indicating that what is passed into gatt.c must necessarily be in handle order.

carlescufi, 2023-07-11:
@daee-demant the settings maintainer, @de-nordic, is very busy at the moment. Could you or anyone at Oticon propose a Pull Request to fix this?

carlescufi, 2023-07-11:
> @carlescufi We are in no way opposed to doing it here, but we are also under a heavy load right now, so it might be a while before we can get around to it. It's unlikely to happen within the next two months. Just so you know.

Sounds good, this is not a high-priority issue in principle so it can wait.

Laczen, 2023-07-11:
This might be a more general problem, if I understand it correctly there is some assumption that the written data will be read back in a certain order. This is not guaranteed in e.g. nvs and might also lead to problems when using nvs as a backend.

kruithofa, 2023-07-20:
@daee-demant will you create a PR, and request comments from @de-nordic ?

de-nordic, 2023-07-20:
I do not think this is problem with settings here. Settings  may change order of info written to flash while running garbage collection. Because garbage collection affects ordering this may require garbage collection to touch stuff it would not have touched previously, to fix order.

jori-nordic, 2023-07-21:
> while running garbage collection

Settings have garbage collection? Not trying to hijack the thread, just curious how it works. It doesn't seem to be much documented, I could only find [this](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/services/settings/index.html#garbage-collection)

Laczen, 2023-07-21:
> Settings have garbage collection? Not trying to hijack the thread, just curious how it works. It doesn't seem to be much documented, I could only find [...]

Settings makes sure that it keeps a copy of each of the items that are stored in the backend, this is either done by the backend itsself (nvs) or by copying when a certain limit is reached (filesystem backend) or when a new sector is needed (fcb backend). While doing this there is no guarantee that the order of entries is maintained (a old sector might only contain a part of the entries, and copying them will move the starting entries to the "end" of the storage).

wdhpawa, 2023-08-07:
> I do not think this is problem with settings here.

Right, I guess the problem is not with the settings system itself, but with the way the Zephyr GATT module restores the CCC settings, i.e. incorrectly assuming that they arrive from the backend in a certain order.

jori-nordic, 2023-08-08:
Yes, it's not just GATT. For the whole Bluetooth stack we misunderstood (perhaps because of misleading API) how the settings subsystem works.
The only thing it is is a key-value store. That's it. It doesn't guarantee the order of reading nor writing.

But the API strongly hints that it has "namespaces" or that it is filesystem-like, by giving examples of keys as paths: ie. `id/serial` from the first paragraphs in the doc.

Someone not reading carefully - bt maintainers in that case, myself included - might see this and think that it saves and restores per-directory, and assume that it would store (and restore) the individual keys in-order. But settings makes no such promises in its API. 

So the solution is either a new settings-like subsystem, this time with atomic multi-key save & restore, or a data format change for which the migration code will have to be included. That new format would write all related settings (e.g. for an identity or for a paired device) into a single kv pair.

If we do end up making a new settings-like subsystem, my wishlist would also include the ability to delete a key while loading/restoring, and a safe-write system where if there is a failure to write the whole atomic group (e.g. power loss), then the previous value is restored.

jori-nordic, 2023-09-25:
@daee-demant are you fine with this? https://github.com/zephyrproject-rtos/zephyr/pull/63057

We also don't have time to prioritize working on a big settings rework.

wdhpawa, 2023-09-25:
@daee-demant is on vacation, but yes I think we are fine for now with the note in the documentation requiring the backend to restore items in handle order - assuming "handle" means CCC attribute handle and doesn't apply to other types of items. We'd have trouble doing anything more general than that, though, e.g. restoring items in the order they were saved.

PR #63057 (2023-09-26): Bluetooth: host: document settings backend requirement
This requirement was implicit as the only backend the host is currently tested with behaves this way.

Fixes #60243
closed 2023-09-26 by carlescufi