zephyrproject-rtos/zephyr #60246: Devices on a tca9548a I2C mux channel are assigned to the wrong bus

68ec020, 2023-07-11:
I use zephyr-v3.4.0 in a project. I have an "asahi-kasei,akm09918c" connected to one channel of a "ti,tca9548a" I2C multiplexer. If the sensor is accessed, the mux isn't changed to the channel the sensor is connected to. I looked into this and I think I found the reason: The bus assigned to the akm09918c is the parent I2C bus of the mux and not the channel I2C bus. 

Device tree overlay:
```
&i2c1 {
    status = "okay";
    pinctrl-0 = < &i2c1_default > ;
    pinctrl-names = "default";
    clock-frequency = < I2C_BITRATE_STANDARD >;
    
    mux: tca9548a@70 {
        compatible = "ti,tca9548a";
        reg = < 0x70 >;
        status = "okay";
        #address-cells = < 0x1 >;
        #size-cells = < 0x0 >;

        ch0: mux_i2c@0 {
            compatible = "ti,tca9546a-channel";
            reg = < 0x0 >;
            status = "okay";
            #address-cells = < 0x1 >;
            #size-cells = < 0x0 >;

            sens0: akm09918c@C{
                compatible = "asahi-kasei,akm09918c";
                status = "okay";
                reg = < 0x0C >;
            };
        };
    };
};
```

A small test app:
```
#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(app);

#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/drivers/sensor.h>
#include <zephyr/drivers/i2c.h>

struct akm09918c_config
{
    i2c_dt_spec i2c;
};

int main()
{
    const struct device *sens0 = DEVICE_DT_GET(DT_NODELABEL(sens0));
    akm09918c_config *sens0_cfg = (akm09918c_config *)sens0->config;

    printk("\ncfg->i2c.bus: %p\n", sens0_cfg->i2c.bus);
    printk("DT ch0:       %p\n", DEVICE_DT_GET(DT_NODELABEL(ch0)));
    printk("DT i2c1:      %p\n", DEVICE_DT_GET(DT_NODELABEL(i2c1)));

    while (true)
    {

        k_sleep(K_MSEC(1000));
    }

    return 0;
}
```

Generates this output:
```
[00:00:00.000,000] <err> AKM09918C: Failed to soft reset
*** Booting Zephyr OS build zephyr-v3.4.0 ***

cfg->i2c.bus: 0x10008dec
DT ch0:       0x10008e34
DT i2c1:      0x10008dec
```

As you can see the bus of the akm09918c is i2c1 (parent bus of the mux) and not ch0 (channel bus) as it should be.

Thank you in advance

Jan

68ec020, 2023-07-11:
BTW, I don't know how, but VS Code "knows" something is wrong:
```
Node should only occur on the i2c bus.
/soc/i2c@40004000/tca9548a@70/mux_i2c@0/akm09918c@E/
```

jhedberg, 2023-07-11:
@kurddt FYI

kurddt, 2023-08-03:
@68ec020 Did you test with previous version of Zephyr ?

sidmodi-mw, 2023-08-21:
Looks like the issue was documented in the [release notes](https://github.com/zephyrproject-rtos/zephyr/blob/4fc9686c410d990179c0f162b775bffeeb7fd95f/doc/releases/release-notes-3.2.rst#id1573:~:text=%3Agithub%3A%6049046%60%20%2D%20Cannot%20use%20devices%20behind%20I2C%20mux%20(TCA9548A)) for v3.2. Has there been any work done to it since then? The release notes for v3.3 and v3.4 don't mention it at all.

68ec020, 2023-08-23:
My college just realised we used the wrong channel device in the device tree. Our mux is a "ti,tca9548a" and the channel device was "ti,tca9546a-channel". So, there is nothing wrong and if we use the matching channel device, it works. Sorry for the confusion.

SzymonRichert, 2023-10-31:
> tca9548a

So if there is nothing wrong, please close this bug.

teburd, 2023-11-07:
I'm closing the issue as it doesn't seem to be a problem, please reopen if I'm wrong

closed 2023-11-07 by teburd