zephyrproject-rtos/zephyr #60274: soc: Add support for new Microchip PIC32CX SG41

pmcgaugh-daikin, 2023-07-11:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
This issue is related to the addition of the new Microchip Microcontroller, the PIC32CX ARM M4 family.

**Describe why you are asking for this support?**
This is a bigger integration and relates heavily to the sam0 drivers and atmel samd/e chips and want to ensure the integration is done in the most efficient way and in a way that the community agrees is most efficient. Speaking with Microchip engineers some of these chips variations are directly in place swap with certain variations of SAM chip.

For example, we are using SAMD51 chip and are looking at swapping with a PIC32CX SG41 chip. We did a quick test build and were able to run a rudimentary version of our application on that chip with few if any changes.

I am willing to support this SoC but hope to gain support from Microchip directly for this new product. Although I cannot say for sure, I hope they will be able to more directly provide support to the broader community if direct issues come up in relation to this chip

Depending on how heavily this chip integration leans on the sam0/samd/e chips it might benefit from tight support with that product family.

**Additional context**
[PIC32X Webpage](https://www.microchip.com/en-us/product/PIC32CX1025SG41128)
[Datasheet Download Link](https://ww1.microchip.com/downloads/aemDocuments/documents/MCU32/ProductDocuments/DataSheets/PIC32CX-SG41-SG60-SG61-Family-Data-Sheet-DS60001715.pdf)

pmcgaugh-daikin, 2023-07-12:
@nandojve I know you did a lot of work on the sam0 chips that this library could be heavily based upon or integrated with. I would like your input on how best manage this integration.

I will be inviting our Microchip partners to this conversation as they're willing to help us get this integrated for use on one of our new product lines. I want them to detail out which SOC this new SOC will be heavily based upon and any deviations we might need to account for.

nandojve, 2023-07-12:
Hi @pmcgaugh-daikin ,

This will require a new SoC support on Zephyr and, maybe, a new hal on my view. So, if you can get from MCHP a pack version with API compatible with the Atmel's one and those packs have a license that project can accept than it is possible to share current drivers. What I saw is that current packs from [MPLAB-Harmony](https://github.com/Microchip-MPLAB-Harmony/dev_packs/tree/master) have a proprietary license. Maybe MCHP provide a public version with is Apache like license but I don't have knowledge about it.

I think above is the starting point to evaluate if it is feasible.

nandojve, 2023-07-13:
You can understand better looking at this PR https://github.com/zephyrproject-rtos/hal_atmel/pull/30

MyGh64605, 2023-07-13:
We are reviewing our options internally at this time.
I am setting up a meeting with the appropriate Microchip personnel
@John.Billingsley@microchip.com