zephyrproject-rtos/zephyr #60295: User defined thrad tracing functions aren't called

alpsayin, 2023-07-12:
`sys_trace_thread_switched_in_user` and `sys_trace_thread_switched_in_user` are not called if `CONFIG_UART_CONSOLE`=`y`

Looking at the void `sys_trace_k_thread_switched_in(void)` implementation I feel like there's an oddity with the code block that disables calling user functions.
```
/* FIXME: Limitation of the current x86 EFI cosnole implementation. */
#if !defined(CONFIG_X86_EFI_CONSOLE) && !defined(CONFIG_UART_CONSOLE)
```
I got the impression the author (@enjiamai)  wanted to `! (X86_EFI_CONSOLE && UART_CONSOLE)` logically but mistakenly pushed the `!` into the condition which was supposed to result in `!X86_EFI_CONSOLE || !UART_CONSOLE`
I got this impression because I'm failing to see why `UART_CONSOLE` should disable calling user defined trace functions.
Please let me know if I'm wrong.

Please also mention any information which could help others to understand
the problem you're facing:
`west build -p -b qemu_riscv32 -T samples/subsys/tracing/sample.tracing.user`
p.s. this also won't work due to incorrect `prj_user.conf` file which I'm opening a ticket now for it too. But it's an obvious+trivial fix.
- What have you tried to diagnose or workaround this issue?
```
- #if !defined(CONFIG_X86_EFI_CONSOLE) && !defined(CONFIG_UART_CONSOLE)
+ #if !defined(CONFIG_X86_EFI_CONSOLE) || !defined(CONFIG_UART_CONSOLE)
```
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
Yes. #46918

**To Reproduce**
`west build -p -b qemu_riscv32 -T samples/subsys/tracing/sample.tracing.user`

**Expected behavior**
`sys_trace_thread_switched_in_user` and `sys_trace_thread_switched_in_user` would be called and print.

**Impact**
A local showstopper for thread tracing.

**Logs and console output**
Lack of prints isn't that interesting.

**Environment (please complete the following information):**
 Linux, SDK 0.16.1

**Additional context**
None

ycsin, 2023-07-21:
I had another look at the `tracing_user` related change in #46918, and find that change to be weird, the trancing_user APIs are supposed to be implemented by the user, it just so happens that its [sample](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/tracing/src/tracing_user.c#L9) is using `printk`. If the other changes in #46918 is causing that sample to fail, I think the better way should be to filter out that testcase?

cc @andyross @enjiamai @jhedberg @cfriedt

alpsayin, 2023-07-24:
> If the other changes in https://github.com/zephyrproject-rtos/zephyr/pull/46918 is causing that sample to fail, I think the better way should be to filter out that testcase?

That was my conclusion too. Thanks for the PR btw 💯

ycsin, 2023-08-19:
Closed by #60666

closed 2023-08-19 by ycsin