zephyrproject-rtos/zephyr #60297: Incorrect EXTRA_ARGS in samples/subsys/tracing/sample.yaml

alpsayin, 2023-07-12:
**Describe the bug**
Extremely trivial. `EXTRA_ARGS` section has `CONF_FILE` parameters defined with quotes which causes build failure.
e.g.
```   
 extra_args: CONF_FILE="prj_user.conf"
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 Linux.
 - What have you tried to diagnose or workaround this issue?
Yes, removed the quotes and voila.

```
-    extra_args: CONF_FILE="prj_user.conf"
+    extra_args: CONF_FILE=prj_user.conf
```
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
Not really a regression, because there seems to be multiple `EXTRA_ARGS` that use quotes from multiple authors, so somehow this makes me think that I'm doing something wrong. So please let me know...

**To Reproduce**
Steps to reproduce the behavior:
``` west build -p -b qemu_riscv32 -T samples/subsys/tracing/sample.tracing.user```

**Expected behavior**
Successful configuration.

**Impact**
Technically a "showstopper" but really an annoyance because it's trivial at worst.

**Logs and console output**
```
-- Including generated dts.cmake file: /home/asayin/zephyr_sdks/microblaze_devel/zephyr/build/zephyr/dts.cmake
CMake Error at /home/asayin/zephyr_sdks/microblaze_devel/zephyr/cmake/modules/kconfig.cmake:275 (message):
  File not found:
  /home/asayin/zephyr_sdks/microblaze_devel/zephyr/samples/subsys/tracing/"prj_user.conf"
Call Stack (most recent call first):
  /home/asayin/zephyr_sdks/microblaze_devel/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
  /home/asayin/zephyr_sdks/microblaze_devel/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/asayin/zephyr_sdks/microblaze_devel/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /home/asayin/zephyr_sdks/microblaze_devel/cmake/bin/cmake -DWEST_PYTHON=/home/asayin/zephyr_sdks/microblaze_devel/python/bin/python3.8 -B/home/asayin/zephyr_sdks/microblaze_devel/zephyr/build -GNinja -DBOARD=qemu_riscv32 '-DCONF_FILE="prj_user.conf"' -S/home/asayin/zephyr_sdks/microblaze_devel/zephyr/samples/subsys/tracing
```

**Environment (please complete the following information):**
Linux, SDK 0.16.1

**Additional context**
Somewhat relates to #60295

nordicjm, 2023-07-12:
This doesn't look right, @PerMac any clue what's going on, since quotes are needed when testing elsewhere e.g. https://github.com/zephyrproject-rtos/zephyr/pull/59747 ?

PerMac, 2023-07-14:
I believe it is an issue with west_commands for `-T` where it tries to load twister's configs from yaml. Calling this test through twister passes and has `CONFIG_TRACING_USER=y` in its .config.

PerMac, 2023-07-14:
I can look at it further after the weekend

PR #60439 (2023-07-19): west_commands: Fix parsing of extra args for test-item
An arg --test-item makes west loading twister's test configurations form sample/testcase.yaml. It has to mirror twister's behavior. It was not the case with "extra_args" section. Quotation  marks were not removed in west as they were in twister. The quotation marks have to be removed from the extra_args section but left in extra_configs. The commit adds differentiation for those.

fixes: #60297
commit 93031ad5 (2023-07-19): west_commands: Fix parsing of extra args for test-item

An arg --test-item makes west loading twister's test configurations
form sample/testcase.yaml. It has to mirror twister's behavior.
It was not the case with "extra_args" section. Quotation  marks were
not removed in west as they were in twister. The quotation marks
have to be removed from the extra_args section but left in
extra_configs. The commit adds differentiation for those.

fixes: #60297
closed 2023-07-19 by nashif