zephyrproject-rtos/zephyr #60299: Set time-to-live for Multicasts

bonetor, 2023-07-12:
**Describe the bug**
As mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/53587 I implemented a Multicast / socket connection to receive Multicasts send via UDP by a client software running on a PC and I send back UDP packages to the client software as an answer to the received packages.

The _time-to-live_ of the package send back is set to 64. I guess that is due to the initial configuration `NET_INITIAL_TTL` set at `zephyr/subsys/net/ip/Kconfig.ipv4`. This configuration is set for the IPv4 interface. I am using only one interface.

I want to reduce the TTL for the multicast / socket connection. I checked `zsock_setsockopt` and the available options, but did not find any matching. 

Is there a possibility to set the TTL for the multicasts, e.g. like the IP_MULTICAST_TTL option described at https://man7.org/linux/man-pages/man7/ip.7.html?

I am using:
- Zephyr v3.1.0
- Toolchain Zephyr SDK
- STM32H747 DISCO Board

rlubos, 2023-07-19:
There is currently no way to set a custom TTL for IPv4 multicast packets. At the `net_context` level there is a `net_context_set_ipv4_ttl()`, but the `ipv4_ttl` field it sets is applied for both, unicast and multicast packets.
It'd be needed to:
* Add a new field, `ipv4_multicast_ttl` to the `net_context` structure,
* Add getter/setter function to the `net_context` API
* Implement `IP_MULTICAST_TTL` that makes use of above
* And finally, update `net_context_create_ipv4_new()` to differentiate between unicast/multicast destinations

rlubos, 2023-07-19:
Also, it's a feature request, not a bug, so changing the label.

PR #65886 (2023-12-04): Add support for multicast IPv4 TTL and IPv6 hop limit
Add option support for adjusting the IPv4 multicast time-to-live value. Add also tests for both unicast and multicast TTL settings.
Add also similar IPv6 hop limit option for both unicast and multicast IPv6 packets + tests.
Add checks and tests for IPv4 TTL 0 or IPv6 hop limit 0 when sending packet, the packet should dropped in this case.

Fixes #60299
commit de0268de (2023-12-04): net: context: Add support for adjusting IPv4 multicast ttl

Add option support for adjusting the IPv4 multicast
time-to-live value.

Fixes #60299
closed 2023-12-04 by henrikbrixandersen