zephyrproject-rtos/zephyr #60330: upsquared: tests/kernel/spinlock/kernel.multiprocessing.spinlock fails.

gbarkadiusz, 2023-07-13:
**Describe the bug**
`tests/kernel/spinlock/kernel.multiprocessing.spinlock` fails.

**Bisect report**
```
28a8b151cf54d8482c1518e2a120a6c54a463c93 is the first bad commit
commit 28a8b151cf54d8482c1518e2a120a6c54a463c93
Author: Christopher Friedt <cfriedt@meta.com>
Date:   Sun Jul 2 10:58:38 2023 +0200

    tests: kernel: spinlock: add test for k_spin_trylock()
    
    Add a testcase to exercise two cases:
    * when `k_spin_trylock()` fails (lock is busy)
    * when `k_spin_trylock()` succeeds (lock is acquired)
    
    We use the same machinery for checking for a recursive mutex
    as `k_spin_lock()`.
    
    Signed-off-by: Christopher Friedt <cfriedt@meta.com>

 tests/kernel/spinlock/src/main.c | 58 +++++++++++++++++++++++++++++++++++++---
 1 file changed, 54 insertions(+), 4 deletions(-)
```

**To Reproduce**
` twister --device-testing --hardware-map $HOME/up2_double.map -s tests/kernel/spinlock/kernel.multiprocessing.spinlock -v`

**Logs and console output**
```
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-1052-gf2e0f787bf83
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report <path_to>/zephyrproject/zephyr/twister-out/testplan.json

Device testing on:

| Platform   | ID    | Serial device   |
|------------|-------|-----------------|
| up_squared | up2   | /dev/ttyUP2     |
| up_squared | up2-2 | /dev/ttyUP2_2   |

INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 up_squared                tests/kernel/spinlock/kernel.multiprocessing.spinlock  FAILED Failed (device 23.750s)
ERROR   - see: <path_to>/twister-out/up_squared/tests/kernel/spinlock/kernel.multiprocessing.spinlock/handler.log

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 0 of 1 test configurations passed (0.00%), 1 failed, 0 errored, 0 skipped with 0 warnings in 29.69 seconds
INFO    - In total 8 test cases were executed, 0 skipped on 1 out of total 584 platforms (0.17%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
```
hanlder.log contains.

```
...
    ===================================================================
    58  START - test_trylock
    59  Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/spinlock/src/main.c:222: spinlock_test_trylock: trylock_failures > 0 is false
    60  FAIL - test_trylock in 1.034 seconds
    61  ===================================================================
    62  TESTSUITE spinlock failed.
    63  ------ TESTSUITE SUMMARY START ------
    64  SUITE FAIL -  85.71% [spinlock]: pass = 6, fail = 1, skip = 1, total = 8 duration = 2.112 seconds
    65  - PASS - [spinlock.test_spinlock_basic] duration = 0.001 seconds
    66  - PASS - [spinlock.test_spinlock_bounce] duration = 1.021 seconds
    67  - SKIP - [spinlock.test_spinlock_lock_time_limit] duration = 0.001 seconds
    68  - PASS - [spinlock.test_spinlock_mutual_exclusion] duration = 0.001 seconds
    69  - PASS - [spinlock.test_spinlock_no_recursive] duration = 0.018 seconds
    70  - PASS - [spinlock.test_spinlock_release_error] duration = 0.018 seconds
    71  - PASS - [spinlock.test_spinlock_unlock_error] duration = 0.018 seconds
    72  - FAIL - [spinlock.test_trylock] duration = 1.034 seconds
    73  ------ TESTSUITE SUMMARY END ------
    74  ===================================================================
    75  RunID: d685f227c8978498acdaab3119a8c20d
    76  PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.0
 - Zephyr version: zephyr-v3.4.0-1052-gf2e0f787bf83

fabiobaltieri, 2023-07-18:
@cfriedt could you look into this?

cfriedt, 2023-07-19:
Reduced priority, since there is a fix in review, and also, because it's probably an outlier that the success rate is 100%.

cfriedt, 2023-07-19:
I'm going to reclassify this as a platform misconfiguration bug. AFAICT, no bug exists in the test under the assumption that SMP is properly configured for the hardware.

Also, AFAICT, Apollo Lake does not support SMP or HyperThreading either.
https://en.wikipedia.org/wiki/List_of_Intel_Atom_processors

My suggestion would be to remove the offending `CONFIG_SMP=y` line from `up_squared_defconfig`.

```diff
diff --git a/boards/x86/up_squared/up_squared_defconfig b/boards/x86/up_squared/up_squared_defconfig
index dfc5db7d39..2d26584b73 100644
--- a/boards/x86/up_squared/up_squared_defconfig
+++ b/boards/x86/up_squared/up_squared_defconfig
@@ -8,6 +8,5 @@ CONFIG_CONSOLE=y
 CONFIG_UART_CONSOLE=y
 CONFIG_SERIAL=y
 CONFIG_X2APIC=y
-CONFIG_SMP=y
 CONFIG_BUILD_OUTPUT_EFI=y
 CONFIG_BUILD_NO_GAP_FILL=y
```

cfriedt, 2023-09-18:
I will double check, but I think that this might have a solution. I don't have the hardware the test though @jhedberg - do you have it?

jhedberg, 2023-09-25:
> I will double check, but I think that this might have a solution. I don't have the hardware the test though @jhedberg - do you have it?

@cfriedt yes, I think I have a board somewhere (either at the office or at home). Just let me know what you want me to test.

cfriedt, 2023-09-25:
> > I will double check, but I think that this might have a solution. I don't have the hardware the test though @jhedberg - do you have it?
> 
> @cfriedt yes, I think I have a board somewhere (either at the office or at home). Just let me know what you want me to test.

Oh - it's just that this bug report is a bit old and I think this issue might already be solved. So if you can successfully run the test mentioned in the description, we can probably close.

Bonus if you can find which PR fixed it 😅

KamilxPaszkiet, 2023-10-11:
> Oh - it's just that this bug report is a bit old and I think this issue might already be solved. So if you can successfully run the test mentioned in the description, we can probably close.
>
> Bonus [...]

Unfortunately, we still have a reproduction of this bug

cfriedt, 2023-10-11:
@gbarkadiusz - are you able to provide more details? E.g. the particular assertion that fails, a stack trace, or values that are inconsistent?

Can you rebase on the latest Zephyr from `main` and re-run your experiment?

I would suggest running your `twister` similarly to how it is run in Zephyr CI, with `-i --retry-failed 3`.

I would like to help remediate this issue but unfortunately do not have the hardware in question, would need to see additional diagnostics, and am otherwise unable to reproduce.

I know that this particular platform has had some issues in the past with SMP.

@jhedberg - did you get a chance to look into it?

KamilxPaszkiet, 2023-10-23:
I added a second bug from the same category
`kernel.multiprocessing.spinlock.minimallibc`

finikorg, 2023-10-25:
This is fixing the issue for EHL: https://github.com/zephyrproject-rtos/zephyr/pull/64354
Could you @KamilxPaszkiet  please verify it?

finikorg, 2023-10-25:
> This is fixing the issue for EHL: #64354 Could you @KamilxPaszkiet please verify it?

If this does not help, I have another PR which works for RPL and ADL: https://github.com/zephyrproject-rtos/zephyr/pull/64368

KamilxPaszkiet, 2023-10-25:
> > This is fixing the issue for EHL: #64354 Could you @KamilxPaszkiet please verify it?
> 
> If this does not help, I have another PR which works for RPL and ADL: #64368

yes, this pr solves the problem ;)

finikorg, 2023-10-25:
> > > This is fixing the issue for EHL: #64354 Could you @KamilxPaszkiet please verify it?
> > 
> > 
> > If this does not help, I have another PR which works for RPL and ADL: #64368
> 
> yes, this pr solves the problem ;)

So which PR has fixed the issue?

KamilxPaszkiet, 2023-10-26:
> > > > This is fixing the issue for EHL: #64354 Could you @KamilxPaszkiet please verify it?
> > > 
> > > 
> > > If this does not help, I have another PR which works for RPL and ADL: #64368
> > 
> > 
> > yes, this pr solves the problem ;)
> 
> So which PR has fixed the issue?

Yes, this [PR](https://github.com/zephyrproject-rtos/zephyr/pull/64368) fix the issue

finikorg, 2023-10-26:
OK, then do not close until PR is merged.

KamilxPaszkiet, 2023-10-26:
> OK, then do not close until PR is merged.

ok, sorry

cfriedt, 2023-10-26:
Just thought I would leave this here. The correct fix for this bug is arguably Ticketed Spinlocks.

https://github.com/zephyrproject-rtos/zephyr/pull/61541

jhedberg, 2023-11-22:
Just tried running this with the latest tree, and the test works fine:
```
------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [spinlock]: pass = 7, fail = 0, skip = 1, total = 8 duration = 1.219 seconds
 - PASS - [spinlock.test_spinlock_basic] duration = 0.001 seconds
 - PASS - [spinlock.test_spinlock_bounce] duration = 0.117 seconds
 - SKIP - [spinlock.test_spinlock_lock_time_limit] duration = 0.001 seconds
 - PASS - [spinlock.test_spinlock_mutual_exclusion] duration = 0.001 seconds
 - PASS - [spinlock.test_spinlock_no_recursive] duration = 0.012 seconds
 - PASS - [spinlock.test_spinlock_release_error] duration = 0.012 seconds
 - PASS - [spinlock.test_spinlock_unlock_error] duration = 0.012 seconds
 - PASS - [spinlock.test_trylock] duration = 1.063 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

jhedberg, 2023-11-22:
> Just tried running this with the latest tree, and the test works fine:

Sorry, just as I wrote this, I realised that the multiprocessing test doesn't get enabled with the default configuration for that application. I'll try to enable it and run it again.

jhedberg, 2023-11-22:
jjkk

> > Just tried running this with the latest tree, and the test works fine:
> 
> Sorry, just as I wrote this, I realised that the multiprocessing test doesn't get enabled with the default configuration for that application. I'll try to enable it and run it again.

Actually no, even though there was no "multiprocessing" in my output it seems these are the same tests (I also checked that I have SMP enabled in the config). So it seems this is passing with the latest tree. I don't have a way to run this through twister however (`--device-testing`), rather I just built the app and booted the EFI file through USB on my up_squared board. Anyway, @gbarkadiusz could you check with the latest tree please?

gbarkadiusz, 2023-11-22:
There is no reproduction anymore.