zephyrproject-rtos/zephyr #60335: ESP32C3: Not able to boot with Zephyr 3.4.0

darshan-shah-git, 2023-07-13:
**Describe the bug**
With Zephyr 3.4.0, device is not booting up on ESP32-C3.

- Device: ESP32C3-DevKitM
- Example: task_wdt with updated configuration

**To Reproduce**
Steps to reproduce the behavior:
[task_wdt.zip](https://github.com/zephyrproject-rtos/zephyr/files/12037645/task_wdt.zip)

1. Extract attached sample application
2. west build -b esp32c3_devkitm --sysbuild . -p --
3. west flash
4. See error

**Expected behavior**
Device should boot-up and run task wdt example

**Impact**
Device is not able to boot-up. It is blocking further development.

**Logs and console output**
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd8000,len:0x10a0
load:0x403cadc8,len:0x3b7c
load:0x403d0000,len:0x16dc
load:0x403d1700,len:0x228
entry 0x403cc528
[esp32c3] [INF] MCUboot 2nd stage bootloader
[esp32c3] [INF] compiled at Jul 13 2023 14:47:35
[esp32c3] [INF] Chip revision: 4
[esp32c3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 4MB
[esp32c3] [INF] Swap type: none
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] Application start=40399D10h
[esp32c3] [INF] DRAM segment: paddr=0003C8E4h, vaddr=3FCC7608h, size=026BCh (  9916) load
[esp32c3] [INF] IRAM segment: paddr=00022860h, vaddr=40380000h, size=1A084h (106628) load
[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=2281Eh (141342) map
[esp32c3] [INF] IROM segment: paddr=00050000h, vaddr=42040000h, size=866EEh (550638) map

```

**Environment (please complete the following information):**
 - OS: Linux 18.04
 - Toolchain - zephyr 0.16.1 
 - Version - Zephyr 3.4.0 Tag

**Additional context**
- Notice that if we disable UART1, device is able to boot.

darshan-shah-git, 2023-07-14:
@sylvioalves are you able to reproduce at your end?

fabiobaltieri, 2023-07-18:
@sylvioalves can you look into this? Not sure if stock mcuboot is supported on this platform or not.

jhedberg, 2023-09-26:
> I am able to boot after disabling usb-serial in overlay file.

@darshan-shah-git do you mean that this resolves the issue, or it's just a workaround? (trying to decide whether to close this issue or not)

darshan-shah-git, 2023-09-29:
> > I am able to boot after disabling usb-serial in overlay file.
> 
> @darshan-shah-git do you mean that this resolves the issue, or it's just a workaround? (trying to decide whether to close this issue or not)

No it was work-around.

INemesisI, 2023-11-10:
I am 90% sure I am running in a similar issue, specifically when enabling mcuboot on zephyr 3.5

As my main application does not boot when enabling deferred logging, I tested the logger sample and was able to find a stock sample which crashes on boot:

```
west build -p -b esp32c3_devkitm zephyr/samples/subsys/logging/logger --sysbuild
```

Here is the console log:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd8000,len:0x1028
load:0x403cada8,len:0x3ce4
load:0x403d0000,len:0x1a24
entry 0x403cc736
[esp32c3] [INF] MCUboot 2nd stage bootloader
[esp32c3] [INF] compiled at Nov 10 2023 16:00:11
[esp32c3] [INF] Chip revision: 4
[esp32c3] [INF] SPI Flash speed: 40MHz, mode: DIO, size: 4MB
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] Application start=403829ECh
[esp32c3] [INF] DRAM segment: paddr=00004CACh, vaddr=3FC879D0h, size=003B4h (   948) load
[esp32c3] [INF] IRAM segment: paddr=00001860h, vaddr=40380000h, size=0344Ch ( 13388) load
[esp32c3] [INF] DROM segment: paddr=00010040h, vaddr=3C000040h, size=01820h (  6176) map
[esp32c3] [INF] IROM segment: paddr=00020000h, vaddr=42020000h, size=049CAh ( 18890) map
*** Booting Zephyr OS build zephyr-v3.5.0-1313-g5bd18886e96a ***
 [00:00:00.000,000] <err> os: 
 [00:00:00.000,000] <err> os:  mcause: 5, Load access fault
 [00:00:00.000,000] <err> os:   mtval: c
 [00:00:00.000,000] <err> os:      a0: 00000000    t0: 4005890e
 [00:00:00.000,000] <err> os:      a1: 00000088    t1: 42023856
 [00:00:00.000,000] <err> os:      a2: 00000000    t2: 00000020
 [00:00:00.000,000] <err> os:      a3: 00000005    t3: 00000004
 [00:00:00.000,000] <err> os:      a4: 00000000    t4: 0000004c
 [00:00:00.000,000] <err> os:      a5: 3fc83ab8    t5: 148001a2
 [00:00:00.000,000] <err> os:      a6: 3fc87a5c    t6: 774afc5e
 [00:00:00.000,000] <err> os:      a7: 0000000e
 [00:00:00.000,000] <err> os:      ra: 42020eac
 [00:00:00.000,000] <err> os:    mepc: 40381274
 [00:00:00.000,000] <err> os: mstatus: 00001800
 [00:00:00.000,000] <err> os: 
 [00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
 [00:00:00.000,000] <err> os: Current thread: 0x3fc83ab8 (unknown)
 [00:00:00.095,000] <err> os: Halting system
```

I hooked up GDB, it crashes when trying to start the static sample thread. It seems like the memory is misaligned by one byte, as the thread_data object in `zephyr/rystaos/zephyr/kernel/thread.c:860` gives me the following object:
```
>>> print thread_data
$1 = (struct _static_thread_data *) 0x3c0026b8 <_k_thread_data_log_demo_thread_id>
>>> print *(thread_data)
thread_data = {
  init_thread = 0x3fc86250 <_k_thread_stack_log_demo_thread_id>,
  init_stack = 0x400,
  init_stack_size = 1107427752,
  init_entry = 0x0,
  init_p1 = 0x0,
  init_p2 = 0x0,
  init_p3 = 0xe,
  init_prio = 0,
  init_options = 1006633908,
  init_name = 0x1 "",
  init_delay = {
    ticks = 1152921981348216832
  }
}
```
Every variable has the value the next variable is supposed to have. I printed the address shifted by one byte and, surely, I got the correct object:
```
>>> print *((struct _static_thread_data *) 0x3c0026b4)
$11 = {
  init_thread = 0x3fc85ad8 <_k_thread_obj_log_demo_thread_id>,
  init_stack = 0x3fc86250 <_k_thread_stack_log_demo_thread_id>,
  init_stack_size = 1024,
  init_entry = 0x420201a8 <log_demo_supervisor>,
[...]
```

I noticed this misalignment already multiple times. Most often, when adding a new device to the board. In this case, I had to add a second dummy device to get a properly aligned memory again. The sample runs fine when compiling without mcuboot. 

My main application behaves slightly different. It works fine in deferred mode, but gives no log output when mcuboot is enabled.

jonkensta, 2023-12-29:
I ran the steps to reproduce on a similar board `esp32c3_luatos_core` (USB version without the CH343 chip) on the HEAD of main (c05a483ba4251589b713a1231fb6ad4c75cebbaa) and the given application `task_wdt` was able to boot. I suggest we can close this issue unless more info comes to light.

closed 2024-03-13 by github-actions[bot]