zephyrproject-rtos/zephyr #60345: ATSAME70Q21 - custom bootloader fails -jumps to application.

pavithvishnu, 2023-07-13:
I have created a small bootloader(sample bootloader) which just print msg and jumps to application( hello-world).

This is my partition detail in board file.
&flash0 {
	partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;
        /* Sector 0: bootloader (128 kbytes) */
        boot_partition: partition@0 {
            label = "mcuboot";
            reg = <0x00000000 0x20000>;
        };
        /* Sector 1: Application (128 kbytes) */
        slot0_partition: partition@40000 {
            label = "image-0";
            reg = <0x00040000 0x20000>;
        };
	};
};
This is my bootloader code:
	#include <zephyr/kernel.h>
	#include <soc.h>

	extern void sys_clock_disable(void);
	typedef void (*fpJumpHandler)(void);
	fpJumpHandler fpApplicationResetHandler;

	int main(void)
	{
		printk("Hello from boot loader! \n");
		uint32_t AppBaseAddr = (uint32_t)(0x440000);
		K_MSEC(5000);
		sys_clock_disable();
		__disable_irq();
		/** Barriers */
		__DSB();
		__ISB();
		/** Update vector table */
		SCB->VTOR = AppBaseAddr & SCB_VTOR_TBLOFF_Msk;
		/** Barriers */
		__DSB();
		__ISB();

		/** Enable interrupts */
		__enable_irq();
		__set_MSP(AppBaseAddr);
		/** Call Application reset handler */
		fpApplicationResetHandler = (fpJumpHandler)(*((uint32_t*)(temp)));
		(*fpApplicationResetHandler)();

		return 0;
	}

As per the datasheet of ATSAME70, by default, in zephyr the base address is set 0x400000 (hex file is generated as per that), so that means bootloader is loaded in 0x400000,  and my hello-world app is loaded in 0x440000,
after flashing bootloader and app in hardware. Im getting usage access fault after execution of (*fpApplicationResetHandler)(); and board gets reset and starts again from boot loader. 

Unable to jump to application.

can any one support here ?

pavithvishnu, 2023-07-13:
Not sure, how to verify that.?

Laczen, 2023-07-13:
> @Laczen Not sure, how to verify that.?

The hex file of the application should have a offset of 0x440000.

Laczen, 2023-07-13:
> > > @Laczen Not sure, how to verify that.?
> > 
> > 
> > The hex file of the application should have a offset of 0x440000.
> 
> I have changed the offset 0x440000 while compiling the app , i have also verified the offset in the generated hex file.

What do you mean by changed the offset while compiling the app ? The app should be compiled for offset 0x440000.