zephyrproject-rtos/zephyr #60358: tests: drivers.gpio.gpio_get_direction.test_input_output failure

LukaszMrugala, 2023-07-13:
**Describe the bug**
The drivers.gpio.gpio_get_direction.test_input_output test is failing on MEC17.
 - Platform: MEC172x 
 - 79ee5a876fce82e4bd22f865032567ba2cab9327 is the first bad commit
 - Before that, the test in question has been skipped

**To Reproduce**
Run the test in question on MEC17. Tested on two machines using dediprog runner.

**Expected behavior**
Test passes.

**Impact**
Failure of a GPIO test puts in question GPIO implementation.

**Logs and console output**
```
^@*** Booting Zephyr OS build v3.4.0-rc3-85-g785d9bdc6709 ***
Running TESTSUITE gpio_get_direction
===================================================================
START - test_disconnect
PASS - test_disconnect in 0.000 seconds
===================================================================
START - test_input
PASS - test_input in 0.000 seconds
===================================================================
START - test_input_output
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_get_direction/src/main.c:121: gpio_get_direction_test_input_output: (true not equal to rv)
gpio_pin_is_input() failed: 0
FAIL - test_input_output in 0.017 seconds
===================================================================
START - test_output
PASS - test_output in 0.001 seconds
=============^@*** Booting Zephyr OS build v3.4.0-rc3-85-g785d9bdc6709 ***
Running TESTSUITE gpio_get_direction
===================================================================
START - test_disconnect
PASS - test_disconnect in 0.000 seconds
===================================================================
START - test_input
PASS - test_input in 0.000 seconds
===================================================================
START - test_input_output
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_get_direction/src/main.c:121: gpio_get_direction_test_input_output: (true not equal to rv)
gpio_pin_is_input() failed: 0
FAIL - test_input_output in 0.017 seconds
===================================================================
START - test_output
PASS - test_output in 0.001 seconds
===================================================================
TESTSUITE gpio_get_direction failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  75.00% [gpio_get_direction]: pass = 3, fail = 1, skip = 0, total = 4 duration = 0.018 seconds
- PASS - [gpio_get_direction.test_disconnect] duration = 0.000 seconds
- PASS - [gpio_get_direction.test_input] duration = 0.000 seconds
- FAIL - [gpio_get_direction.test_input_output] duration = 0.017 seconds
- PASS - [gpio_get_direction.test_output] duration = 0.001 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 29d91c2c368ab77dac7739aca0f983ec
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: 785d9bdc6709ed47616aec2ef09fabcea4e06ddf

Manimaran-A, 2023-07-18:
Ok, i check once and update it.

Manimaran-A, 2023-08-08:
Just an update, started looking in to this failure. Shortly create PR.

Manimaran-A, 2023-08-09:
Created PR #61314 to fix the failure.

PR #61314 (2023-08-10): drivers: gpio: mchp: Twister fix for test_input_output failure
Added logic to return ENOTSUP, if input-output GPIO direction requested.

fix #60358
closed 2023-08-10 by nashif