zephyrproject-rtos/zephyr #60359: tests: tests.drivers.adc.adc_api failure and timeout

LukaszMrugala, 2023-07-13:
**Describe the bug**
tests.drivers.adc.adc_api fails. Specific subtests failing are:
 - test_adc_asynchronous_call - FAIL
 - test_adc_repeated_samplings - ASSERTION FAIL, Kernel Panic

Platform: MEC 172x

Error commit history:
 - Test skipped before                                cdcfc2926fe6c051afd0c26e7e105e0800346823
 - CMAKE build failure since                      cdcfc2926fe6c051afd0c26e7e105e0800346823
 - Malformed build during building since f429f339fc8fb9adcb5d87559cabe0f164106601
 - CMAKE build failure since                      336aa9dc8899acf432b4865728225fe6d7786ab3
 - Timeout since                                        1cfc85bab89deffea303e115f7d9d8a81a7b8cb0

**To Reproduce**
Run the test in question on MEC17. Tested on two machines using dediprog runner.

**Expected behavior**
Test passes.

**Impact**
Failure of a ADC test puts in question ADC implementation.

**Logs and console output**
```
^@*** Booting Zephyr OS build v3.4.0-rc3-85-g785d9bdc6709 ***
Running TESTSUITE adc_basic
===================================================================
START - test_adc_asynchronous_call
Samples read: 0x0fff 0x0fff 0x0fff 0x0fff 0xffff8000
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/adc/adc_api/src/test_adc.c:72: check_samples: (INVALID_ADC_VALUE equal to sample_value)
[4] should be filled
FAIL - test_adc_asynchronous_call in 0.021 seconds
===================================================================
START - test_adc_invalid_request
Samples read: 0x0fff 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000
PASS - test_adc_invalid_request in 0.007 seconds
===================================================================
START - test_adc_repeated_s^A^@*** Booting Zephyr OS build v3.4.0-rc3-85-g785d9bdc6709 ***
Running TESTSUITE adc_basic
===================================================================
START - test_adc_asynchronous_call
Samples read: 0x0fff 0x0fff 0x0fff 0x0fff 0xffff8000
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/adc/adc_api/src/test_adc.c:72: check_samples: (INVALID_ADC_VALUE equal to sample_value)
[4] should be filled
FAIL - test_adc_asynchronous_call in 0.021 seconds
===================================================================
START - test_adc_invalid_request
Samples read: 0x0fff 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000
PASS - test_adc_invalid_request in 0.007 seconds
===================================================================
START - test_adc_repeated_samplings
repeated_samplings_callback: done 1
Samples read: 0x0fff 0x0fff 0xffff8000 0xffff8000 0xffff8000 0xffff8000
repeated_samplings_callback: done 2
Samples read: 0x0fff 0x0fff 0x0fff 0xffff8000
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/adc/adc_api/src/test_adc.c:72: check_samples: (INVALID_ADC_VALUE equal to sample_value)
[3] should be filled
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1738
aborting essential thread 0x118210
E: r0/a1:  0x00000004  r1/a2:  0x000006ca  r2/a3:  0x00118210
E: r3/a4:  0x000c319d r12/ip:  0x00000000 r14/lr:  0x000c514f
E:  xpsr:  0x4100005e
E: Faulting instruction address (r15/pc): 0x000c6840
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x118210 (idle)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: 785d9bdc6709ed47616aec2ef09fabcea4e06ddf

fabiobaltieri, 2023-07-18:
@jvasanth1 could you look into this? Hard to trace it down to a specific regression so it may need to be look into from the start.

pkunieck, 2023-10-17:
Can we have an update on this issue?

pkunieck, 2023-10-24:
Any new updates? @jvasanth1

jvasanth1, 2023-10-24:
Somehow we didn't notice this issue; will start looking into this and update

jvasanth1, 2023-10-24:
We are seeing this issue only when CONFIG_PM=y.

Could you confirm if CONFIG_PM=y for this test?

nashif, 2023-10-25:
yes, CONFIG_PM is enabled, this is set by the SoC which is wrong. Platforms should never enable PM by default as it might interfere with testing in general, only an application should enable PM, not the default SoC config.

PR #64370 (2023-10-26): soc: microchip_mec: do not enable PM in soc
PM should not be enabled by default at the soc level. This is an
application decision and not a platform decision whether to enable or
disable PM features.

Enabling PM by default will result in descripencies and test scope and
failures in tests that do not account for PM being enabled.

Fixes #60359
commit 51f6c426 (2023-10-26): soc: microchip_mec: do not enable PM in soc

PM should not be enabled by default at the soc level. This is an
application decision and not a platform decision whether to enable or
disable PM features.

Enabling PM by default will result in descripencies and test scope and
failures in tests that do not account for PM being enabled.

Fixes #60359
closed 2023-10-26 by MaureenHelm