zephyrproject-rtos/zephyr #60372: build system bug

JerryDennisEasley, 2023-07-14:
I have just installed west as was directed here: https://docs.zephyrproject.org/latest/develop/getting_started/index.html

When I attempted to build the blinky sample, it failed apparently because a %HOMEPAGE% that contains an embedded space is incorrectly parsed.  This is evinced by these two if statements starting at about line 77 in the Kconfig.makefile:

if(CONF_FILE)
  string(CONFIGURE "${CONF_FILE}" CONF_FILE_EXPANDED)
  string(REPLACE " " ";" CONF_FILE_AS_LIST "${CONF_FILE_EXPANDED}")
endif()

if(EXTRA_CONF_FILE)
  string(CONFIGURE "${EXTRA_CONF_FILE}" EXTRA_CONF_FILE_EXPANDED)
  string(REPLACE " " ";" EXTRA_CONF_FILE_AS_LIST "${EXTRA_CONF_FILE_EXPANDED}")
endif()

This code seems to be processing a list of path names and using a space as a delimiter.  This will fail when the %HOMEPATH% contains an embedded space as does mine, "C:\Users\Jerry Easley"; it splits this string into two "C:\Users\Jerry" and "Easley" neither of which can be found.  Since both Windows and Linux allow imbedded spaces within path names, I think this is a bug.

I also had an installation failure due to the same problem, however, I was able to work around that problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Just create a %HOMEPAGE% which contains an embedded space such as "C:\Users\Firstname Lastname"
2. install west
3. build blinky

**Expected behavior**
I expect that you will receive a "Fatal Error" on about line 275 in the Kconfig.makefie

**Impact**
I guess I could work around this problem by changing my directory structure and eliminating the embedded space but I'm afraid I would have reinstall quite a bit of other software

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain Zephyr SDK

Regards,
Jerry

marc-hb, 2023-07-14:
Thanks for the detailed bug report! The title is surprisingly vague though, can you please rephrase with some of the specific keywords you used in the description?

You found one place where the build system does not support whitespace in paths. I'm afraid there are others :-( This limitation should probably be documented. I thought I remembered seeing that mentioned in the documentation but I can't find it right now.

nordicjm, 2023-07-14:
This is a known limitation and is listed in the documentation https://docs.zephyrproject.org/latest/develop/application/index.html#creating-an-application-by-hand

JerryDennisEasley, 2023-07-14:
Well I guess I can live with this limitation, but I would like to point out that I was not apprised of this problem; neither of the notices you cite are in the "Getting Started" documentation; it would be nice to put a similar notice in the getting started documentation, better yet put in a check in the installation procedure.

marc-hb, 2023-07-15:
Thanks @nordicjm for digging out two places where this limitation is documented!

@JerryDennisEasley I agree that this is not documented in the right place. By the time the user reads the "Beyond Getting Started" guide or the "Creating an application by hand" page then it's too late: the build will have already failed - as you experienced. This note must be moved earlier; in the getting started guide at the point where zephyr code gets downloaded for the first time. Can you please submit a documentation fix that simply moves this note? Thanks in advance.

I renamed and re-opened this issue as a documentation enhancement.