zephyrproject-rtos/zephyr #60407: Add WIFI driver support for  Cypress CYW43XXX

mariopaja, 2023-07-15:
Add WIFI driver support for Cypress CYW43XXX (spi/sdio). It seems like several boards (rpi pico w, arduino giga r1 wifi, infineon PSoC6 ...) are using this wifi module. Additionally, it seems like this module is popular in embedded community.

The module has already bluetooth support and I guess it would be beneficial to also have wifi support.

ifyall, 2023-07-19:
This is in progress in PR 59444! Please review the PR and let us know your thoughts.

mariopaja, 2023-07-19:
> This is in progress in [PR 59444](https://github.com/zephyrproject-rtos/zephyr/pull/59444)! Please review the PR and let us know your thoughts.

On a brief look, it looks great. However, I noticed that at the moment only SDIO is implemented. Will there be also an implementation for SPI?

ifyall, 2023-07-19:
> 

@mariopaja, we don't have that in our short/medium term roadmap. We would be open to some collaboration on that!

PatrickE94, 2023-07-31:
@ifyall I would be open to collaborate on this (adding SPI support). Is anyone else actively working on this?