zephyrproject-rtos/zephyr #60409: Nomination for contributor triage rights for marco85mars20

nixward, 2023-07-16:
# Nomination

I would like to nominate contributor marco85mars20 with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

## GitHub User

Marco Argiolas
[marco85mars20](https://github.com/marco85mars20)
FTP Technologies

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Apr+is%3Aclosed+author%3Amarco85mars20

Thanks

fabiobaltieri, 2023-07-16:
unassigning from myself, usually @nashif or @stephanosio look into these

I thought you were going to self nominate :-)

Can we give triage permissions to @nixward as well while at it? Plenty of code in the repo: https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Apr+is%3Aclosed+author%3Anixward+

nixward, 2023-07-16:
No problem @fabiobaltieri. I believe I already have triage permissions.

nashif, 2023-07-18:
invite sent https://github.com/zephyrproject-rtos/zephyr/invitations

closed 2023-07-18 by nashif