zephyrproject-rtos/zephyr #60423: File change by USB MSC (mass storage class) doesn't reflect to filesystem subsys

pirun, 2023-07-17:
**Describe the bug**
A clear and concise description of what the bug is.

When enable filesystem shell and MSC, try to use filesystem shell to read file created by MSC.
But new file could not be found or read by Zephyr filesystem subsys.
Copy new file into Zephyr MSC, the new file could be read content by any OS connect to it immediately.
Either Ramdisk or Flashdisk has this bug.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
nRF5340DK

**To Reproduce**
MSC Ramdisk with filesystem shell
Steps to reproduce the behavior:
1. cd sample/subsys/usb/mass
2. west build -b nrf5340dk_nrf5340_cpuapp -- -DCONF_FILE=usbd_next_prj.conf -DCONFIG_APP_MSC_STORAGE_RAM=y -DCONFIG_DISK_RAM_VOLUME_SIZE=64 -DCONFIG_FS_LOG_LEVEL_DBG=y -DCONFIG_USE_SEGGER_RTT=y -DCONFIG_SHELL_LOG_BACKEND=n -DCONFIG_USBD_MSC_LOG_LEVEL_DBG=y -DCONFIG_RAMDISK_LOG_LEVEL_DBG=y -DCONFIG_DISK_LOG_LEVEL_DBG=y -DCONFIG_SHELL=y -DCONFIG_HEAP_MEM_POOL_SIZE=2048 -DCONFIG_FILE_SYSTEM_SHELL=y -DCONFIG_DK_LIBRARY=y && west flash --erase
3. Connect USB to Windows or Linux Host PC.
4. Copy any file smaller than 30KB.
5. You can see and read file with Host PC file manager.
6. list file in filesystem with command
`fs ls RAM:/`
7. Could not find any file.

**Expected behavior**
File 1111 is in RAM:/.
File 1111 could be read by filesystem API.

**Impact**
For those applications which want to use Mass storage as bulk data input, then forward bulk data with another transport.
The unsynchronized behavior between MSC and filesystem could make this feature unusable. 

**Logs and console output**
[rtt_log_ramdisk.txt](https://github.com/zephyrproject-rtos/zephyr/files/12065847/rtt_log_ramdisk.txt)
Screesnhot of using command to list file in filesystem.
[image]

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain Zephyr SDK 0.16.0
 - Version used: Nordic Zephyr b37ce40878ec52295afe6350d17712d60eb6639a

pirun, 2023-07-17:
**To Reproduce**
MSC Flash disk with filesystem shell
Steps to reproduce the behavior:

1. cd sample/subsys/usb/mass
2. west build -p -b nrf5340dk_nrf5340_cpuapp -- -DCONF_FILE=usbd_next_prj.conf -DCONFIG_APP_MSC_STORAGE_FLASH_FATFS=y  -DCONFIG_FS_LOG_LEVEL_DBG=y -DCONFIG_USE_SEGGER_RTT=y -DCONFIG_SHELL_LOG_BACKEND=n -DCONFIG_USBD_MSC_LOG_LEVEL_INF=y -DCONFIG_RAMDISK_LOG_LEVEL_DBG=y -DCONFIG_DISK_LOG_LEVEL_DBG=y -DCONFIG_SHELL=y -DCONFIG_HEAP_MEM_POOL_SIZE=2048 -DCONFIG_FILE_SYSTEM_SHELL=y -DCONFIG_HEAP_MEM_POOL_SIZE=2048  && west flash --erase
3. Connect USB to Windows or Linux Host PC.
4. Copy any file smaller than 30KB.
5. You can see and read file with Host PC file manager.
6. list file in filesystem with command
fs ls NAND:/
Could not find any file.
**Expected behavior**
File 1111 is in NAND:/.
File 1111 could be read by filesystem API.
7. After power cycle , 1111 could be found and read by filesysem shell.

Laczen, 2023-07-17:
@pirun, this is not supposed to work. A msc is not a filesystem, if a file is copied from the pc host to a msc the host creates a filesystem on the msc that is readable by the host but this is not a fat filesystem that can be read by zephyr.

pirun, 2023-07-17:
MSC sample enables `FILE_SYSTEM` and `FAT_FILESYSTEM_ELM` https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/usb/mass/Kconfig#L19.
MSC sample mounts filesystem https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/usb/mass/src/main.c#L171.
I though I can use the same mount point to access file by Zephyr.

nordicjm, 2023-07-17:
@jfischer-no can explain to you why this is not a bug

jfischer-no, 2023-07-17:
> * Copy any file smaller than 30KB.
> 
> * You can see and read file with Host PC file manager.
> 
> * list file in filesystem with command
>   fs ls NAND:/
>   Could not find any file.
>   **Expected behavior**
>   File 1111 is in NAND:/.
>   File 1111 could be read by filesystem API.
> 
> * After power cycle , 1111 could be found and read by filesysem shell

Looks like FAT_FILESYSTEM_ELM is caching the root directory or directory sector, unlikely it is the host OS.
Not sure who to blame.

Laczen, 2023-07-17:
> Looks like FAT_FILESYSTEM_ELM is caching the root directory or directory sector, unlikely it is the host OS. Not sure who to blame.

Does this mean a remount should get it working ?

pirun, 2023-07-17:
> Looks like FAT_FILESYSTEM_ELM is caching the root directory or directory sector, unlikely it is the host OS. Not sure who to blame.

I believe someone is caching filesystem. Is there is any API to trigger fs to refresh cache?
Remount might not be an option we need. The data in MSC will be changed rapidly.

Laczen, 2023-07-17:
Remount should be the solution (if this even works). Allowing modifications to the fatfs from two sides (pc host + zephyr) will almost certainly break the fs at some point.

pirun, 2023-07-17:
Modification will be only MSC side. The zephyr is only content reader.

jfischer-no, 2023-07-17:
> Attempting to use the underlying storage simultaneously by USB host and application is recipe for data loss. It doesn't matter that you want the application to only be reading the data - the filesystem data can be in any inconsistent state because the only knowledge about it is in host. MSC is not the protocol to achieve what you want.

Unfortunately, our sample may lead users to believe that it could work. Anyway, let's move it to discussions.