zephyrproject-rtos/zephyr #60430: Endless tracing loop in sysview sample

tmon-nordic, 2023-07-17:
**Describe the bug**
Since 15b8bff8325e7d3832a9f78c95c963571fcd2ec9 the sysview tracing sample fails on nrf52840dk_nrf52840.

**To Reproduce**
Steps to reproduce the behavior:
1. git checkout 15b8bff8325e7d3832a9f78c95c963571fcd2ec9
2. west update; west twister -p nrf52840dk_nrf52840 -s samples/subsys/tracing/sample.tracing.format.sysview --device-testing --verbose --device-serial /dev/ttyACM0
3. Repeat previous step if necessary (looks like the issue triggers only on 2nd and subsequent builds)

**Expected behavior**
Test should pass.

**Impact**
Failing CI.

**Logs and console output**
INFO    - 1/1 nrf52840dk_nrf52840       samples/subsys/tracing/sample.tracing.format.sysview  FAILED Timeout (device 66.304s)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA or Version used: 15b8bff8325e7d3832a9f78c95c963571fcd2ec9

**Additional context**
The endless recursive loop is:

1. `uart_console_init()` contains `device_is_ready()` call results in `SEGGER_SYSVIEW_RecordU32` followed by `z_device_is_ready` followed by `SEGGER_SYSVIEW_RecordEndCallU32`
2.  `SEGGER_SYSVIEW_RecordU32()` calls `RECORD_START(SEGGER_SYSVIEW_INFO_SIZE + SEGGER_SYSVIEW_QUANTA_U32);` which calls `zephyr_rtt_mutex_lock()`
3. `z_impl_k_mutex_lock()` calls `SYS_PORT_TRACING_OBJ_FUNC_ENTER(k_mutex, lock, mutex, timeout);` which results in `SEGGER_SYSVIEW_RecordU32x2()`
4. `SEGGER_SYSVIEW_RecordU32x2` calls `RECORD_START(SEGGER_SYSVIEW_INFO_SIZE + 2 * SEGGER_SYSVIEW_QUANTA_U32);` that calls `zephyr_rtt_mutex_loc()` and the execution goes back to 3

Execution stops when stack goes out of bounds.

jhedberg, 2023-09-26:
@tmon-nordic @carlescufi anyone from Nordic that could take a look at this? (it's unassigned for now)

tmon-nordic, 2023-10-05:
Cannot reproduce it with latest main (38470a42316c4e15077b238fda229eab8a6e0b4f). Reverse git bisect revealed that this issue was fixed in commit 6262304125fbcce27ded4875e974ce0c62edd682

closed 2023-10-05 by tmon-nordic