zephyrproject-rtos/zephyr #60432: ATSAME70Q21 - custom bootloader fails -jumps to application.

Aravindh1910, 2023-07-17:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/60366

I am also working on similar application on  ATSAME70Q21.
I am using mcuboot as bootloader and helloworld as my application. Now I am facing similar kind of issue discussed in above thread. In my case, Bootloader is Jumping to the corresponding address of the application , after that  I am facing memory usage issue,

[image]

The following is (.lst) file (where it gets blocked)

/* do byte-sized copying until finished */

 

    while (n > 0) {
  4437ae:    1e43          subs    r3, r0, #1
  4437b0:    440a          add    r2, r1
{
  4437b2:    b510          push    {r4, lr}
    while (n > 0) {
  4437b4:    4291          cmp    r1, r2
  4437b6:    d100          bne.n    4437ba <memcpy+0xc>
        *(d_byte++) = *(s_byte++);
        n--;
    }

    return d;
}
  4437b8:    bd10          pop    {r4, pc}
        *(d_byte++) = *(s_byte++);
  4437ba:    f811 4b01     ldrb.w    r4, [r1], #1
  4437be:    f803 4f01     strb.w    r4, [r3, #1]!
        n--;
  4437c2:    e7f7          b.n    4437b4 <memcpy+0x6>

can any help m out ?

@pavithvishnu are you be to jump to the application.
@Laczen can you help me out

Laczen, 2023-07-17:
@Aravindh1910, for my understanding: when you compile and flash hello_world without mcuboot everything works as expected ?

Aravindh1910, 2023-07-17:
@Laczen  Yes, If I compile hello_world and flashed it without mcuboot it works!

Aravindh1910, 2023-07-18:
@Laczen  and @pavithvishnu,
I have a findings below,
1. able to compile hello_world and Flash it without mcuboot it works.
2. able to compile and flash hello_world with mcuboot (without USB enabled) it works!.
3. able to compile and flash hello_world with mcuboot (with USB enabled) it doesn't worked!.

To summarize, if I enabled USB in Configuration, jump from mcuboot to Application doesn't work.
But, For my requirement I need to enable USB. USB configured as below.

filename : board_defconfig
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_HID=y
CONFIG_ENABLE_HID_INT_OUT_EP=y
CONFIG_HID_INTERRUPT_EP_MPS=256

May I know How to disable this before jump to application. 
(FYI, usb_disable API also not works)

Laczen, 2023-07-18:
@jfischer-no, could there be something missing in mcuboot that causes this malfunction ?

pavithvishnu, 2023-07-18:
@Aravindh1910  you are right, I have also tried with the same configuration  which you have mentioned above. I'm also facing the same issue. if usb stack is enable, jump to application is not working.