zephyrproject-rtos/zephyr #60442: mcuboot: shell: Output does not look valid

nordicjm, 2023-07-17:
**Describe the bug**
It seems that the `mcuboot` shell output produces bogus data which does not represent the actual state of the mcuboot images, for example, with 2 images where the main image is confirmed and active:
```
mcuboot
swap type: none
confirmed: 1

primary area (1):
  version: 0.0.0+0
  image size: 119200
  magic: good
  swap type: revert
  copy done: set
  image ok: set

secondary area (2):
  version: 0.0.0+0
  image size: 50652
  magic: unset
  swap type: none
  copy done: unset
  image ok: unset
```
This is from a device with swap using move mode, the output does not make sense, upon reboot, the image would not be reverted.

**To Reproduce**
Build `samples/mgmt/mcumgr/smp_svr` with `overlay-bt.conf` overlay then issue `mcuboot` command from shell

**Expected behavior**
Output should be an easy to understand and realistic representation of the images on the device

**Impact**
Annoyance, confusion

**Environment (please complete the following information):**
 - Commit SHA or Version used: ea2aac6e5131bf916abd1e37334e94cc3481e1a8

jhedberg, 2023-09-26:
@d3zd3z any comment on this one?

closed 2023-12-11 by github-actions[bot]

aunsbjerg, 2024-08-14:
@nordicjm, was this ever solved? I believe I am seeing the same issue in a Zephyr 3.5 application.

nordicjm, 2024-08-14:
> @nordicjm, was this ever solved? I believe I am seeing the same issue in a Zephyr 3.5 application.

Not sure but try on 3.6 or 3.7

aunsbjerg, 2024-08-14:
> Not sure but try on 3.6 or 3.7

I just tried this, but I have the same issue.

This is the log output after performing an upgrade and then confirming the new image using `mcuboot confirm`:

```log
uart:~$ mcuboot 
swap type: none
confirmed: 1

primary area (1):
  version: 1.0.0+1
  image size: 628652
  magic: good
  swap type: test
  copy done: set
  image ok: set

secondary area (2):
  version: 1.0.0+0
  image size: 628652
  magic: unset
  swap type: none
  copy done: unset
  image ok: unset
```

It seems strange that the primary area swap type is `test` when the overall swap type is `none` and the primary image is confirmed. 

I am not using mcumgr but a custom "manager" built on top of the public zephyr IMG_MANAGER subsystem. It is entirely possible I've messed up somewhere. I'll see if I can reproduce using vanilla mcumgr and the smp sample tomorrow in the office.

aunsbjerg, 2024-08-15:
Just did a test on a frdm-k64f board on the latest main (17f11332a92) using the smp_srv sample. I see the same inconsistency in the output:

```
uart:~$ mcuboot 
swap type: none
confirmed: 1

primary area (1):
  version: 1.2.3+5
  image size: 69688
  magic: good
  swap type: test
  copy done: set
  image ok: set

secondary area (2):
  version: 0.0.0+0
  image size: 69688
  magic: unset
  swap type: none
  copy done: unset
  image ok: unset
```