zephyrproject-rtos/zephyr #60449: STM32H7: Flash error flags are set spontaneously

mjaun, 2023-07-17:
**Describe the bug**
Summary: The RDSERR (secure error) and/or RDPERR (read protection error) flags of the flash status register are set from time to time.

We noticed this issue in our project because the bootloader API `mcuboot_swap_type()` would return something unexpected. Under the hood this function uses `flash_area_read()` which is provided by the STM32 flash driver. This one in turn checks the above mentioned error flags after execution which leads to a failure.

**To Reproduce**
From our project we extracted a sample to reproduce the issue. After quite some playing around we now have a minimal sample which does actually nothing except very short sleeps in the main thread. Before running the loop with sleeps the sample registers an interrupt to get notified if the mentioned flash error flags are set.

It can be found here: https://github.com/mjaun/zephyr/tree/flash_error/samples/flash_error

It reproduces the issue at least on the `nucleo_h743zi` board and also on the `nucleo_h743zi_minimal` board which is provided on the same branch and disable almost all drivers.

**Expected behavior**
1. Most importantly the flash errors should not occur
2. Additionally it might make sense to clear the flags before read/write operations

**Impact**
We have some workarounds in place, because the bootloader API might not work properly.

**Logs and console output**
Output from the mentioned sample:

```
*** Booting Zephyr OS build abae9b562b84 ***
ISR triggered: 5581 ms
FLASH_SR1: 0x01000000
FLASH_SR2: 0x00000000
ISR triggered: 50320 ms
FLASH_SR1: 0x01000000
FLASH_SR2: 0x00000000
ISR triggered: 95060 ms
FLASH_SR1: 0x01000000
FLASH_SR2: 0x00000000
ISR triggered: 139799 ms
FLASH_SR1: 0x01000000
FLASH_SR2: 0x00000000
ISR triggered: 184538 ms
FLASH_SR1: 0x01000000
FLASH_SR2: 0x00000000
```

The timings are always the same also across multiple devices. Note that the time between each occurrence is always almost the same. I observed the timings changing by modifying SYSCLK or also changing the application code. Some changes make the issue disappear (e.g. enabling CONFIG_LOG or CONFIG_DEBUG).

Currently I don't have a clue why these errors occur at all. Breaking in the ISR shows that the application is in idle thread.

**Environment (please complete the following information):**
- Zephyr v3.3.0 but also on main (8e40304)
- STM32H743ZI

mjaun, 2023-07-17:
Just noted the comment in [this discussion](https://community.st.com/t5/stm32-mcu-products/spurious-rdperr-and-rdserr-when-all-protection-and-security/m-p/279852), that such behavior can be triggered by speculative access to the system memory.

Indeed adding the following MPU region seems to resolve the issue at least in our sample:

```
MPU_REGION_ENTRY("SYSTEM", 0x1FF00000, { (STRONGLY_ORDERED_SHAREABLE | REGION_1M | NO_ACCESS_Msk) })
```

Beforehand I already tried setting up a background region as suggested by ST in [this tutorial](https://www.youtube.com/watch?v=6IUfxSAFhlw). For unknown reasons they don't protect the system memory in this example. Anyway the solution to the issue here could be to define a reasonable background MPU region.

erwango, 2023-07-18:
Thanks for raising this.
Can you check https://github.com/zephyrproject-rtos/zephyr/pull/57467 to see if it can fix your issue ?

mjaun, 2023-07-18:
> Thanks for raising this. Can you check #57467 to see if it can fix your issue ?

The commits from #57467 are contained in the provided sample. They don't seem to fix the issue. As mentioned it seems to be the system memory region which causes the issue not external memory regions.

FRASTM, 2023-07-19:
also `CONFIG_DCACHE=n` makes the pb disappear

FRASTM, 2023-08-08:
> We've also been observing this behaviour on our STM32H7343 when communicating with the bootloader API. Would be great to have this fixed.

The PR https://github.com/zephyrproject-rtos/zephyr/pull/60765  fixes for all the stm32H7, isn't it. 
Now Expecting reviewers/approvers

closed 2023-08-30 by carlescufi