zephyrproject-rtos/zephyr #60502: Align west --test-item with twister workflow

PerMac, 2023-07-18:
**Is your enhancement proposal related to a problem? Please describe.**
`Sample/testcase.yamls` contain build configuration for twister tests. It is required that a user could build a given configuration without twister but with west. A west-command `--test-item` was added to serve this purpose. However, this method doesn't align with twister's behavior and it is not warranted that it will work, or worse: a user will get a different build without realizing it. In recent month several issues were reported regarding `--test-item`:
#52614, #56248, #59761, #60297
 
**Describe the solution you'd like**
IMO a single source of truth should be used when building via twister and west. Since west should be usable without twister, and twister already requires west, I think the source can be in west. When building, twister will directly call west and no scripting for building will be present in twister. With the direct call (through cli) we will assure that calling this command by user (without using twister) will give the same result.

**Describe alternatives you've considered**
- A west module is loaded into twister and the building is called with available west methods. IMO it can be less clear for a user to reproduce 
- we do the aligning in west and keep the code for building both in twister and in west. However this will be problematic to keep in sync as it was already seen in reported issues. We then would have to provide tests verifying if the same output was achieved with west and with twister, which can end up in a cat and mouse game.

**Additional context**
twister does several steps during the build stage, which are not mirrored in west. Some missing pieces are easier to find. E.g. west ignores the common section of yamls. In addition twister compiles extra args and konfigs from multiple places into a single list https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/config_parser.py#L205 in a certain order. This proces is not reproduced in west and can cause harder to catch mismatches in build products.

PerMac, 2023-07-18:
cc @nordicjm @nashif @hakehuang

hakehuang, 2023-07-18:
@PerMac , we can use  `west twister ...` , i remember this has been supported by google team @aborisovich

PerMac, 2023-07-18:
`west twister` is only a west wrapper over twister, i.e. west directly calling twister. However, `west --test-item` doesn't use twister. It suppose to be a mean to build/run with configurations described in .yaml files, which are native to twister.

hakehuang, 2023-07-18:
> `west twister` is only a west wrapper over twister, i.e. west directly calling twister. However, `west --test-item` doesn't use twister. It suppose to be a mean to build/run with configurations described in .yaml files, which are native to twister.

in this case, you will have to re-implement those already in the testcase.yaml right?

nashif, 2023-07-18:
`west --test-item` was added as a convinience option. The yaml file is very twister specific and we should continue to maintain building and parsing this file in twister directly. Single source of truth is nice, but splitting twister functionality across different tools is going to be counter productive and I do not see any significant benefit. 

So, I am -1 on this proposal.

PerMac, 2023-07-18:
I see your point. But in such form I don't see `west --test-item` as a convenience. Rather a dangerous cumber. Cumber, since we will constantly have to maintain it and try to keep close to twister workflow, and dangerous since (as linked issues showed and since there are still some not solved issues, like handling common section) can produce different results and give false impression that something is fine/wrong.

What is the alternative then? Leave the status quo with not aligned functionality?

nashif, 2023-07-18:
> What is the alternative then? Leave the status quo with not aligned functionality?

The alternative is to just focus on what west needs and implement it directly in west like we have right now. There is no need to move all of this logic into west from twister and create more problems along the way.

PerMac, 2023-07-20:
@nashif 
>  The yaml file is very twister specific and we should continue to maintain building and parsing this file in twister directly.

What about the other way then? I.e. `west --test-item` using twister library?

nashif, 2023-07-20:
> What about the other way then? I.e. `west --test-item` using twister library?

that makes more sense, but then it is about whether west wants to depend on another library.

nashif, 2023-07-20:
``--test-item`` option is very useful and can't be replaced with just running twister. A developer can build any scenario directly and does not need to worry about twister internals, i.e. I could debug, do incremental changes and rebuild etc. etc., using the same development workflow for everything.

The only thing this option needs to worry about are options related to building, nothing else. So filtering and other twister options are out of the question. The option is there to just build the scenario in a similar way as with twister. The option itself has been working great for majority of the tests, recent additions such as sysbuild however or more use of common for build options might have caused issues, but for the most part it works.

nashif, 2024-02-09:
I consider this done, no need to keep this open.

closed 2024-02-09 by nashif