zephyrproject-rtos/zephyr #60522: scripts: pylib: twister: twisterlib: environment: Unguarded member access in TwisterEnv __init__()

LukaszMrugala, 2023-07-18:
Initialisation of `TwisterEnv` is careful as to check whether `options` parameter is not `None` before trying to access it. 

https://github.com/zephyrproject-rtos/zephyr/blob/b8f7c81dd3930ae1dde60dcf1daba32d03176816/scripts/pylib/twister/twisterlib/environment.py#L800-L820

However, at the very end of `__init__`, we've got two accesses that fail to check for whether any `options` exist.

https://github.com/zephyrproject-rtos/zephyr/blob/b8f7c81dd3930ae1dde60dcf1daba32d03176816/scripts/pylib/twister/twisterlib/environment.py#L824-L826

In my opinion, that should be guarded against.

nashif, 2023-07-18:
@LukaszMrugala  can you please subit a PR fixing this?

LukaszMrugala, 2023-07-18:
> @LukaszMrugala can you please subit a PR fixing this?

Done in PR #60526

PR #60526 (2023-07-20): scripts: pylib: twister: twisterlib: environment: TwisterEnv fixes
Fixes #60522 by guarding member access against None parameters.
Shortens the init by using the if...else assignment.
Fixes two typos.
closed 2023-07-20 by nashif