zephyrproject-rtos/zephyr #60546: modem: hl7800: no response from AT commands after wakeup from hibernate

bsphere, 2023-07-18:
I'm using the pinnacle_100_dvk target,
with a simple example that tries to perform a network operation, while the modem is in hibernate (PSM dormant), it wakes up but looks like no response is received for the AT+KUDPSND command resulting in dns timeout.

I tried to diagnose the problem and found out that if i disable CONFIG_PM_DEVICE so it won't shut down UART everything is working as expected. it also works occasionally so I suspect some race condition..

furthermore, if I enable CONFIG_ASSERT, there's a fatal error when the modem wakes up..

example log -
```
Waiting for LTE to be ready...
[00:00:09.094,848] <dbg> modem_hl7800: hl7800_rx: HANDLE +KUDP_IND:  (len:3)
[00:00:09.094,879] <dbg> modem_hl7800: on_cmd_sock_ind: +KUDP_IND ID: 1
Network monitor event 2
[00:00:10.095,001] <dbg> modem_hl7800: allow_sleep_work_callback: Allow sleep
[00:00:10.095,062] <inf> modem_hl7800: Sleep State: HIBERNATE
Resolving host address postman-echo.com ...
[00:00:39.099,670] <dbg> modem_hl7800: allow_sleep: Keep awake
[00:00:39.099,731] <dbg> modem_hl7800: wakeup_hl7800: Waiting to wakeup
[00:00:39.106,811] <inf> modem_hl7800: Sleep State: AWAKE
[00:00:39.106,994] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KUDPSND=1,"82.102.139.10",53,34]
Unable to resolve address [-11], quitting
[00:01:41.107,208] <err> modem_hl7800: Err waiting for CONNECT (-11)
App sleeping for 120 seconds
[00:01:42.107,360] <dbg> modem_hl7800: allow_sleep_work_callback: Allow sleep
[00:01:42.107,391] <inf> modem_hl7800: Sleep State: HIBERNATE
```

I'm using the zephyr revision in NCS v2.4.0, i double checked and made sure the hl7800 driver is similar to the one in zephyr's main branch.  

**Impact**
showstopper

**Environment (please complete the following information):**
- OS: macOS
- Toolchain: NCS v2.4.0

**To Reproduce**
a simple code is enough to show this behavior:
src/main.c:
```
#include <zephyr/kernel.h>

#define MODULE main

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(MODULE);

#include <zephyr/drivers/modem/hl7800.h>

void ntp_cb(struct k_work *work) {
    int r, s;
    mdm_hl7800_get_signal_quality(&r, &s);
}

K_WORK_DEFINE(ntp_work, ntp_cb);

int main(void) {
   k_sleep(K_SECONDS(45));
   k_work_submit(&ntp_work);
   return 0; 
}
``` 

prj.conf:
```
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_HEAP_MEM_POOL_SIZE=4096
CONFIG_THREAD_NAME=y
CONFIG_THREAD_MONITOR=y
CONFIG_INIT_STACKS=y
CONFIG_LOG=y
CONFIG_LOG_BUFFER_SIZE=10240
CONFIG_MODEM=y
CONFIG_MODEM_HL7800=y
CONFIG_MODEM_HL7800_FW_UPDATE=y
CONFIG_MODEM_LOG_LEVEL_DBG=y
CONFIG_MODEM_HL7800_RAT_M1=y
CONFIG_MODEM_HL7800_CONFIGURE_BANDS=n
CONFIG_MODEM_HL7800_LOW_POWER_MODE=y
CONFIG_MODEM_HL7800_PSM=y
CONFIG_MODEM_HL7800_PSM_PERIODIC_TAU="10100010" # 2 minutes
CONFIG_MODEM_HL7800_PSM_ACTIVE_TIME="00001111" # 30 seconds
CONFIG_MODEM_HL7800_SET_APN_NAME_ON_STARTUP=y
CONFIG_MODEM_HL7800_APN_NAME="uinternet"

# Networking
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_SOCKETS=y
CONFIG_NET_TCP=y

# Enable the DNS resolver
CONFIG_DNS_RESOLVER=y
CONFIG_NET_SOCKETS_DNS_TIMEOUT=20000

#CONFIG_LOG_BACKEND_UART=n
CONFIG_LOG_BACKEND_RTT=y

CONFIG_UART_CONSOLE=n

CONFIG_NEWLIB_LIBC=y

CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
```

log output:   (AT+KCELLMEAS times out after the wakeup)
```
00> [00:00:07.549,713] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KBND?]
00> [00:00:07.555,999] <dbg> modem_hl7800: hl7800_rx: HANDLE +KBND:  (len:22)
00> [00:00:07.556,518] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
00> [00:00:07.556,579] <dbg> modem_hl7800: iface_status_work_cb: Network state updated
00> [00:00:07.556,640] <dbg> modem_hl7800: allow_sleep: Keep awake
00> [00:00:07.556,671] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KCELLMEAS=0]
00> [00:00:07.566,497] <dbg> modem_hl7800: hl7800_rx: HANDLE +KCELLMEAS:  (len:26)
00> [00:00:07.566,497] <inf> modem_hl7800: RSSI (RSRP): -85 SINR: 23
00> [00:00:12.567,077] <dbg> modem_hl7800: allow_sleep_work_callback: Allow sleep
00> [00:00:12.567,138] <inf> modem_hl7800: Sleep State: HIBERNATE
00> [00:00:49.816,864] <dbg> modem_hl7800: allow_sleep: Keep awake
00> [00:00:49.816,925] <dbg> modem_hl7800: wakeup_hl7800: Waiting to wakeup
00> [00:00:49.818,786] <inf> modem_hl7800: Sleep State: AWAKE
00> [00:00:49.818,817] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KCELLMEAS=0]
00> [00:00:55.818,969] <err> modem_hl7800: AT+KCELLMEAS ret:-116
00> [00:01:00.819,030] <dbg> modem_hl7800: allow_sleep_work_callback: Allow sleep
00> [00:01:00.819,061] <inf> modem_hl7800: Sleep State: HIBERNATE
```

bsphere, 2023-07-19:
ok, i will and update.
please note that it happens for any wakeup initiated by AT command and not only by networking, see the few lines of code i posted in the issue..

if the wakeup is initiated by a PSM wakeup there is no problem. so the immediate workaround is to interact with the modem only from the wake event handler.

bsphere, 2023-07-19:
update - it also happens with the example built from the manifest. I changed `SLEEP_TIME_SECONDS` to 60 seconds so it'll do the http_get before PSM wakeup:

```
Network monitor event 2
Resolving host address postman-echo.com ...
[00:00:07.944,458] <dbg> modem_hl7800: hl7800_rx: HANDLE +KUDP_IND:  (len:3)
[00:00:07.944,488] <dbg> modem_hl7800: on_cmd_sock_ind: +KUDP_IND ID: 1
[00:00:07.951,324] <dbg> modem_hl7800: allow_sleep: Keep awake
[00:00:07.951,477] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KUDPSND=1,"82.102.139.10",53,34]
[00:00:07.968,078] <dbg> modem_hl7800: hl7800_rx: HANDLE CONNECT (len:0)
[00:00:07.971,221] <dbg> modem_hl7800: send_data: Sent 34 bytes
[00:00:07.980,285] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:08.197,662]Got host address!
Connecting to postman-echo.com
 <dbg> modem_hl7800: hl7800_rx: HANDLE +KUDP_DATA:  (len:5)
[00:00:08.197,662] <dbg> modem_hl7800: start_socket_rx: Start socket RX ID:1 size:130
[00:00:08.197,753] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KUDPRCV=1,130]
[00:00:08.209,991] <dbg> modem_hl7800: hl7800_rx: HANDLE CONNECT (len:0)
[00:00:08.210,021] <dbg> modem_hl7800: sock_read: Socket 1 RX 130 bytes
[00:00:08.210,021] <dbg> modem_hl7800: wait_for_modem_data: cur:0, exp:1
[00:00:08.260,192] <dbg> modem_hl7800: sock_read: Processing RX, buf len: 185
[00:00:08.260,986] <dbg> modem_hl7800: sock_read: Got all data, get EOF and OK (buf len:55)
[00:00:08.261,047] <dbg> modem_hl7800: sock_read: Sock 1 RX done
[00:00:08.261,138] <dbg> modem_hl7800: hl7800_rx: UNHANDLED RX
                                       2b 4b 55 44 50 5f 52 43  56 3a 20 22 38 32 2e 31 |+KUDP_RC V: "82.1
                                       30 32 2e 31 33 39 2e 31  30 22 2c 35 33          |02.139.1 0",53
[00:00:08.261,169] <dbg> modem_hl7800: sockreadrecv_cb_work: Sock 1 RX CB (size: 158)
[00:00:08.271,728] <dbg> modem_hl7800: allow_sleep: Keep awake
[00:00:08.271,911] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KTCPCFG=1,0,"44.212.248.94",80,,,,0,,1]
[00:00:08.309,173] <dbg> modem_hl7800: hl7800_rx: HANDLE +KTCPCFG:  (len:1)
[00:00:08.309,173] <dbg> modem_hl7800: on_cmd_sockcreate: +KTCPCFG: 2
[00:00:08.309,204] <dbg> modem_hl7800: on_cmd_sockcreate: look up new socket by creation id
[00:00:08.309,661] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:08.309,783] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KTCPCNX=2]
[00:00:08.332,672] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
Close HTTP connection after request

Sending GET request...
Response:

[00:00:08.557,495] <dbg> modem_hl7800: hl7800_rx: HANDLE +KTCP_IND:  (len:3)
[00:00:08.557,525] <dbg> modem_hl7800: on_cmd_sock_ind: +KTCP_IND ID: 2
[00:00:08.563,537] <dbg> modem_hl7800: allow_sleep: Keep awake
[00:00:08.563,629] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KTCPSND=2,61]
[00:00:08.576,019] <dbg> modem_hl7800: hl7800_rx: HANDLE CONNECT (len:0)
[00:00:08.581,695] <dbg> modem_hl7800: send_data: Sent 61 bytes
[00:00:08.890,289] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:08.892,181] <dbg> modem_hl7800: hl7800_rx: HANDLE +KTCP_DATA:  (len:5)
[00:00:08.892,211] <dbg> modem_hl7800: start_socket_rx: Start socket RX ID:2 size:370
[00:00:08.892,272] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KTCPRCV=2,370]
[00:00:08.904,998] <dbg> modem_hl7800: hl7800_rx: HANDLE CONNECT (len:0)
[00:00:08.905,029] <dbg> modem_hl7800: sock_read: Socket 2 RX 370 bytes
[00:00:08.905,029] <dbg> modem_hl7800: wait_for_modem_data: cur:0, exp:1
[00:00:08.955,261] <dbg> modem_hl7800: sock_read: Processing RX, buf len: 431
[00:00:08.957,397] <dbg> modem_hl7800: sock_read: Got all data, get EOF and OK (buf len:61)
[00:00:08.957,458] <dbg> modem_hl7800: sock_read: Sock 2 RX done
[00:00:08.957,550] <dbg> modem_hl7800: hl7800_rx: HANDLE +KTCP_DATA:  (len:3)
[00:00:08.957,580] <dbg> modem_hl7800: start_socket_rx: Start socket RX ID:2 size:1
[00:00:08.957,672] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KTCPRCV=2,1]
[00:00:08.959,075] <dbg> modem_hl7800: hl7800_rx: HANDLE +KTCP_NOTIF:  (len:3)
[00:00:08.959,106] <dbg> modem_hl7800: on_cmd_sock_notif: +K**P_NOTIF: 2,4
[00:00:08.959,167] <dbg> modem_hl7800: sockreadrecv_cb_work: Sock 2 RX CB (size: 410)
[00:00:08.969,970] <dbg> modem_hl7800: hl7800_rx: HANDLE CONNECT (len:0)
[00:00:08.970,001] <dbg> modem_hl7800: sock_read: Socket 2 RX 1 bytes
[00:00:08.970,031] <dbg> modem_hl7800: wait_for_modem_data: cur:0, exp:1
[00:00:09.020,111] <dbg> modem_hl7800: sock_read: Processing RX, buf len: 23
[00:00:09.020,202] <dbg> modem_hl7800: sock_read: Got all data, get EOF and OK (buf len:22)
[00:00:09.020,263] <dbg> modem_hl7800: sock_read: Sock 2 RX done
[00:00:09.020,294] <dbg> modem_hl7800: sockreadrecv_cb_work: Sock 2 RX CB (size: 41)
[00:00:09.120,086] <dbg> modem_hl7800: sock_notif_cb_work: Sock 2 trigger NULL packet
[00:00:09.120,117] <dbg> modem_hl7800: sockreadrecv_cb_work: Sock 2 RX CB (size: 0)
HTTP/1.1 302 Found
Date: Wed, 19 Jul 2023 17:10:59 GMT
Content-Type: text/plain; charset=utf-8
Content-Length: 51
Connection: close
Location: https://docs.postman-echo.com
Vary: Accept
set-cookie: sails.sid=s%3Agy4XJTal0I2vAUKkqnxu-2fPC14IfL2V.JRACOv6Mukqty1%2Fc%2B5YEbwynxfnn4rkuRpMXhQaY4p4; Path=/; HttpOnly

Found. Redirecting to https://docs.postman-echo.com
Close connection.
App sleeping for 60 seconds
Resolving host address postman-echo.com ...
[00:01:09.943,450] <dbg> modem_hl7800: allow_sleep: Keep awake
[00:01:09.943,481] <dbg> modem_hl7800: wakeup_hl7800: Waiting to wakeup
[00:01:09.945,404] <inf> modem_hl7800: Sleep State: AWAKE
[00:01:09.945,556] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KUDPSND=1,"82.102.139.10",53,34]
Unable to resolve address [-11], quitting
App sleeping for 60 seconds
```

rerickson1, 2023-07-19:
Thanks for the feedback. What HL7800 firmware version is being used?

bsphere, 2023-07-19:
4.6.9.4

rerickson1, 2023-08-31:
Give this a try: https://github.com/LairdCP/Pinnacle_100_Sample_Applications_Manifest/pull/3

closed 2024-01-15 by github-actions[bot]