zephyrproject-rtos/zephyr #60564: Clarify usage of binary constants in the code base

fabiobaltieri, 2023-07-19:
**Is your enhancement proposal related to a problem? Please describe.**
Documentation clarification.

**Describe the solution you'd like**
There's been some discussion in PR on whether is ok or not to use binary constant in the code (`0b` prefixed constants), where it's been noted that they are not part of the standard but a common extension: https://github.com/zephyrproject-rtos/zephyr/pull/56234#discussion_r1267751157. Should we clarify this as part of the coding style? While these appear to be in use on core code already and made their way in the Linux code base as well, it'd be nice to have an explicit guideline on their usage.

fabiobaltieri, 2023-07-19:
cc @stephanosio @nashif @keith-packard @cfriedt (C library maintainers) @galak (think you did some work on external compilers) any opinion?

jfischer-no, 2023-07-19:
It is not part of C99 or C11, but likely C23. I cannot find it in Linux Kernel code base except in comments.

thedjnK, 2023-07-19:
It is a GCC specific extension, and should be banned as per https://github.com/zephyrproject-rtos/zephyr/issues/52226

kartben, 2023-07-19:
> [...] While these appear to be in use on core code already [...]

More than just core, for better or worse, as it can also be found in a handful of drivers as well (ads114s0x, BD8LB600FS, max31865, ...)

jfischer-no, 2023-07-19:
> > [...] While these appear to be in use on core code already [...]
> 
> More than just core, for better or worse, as it can also be found in a handful of drivers as well (ads114s0x, BD8LB600FS, max31865, ...)

ads114s0m, BD8LB600FS because it is the same author, and none said no.

cfriedt, 2023-07-19:
> It is not part of C99 or C11, but likely C23. I cannot find it in Linux Kernel code base except in comments. 

https://en.m.wikipedia.org/wiki/C23_(C_standard_revision)

So yeah, part of C23.

I'm in no rush to #ifdef on a whole bunch of initializers and macros.

If this is breaking something, there should be an actual bug filed.

If there is no bug filed, I would suggest to close the issue.

If there is a bug filed, we should probably add something in the coding style, add a linter rule, and then do a tree-wide change.

There is a process for that.

fabiobaltieri, 2023-07-19:
> It is not part of C99 or C11, but likely C23. I cannot find it in Linux Kernel code base except in comments.

There's very few in 5.15 https://lxr.missinglinkelectronics.com/linux/drivers/gpu/drm/amd/amdgpu/gfx_v9_4_2.c#L555, though a lot more have been added in the latest merge window, wondering if the rule changed recently.

keith-packard, 2023-07-19:
I'd say that when importing existing data or code it would be best to leave things unmodified as it's easier to validate things that way. And, if it were purely a non-standard extension and not likely to be part of the next language revision, we might want to push back a bit harder.

I'd suggest allowing binary constants where they improve clarity and reduce transcription errors, but making it clear that existing standard C forms should be preferred in general.

I suspect these values will have the greatest application closer to the metal, so perhaps the should only be allowed in drivers/ and arch/ ?

keith-zephyr, 2023-07-19:
> I'd say that when importing existing data or code it would be best to leave things unmodified as it's easier to validate things that way. And, if it were purely a non-standard extension and not [...]

+1 on this suggestion.

> 
> I suspect these values will have the greatest application closer to the metal, so perhaps they should only be allowed in drivers/ and arch/ ?

We should avoid special rules for different parts of the tree.

nordicjm, 2023-07-20:
I vote for removal, they are not required and we have multiple toolchains supported in-tree: gcc, armclang, clang, arc tools (whatever they are called) plus ongoing support for IAR. It might be part of a _forthcoming_ C standard, but we are not at that standard and should not be using features that are not widely supported.

jfischer-no, 2023-07-20:
> I vote for removal, they are not required and we have multiple toolchains supported in-tree: gcc, armclang, clang, arc tools (whatever they are called) plus ongoing support for IAR. It might be part [...]

They probably all support it.
https://www.iar.com/knowledge/support/technical-notes/compiler/any-binary-notation-in-c-source/

>The linux kernel is different in that it only supports being compiled with GCC, so they can use whatever GCC-specific extensions they want

https://docs.kernel.org/kbuild/llvm.html

But the thing is, binary constants are not really needed, some parts would be even more readable with decimal representation. 
IIRC compiler-specific extensions must be justified from a safety perspective, FYI @simhein

benediktibk, 2023-07-20:
I'm the author that triggered this, The reason for using the binary literals was that I could use the exact same representation in the driver implementation of some register values as they are also depicted in the data sheet of the hardware. Which in my opinion makes it easier to understand the driver, in case of for instance of some bug fixing.

But I am usually also against non standard extensions. I just wasn't aware that these literals are not part of the standard. In this situation it is a little bit in between, as it will probably be part of the standard pretty soon.

If it should be banned I strongly support the idea of @cfriedt of having a hard check in the continuous integration. Every manual check is more labor and effort which is first of all better spent on something actually useful and secondly bound to fail at one point.

cfriedt, 2023-07-20:
> The linux kernel is different in that it only supports being compiled with GCC, so they can use whatever GCC-specific extensions they want

We build Linux with LLVM regularly :-)

simhein, 2023-07-25:
> But the thing is, binary constants are not really needed, some parts would be even more readable with decimal representation. IIRC compiler-specific extensions must be justified from a safety perspective, FYI @simhein

@jfischer-no Thanks for flagging this. Yes you are correct in this case compiler-specific extensions must be justified or at least documented in the guidelines how and where they can be used. 

> We should avoid special rules for different parts of the tree.
+1

cfriedt, 2023-07-25:
It's also important to note that binary constants have been a part of the C++ standard for nearly a decade (since C++14). Kind of unfortunate that the C standard did not adopt this essentially zero-cost feature until C23 (whatever the next release is?)

I'm sure binary constanta are a part of other languages too (e.g. python).

That implies we would need to have different conventions / considerations / checks on a per-language basis.

jfischer-no, 2023-07-25:
> It's also important to note that binary constants have been a part of the C++ standard for nearly a decade (since C++14). Kind of unfortunate that the C standard did not adopt this essentially zero-cost feature until C23 (whatever the next release is?)
> 
> I'm sure binary constants are a part of other languages too (e.g. python).
> 
> That implies we would need to have different conventions / considerations / checks on a per-language basis.

What "a per-language basis"? Project language is clearly described here 
https://docs.zephyrproject.org/latest/develop/languages/c/index.html#language-standards.
Wondering when C++ people will discover again that result of equal-to operator has type int? :scream:

cfriedt, 2023-07-25:
> What "a per-language basis"? Project language is clearly described here 
> https://docs.zephyrproject.org/latest/develop/languages/c/index.html#language-standards.

`primarily written in C` 👈

Zephyr does support C++, so I would assume C++ code can be written according to the standard too.

`Zephyr does not target a specific version of the C standards` 👈

They will be a part of the next C standard 🤷‍♂️

I'm not in love with binary constants, am not even sure if I've used them in Zephyr, but they can be more descriptive in some cases.

What is the ask here? To explicitly prevent binary constants in core Zephyr?

I guess I'm back to my original questions of
* do binary constants break anything?
* if so, can someone submit a bug report?

fabiobaltieri, 2023-07-25:
> What is the ask here? To explicitly prevent binary constants in core Zephyr?
> 
> I guess I'm back to my original questions of
> 
>     * do binary constants break anything?
> 
>     * if so, can someone submit a bug report?

The practical question is if we need a guideline or CI check for/against them. From what I hear the consensus is no CI checks, just a guideline saying that we prefer to avoid them, but they are allowed if they obviously improve clarity. Which in practically means: if all reviewers are happy with it ok, otherwise don't use them.

Does it sound reasonable? If it turns out they break something we could ban but there does not seem to be a case for starting to convert the few we have in the codebase now.

nordicjm, 2023-07-26:
> Does it sound reasonable? If it turns out they break something we could ban but there does not seem to be a case for starting to convert the few we have in the codebase now.

Has someone tested it against **every** supported compiler in the code base, because I doubt someone has tested it on the 2 very niche drivers in the tree that use them.

I am still of the mind they should be banned from the codebase entirely.

fabiobaltieri, 2023-07-26:
> Has someone tested it against **every** supported compiler in the code base, because I doubt someone has tested it on the 2 very niche drivers in the tree that use them.

XCC and MWDT? https://docs.zephyrproject.org/latest/develop/toolchains/index.html

cfriedt, 2023-07-26:
Do we support IAR yet? (/me shudders)

fabiobaltieri, 2023-07-26:
> Do we support IAR yet? (/me shudders)

Funny how IAR is mentioned in a bunch of board documentation pages. It's obviously just copy pasted from the generic board brochure, but it should probably not be there if the page is part of the Zephyr documentation and you can't build Zephyr with it.

MaureenHelm, 2023-07-27:
dev-review: general agreement to prohibit use in C code and recommend updating the coding guidelines