zephyrproject-rtos/zephyr #60613: tests: posix: common: portability.posix.common timeout due to the hanging in test_pthread_execution

povergoing, 2023-07-20:
**Describe the bug**
`portability.posix.common` sometimes fails with a timeout, and the program hangs in `test_pthread_execution`.

After some investigations, I found there might be a racy about writing the `mu` variables. Multiple threads race to take the mutex, they can call `m = to_posix_mutex(mu);` at the same time. Going into the `to_posix_mutex`, it might allocate several times for `*mu`, because there is no exclusive lock to protect the `*mu`.
https://github.com/zephyrproject-rtos/zephyr/blob/13a6c454529e64cdb9f69cab0f0a9fb8c539b192/lib/posix/mutex.c#L109
Specifically,
1. `*mu` init with `PTHREAD_MUTEX_INITIALIZER`
2. Thread#1 and Thread#2 both pass `if (*mu != PTHREAD_MUTEX_INITIALIZER) `
3. Thread#1 alloc bit == 0, and Thread#2 alloc bit == 1
4. `*mu` is assigned twice, the last one takes effect, but `bit` and `m` will remain different between two threads.
5. Thus, two threads actually got two `k_mutex`s

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b fvp_baser_aemv8r_smp -t run -T zephyr/tests/posix/common/portability.posix.common
6. might need to test many times

**Expected behavior**
Pass

**Impact**
annoyance

**Logs and console output**
```
terminal_0: Listening for serial connection on port 5000
terminal_1: Listening for serial connection on port 5001
terminal_2: Listening for serial connection on port 5002
terminal_3: Listening for serial connection on port 5003
xcb_connection_has_error() returned true
xcb_connection_has_error() returned true
*** Booting Zephyr OS build zephyr-v3.4.0-1262-g029a4cb5370b ***
Secondary CPU core 1 (MPID:0x0) is up
Secondary CPU core 2 (MPID:0x1) is up
Secondary CPU core 3 (MPID:0x2) is up
Running TESTSUITE posix_apis
===================================================================
START - test_posix_pthread_barrier
 PASS - test_posix_pthread_barrier in 0.000 seconds
===================================================================
START - test_pthread_attr_stacksize
 PASS - test_pthread_attr_stacksize in 0.001 seconds
===================================================================
START - test_pthread_create_negative
 PASS - test_pthread_create_negative in 0.001 seconds
===================================================================
START - test_pthread_descriptor_leak
 PASS - test_pthread_descriptor_leak in 4.398 seconds
===================================================================
START - test_pthread_errors_errno
 PASS - test_pthread_errors_errno in 0.001 seconds
===================================================================
START - test_pthread_execution
Thread 1 starting with scheduling policy 1 & priority 15
Thread 2 starting with scheduling policy 1 & priority 15
Thread 0 starting with scheduling policy 1 & priority 15
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - main

**Additional context**

cfriedt, 2023-07-21:
@povergoing - I've occasionally seen weirdness on `qemu_cortex_a53` (maybe just the `_smp` variant). Do you think that would be sufficient to reproduce the issue?

povergoing, 2023-07-23:
> @povergoing - I've occasionally seen weirdness on qemu_cortex_a53 (maybe just the _smp variant). Do you think that would be sufficient to reproduce the issue?

If you meant the weirdness in `START - test_pthread_execution`, I think yes.

cfriedt, 2023-07-23:
@povergoing, I'm not seeing any issues using `qemu_cortex_a53_smp` with that testsuite.

However, I do feel that there is a race condition. Will comment on the PR.

PR #60566 (2023-10-04): lib: posix: mutex: to_posix_mutex should be spinlocked
to_posix_mutex allocates a mutex which will race to change the value of the lock, thus to_posix_mutex should be performed under the spinlock to prevent the racy issue.

Fixes #60613
closed 2023-10-04 by cfriedt