zephyrproject-rtos/zephyr #60616: RFC: API Change: mcumgr

kunoh, 2023-07-20:
## Introduction

**This section targets end users, TSC members, maintainers and anyone else that might
need a quick explanation of your proposed change.**

User data should be accessible dynamically through a parameter for SMP message handlers

### Problem description

**Why do we want this change and what problem are we trying to address?**

Currently user data can only be accessed in a handler through a static global variable. When designing in an object oriented way,
it would be nice to allow user data to be passed on to the handler.

### Proposed change

**A brief summary of the proposed change - the 10,000 ft view on what it will
change once this change is implemented.**

Add a pointer to user data as a parameter to the function pointer `mgmt_handler_fn` and an extra field to the struct `mgmt_handler`

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

Adding an extra parameter to the function pointer `mgmt_handler_fn` is a breaking change, all current supported and implemented handlers will need to have this argument added as well.
All users of SMP who may have implemented user defined handlers, would have to update their handlers.

### Proposed change (Detailed)

**This section is freeform - you should describe your change in as much detail
as possible. Please also ensure to include any context or background info here.
For example, do we have existing components which can be reused or altered.

By reading this section, each team member should be able to know what exactly
you're planning to change and how.**

Add a pointer to user data as a parameter to the function pointer `mgmt_handler_fn` and an extra field to the struct `mgmt_handler`,
see example below
```
typedef int (*mgmt_handler_fn)(struct smp_streamer *ctxt, void *user_data); <-- Added parameter

struct mgmt_handler {
	mgmt_handler_fn mh_read;
	mgmt_handler_fn mh_write;
	void *user_data; <-- Added field
};
```

### Dependencies

**Highlight how the change may affect the rest of the project (new components,
modifications in other areas), or other teams/projects.**

Existing handlers will break and an update of all handlers is therefore needed.

### Concerns and Unresolved Questions

**List any concerns, unknowns, and generally unresolved questions etc.**

## Alternatives

**List any alternatives considered, and the reasons for choosing this option
over them.**

nordicjm, 2023-07-20:
Why can it not call a function to get that? E.g. if your source exposes a struct with the required variables and has a function named `my_file_get_vars()` which returns the struct which the handler can then use?

kunoh, 2023-07-20:
It will reduce the coupling between each message handler and each struct/object will show the operation. If there is no user data this must be handled in a "know it all" function that can call the specific message callback (increased coupling)

Below is an example of how user data can be used:
```cpp
int SystemMgmt::HelloWorldPb(struct smp_streamer *ctxt, void *user_data)
{
    SystemMgmt *self = reinterpret_cast<SystemMgmt*>(user_data);
}
```

nordicjm, 2023-07-20:
I do not think it is wise the burden the many (C handlers that do not need this functionality) with the needs of the few (a C++ handler), there should only be one function handler per single command of a group, if you are using a single function for multiple commands then this is not how handler usage was designed. In my years of making MCUmgr handlers, I have never needed to do this

kunoh, 2023-07-20:
The intention is not to have a single function for multiple commands.

I think user data applies for both C and C++ developers. 
User data will allow a developer to have a dynamic architecture rather than static.

The concept of user data is common throughout multiple protocol implementations like libusb and the intention of this RFC is the same;
https://vovkos.github.io/doxyrest/samples/libusb/struct_libusb_transfer.html#doxid-structlibusb-transfer

nordicjm, 2023-07-20:
Can you give a complete example of your application (i.e. the actual functionality you are trying to implement) would want to use this and how calling a function to e.g. get a struct would not be suitable for it? I'm struggling to understand your usage.

de-nordic, 2023-07-20:
The USB example given above is completely missed because it represents different concept.
The `mgmt_handler` is type used for collecting together pointers to two callbacks as a single type used in an array. It is not passed as context, it is not managed by any kind of memory alloc/free mechanisms, it is not being passed between layers, etc. This thing is not obtained, created, generated or whatever at run time to be passed, retrieved, used and destroyed/returned to pool.
The USB thing (`struct libusb_transfer`) looks like something that will be allocated when transfer occurs, passed somewhere, retrieved , and then returned to pool or freed (actually it is even described that way).

kunoh, 2023-07-21:
An example of the user data usage could be something like this:

```cpp
class Display {
    int StopDisplay();
    int NextFrame();
}
```

```cpp
class DisplayMgmt {
public:
    DisplayMgmt(Display& disp);

private:
    static int StopDisplay(struct smp_streamer *ctxt, void *user_data);
    static int NextFrame(struct smp_streamer *ctxt, void *user_data) ;

private:
    Display& disp_;
    struct mgmt_group display_mgmt_group_;
    struct mgmt_handler display_mgmt_group_handlers_[DISPLAY_NUMBER_OF_COMMANDS];
}
```

```cpp
DisplayMgmt::DisplayMgmt(Display& disp)
    : disp_{disp}
{
    display_mgmt_group_handlers_[DISPLAY_MGMT_ID_STOP_DISPLAY] = {
            .mh_read = &DisplayMgmt::StopDisplay,
            .mh_write = NULL,
            .user_data = this,
    };
    display_mgmt_group_handlers_[DISPLAY_MGMT_ID_NEXT_FRAME] = {
            .mh_read = &DisplayMgmt::NextFrame,
            .mh_write = NULL,
            .user_data = this,
    };

    display_mgmt_group_ = {
        .node = NULL,
        .mg_handlers = display_mgmt_group_handlers_,
        .mg_handlers_count = DISPLAY_NUMBER_OF_COMMANDS,
        .mg_group_id = TM_MGMT_GROUP_ID_DISPLAY,
    };

    mgmt_register_group(&display_mgmt_group_);
}

int DisplayMgmt::StopDisplay(struct smp_streamer *ctxt, void *user_data) {
    DisplayMgmt* self = reinterpret_cast<DisplayMgmt*>(user_data);
    self->disp_.StopDisplay();
}

int DisplayMgmt::StopDisplay(struct smp_streamer *ctxt, void *user_data) {
    DisplayMgmt* self = reinterpret_cast<DisplayMgmt*>(user_data);
    self->disp_.NextFrame();
}
```

```cpp
int main(void) {
    Display disp;
    DisplayMgmt disp_hdlr(disp);
}

nordicjm, 2023-07-21:
So it is only applicable to C++, there can only ever be one of these objects in total because you can't have duplicate group IDs, and though my C++ might be a bit rusty, you should be able to access `this->disp_` from those functions without needing any user data.

kunoh, 2023-07-21:
A static function cannot use `this` and cannot reach anything that is non-static.

There are multiple implementations of user data in Zephyr already, to name a few;

`k_timer`
https://docs.zephyrproject.org/apidoc/latest/group__timer__apis.html

`uart`
https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html#c.uart_callback_set

de-nordic, 2023-07-21:
Both above things are designed to be called from unspecified code, due to some event.

MCUmgr handlers are always called from the MCUmgr processor, due to specified event and are executed in context hey provide by the scope of their code.

The difference is like asking for a ride and then when ride arrives trying to figure out what it is to know how to get in vs asking for a ride in a car.

Anyway, I would avoid changing MCUmgr callback types for now. We can just add `user_data`  to `mgmt_handler` and callbacks can use `mgmt_find_handler` to figure out the object for them, as callback knows what it is.

Additionally: the DisplayMgmt is singleton by its nature, as it may be hard to instantiate it twice, which is also true for any function calling `mgmt_register_group` with the same identifiers.
Generally anything that would be serving any MCUmgr commands needs singleton for its entry; by your example you may not have several instances of DisplayMgmt.
So you may as well change DisplaMgmt to singleton holding static pointer to object of a class currently serving these identifiers and your singleton static callbacks, registered as `mh_read` and `mh_write` would dig out the current object from the singleton held static  pointer and call methods of that object.

kunoh, 2023-08-07:
Sorry for my late reply, just came back from vacation.

Sounds like a good way to solve it, I will update my PR accordingly - thank you!

closed 2024-01-22 by kunoh