zephyrproject-rtos/zephyr #60624: docs: Search ordering has worsened

nordicjm, 2023-07-20:
**Describe the bug**
The doc search system has gone to pot recently, comparing the current `latest` and `3.3` when searching for GPIO yields very different results.
Latest: https://docs.zephyrproject.org/latest/search.html?q=gpio&check_keywords=yes&area=default
3.3: https://docs.zephyrproject.org/3.3.0/search.html?q=gpio&check_keywords=yes&area=default

The correct result, for the GPIO page, is 20th in the latest (and 3.4) results, it is 2nd in the 3.3 docs.

**To Reproduce**
Search

**Expected behavior**
The relevant (in this case GPIO) page to be near the top

**Impact**
Annoyance

gmarull, 2023-07-20:
I think there's an important point to make here:

The GPIO page title changed from `GPIO` to `General-Purpose Input/Output (GPIO)`:

https://docs.zephyrproject.org/latest/hardware/peripherals/gpio.html
https://docs.zephyrproject.org/3.3.0/hardware/peripherals/gpio.html

So this is likely not a new issue.

gmarull, 2023-07-20:
Also, FYI https://github.com/zephyrproject-rtos/zephyr/pull/60018

kartben, 2023-07-20:
> So this is likely not a new issue.

+1. @nordicjm unless you are seeing other examples of regressions like this one, I think we can close this.

nordicjm, 2023-07-21:
> > So this is likely not a new issue.
> 
> +1. @nordicjm unless you are seeing other examples of regressions like this one, I think we can close this.

It's an issue, I don't see why it should be closed, regardless if it's one search item or multiple.

> Also, FYI #60018

Having an absolutely invasive company's search engine is not a solution to anything

kartben, 2023-07-21:
> > > So this is likely not a new issue.
> > 
> > +1. @nordicjm unless you are seeing other examples of regressions like this one, I think we can close this.
> 
> It's an issue, I don't see why it should be closed, regardless if it's one search item or multiple.

At least can we agree it's a dupe of #55173? :)

nordicjm, 2023-07-21:
> > > > So this is likely not a new issue.
> > > 
> > > 
> > > +1. @nordicjm unless you are seeing other examples of regressions like this one, I think we can close this.
> > 
> > 
> > It's an issue, I don't see why it should be closed, regardless if it's one search item or multiple.
> 
> At least can we agree it's a dupe of #55173? :)

Yes, dupe of that

closed 2023-07-21 by kartben