zephyrproject-rtos/zephyr #60628: Issue while re-mounting the SD card after a SD power cycle

shubham-mante, 2023-07-20:
**Describe the bug**
I'm working on the project where I use NRF5340 with SDMMC driver and running into a problem while re-mounting the SD-card during runtime. Here, the SD Card power is controlled through a GPIO pin and during runtime, I need to do unmount, power OFF, power ON and mount SD card as part of application requirement. 

Normally the SD Card is initialized with disk_access_init() function at the beginning, but if the SD card is unmounted, powered off then powered on and tried re-mounting, it throws error fs mount error (-5). The issue I found in sdmmc_subsys.c:40 where the initialization is skipped if card already had status SD_OK during initial booting i.e., inside disk_sdmmc_access_init(), the sd_status (data->status) remains SD_OK even after fs_unmount(). 

After following a workaround as mentioned here (https://devzone.nordicsemi.com/f/nordic-q-a/96911/unable-to-remount-sd-card-after-physically-ejecting-without-rebooting) and by manually setting the sd_status (data->status) to SD_UNINIT before calling fs_mount(), the initialization inside disk_sdmmc_access_init() happens completely as given in below code. 

```
#include <zephyr/sd/sd.h>

#define MMC_NODE		DT_PATH(soc, peripheral_50000000, spi_a000, sdhc_0, mmc)
const struct device* mmc_dev	= DEVICE_DT_GET(MMC_NODE);

extern "C" 
{
enum sd_status {
	SD_UNINIT,
	SD_ERROR,
	SD_OK,
};
struct sdmmc_data {
	struct sd_card card;
	enum sd_status status;
	char *name;
};
}

void MountSD() {
    const struct device *dev = mmc_dev;
	struct sdmmc_data *data = (sdmmc_data*)dev->data;
    data->status = SD_UNINIT; // Set to UNINIT state prior to mounting
    //....
    //Do mounting etc here...
}
```

However, as this is just a workaround, please suggest a correct approach to unmount and re-mount the SD card during runtime considering SD power cycle.

**Additional information :**

    What target platform are you using?
        NRF5340, where SD Card power is controlled through a GPIO pin.

    What have you tried to diagnose or workaround this issue?
        Followed below procedure:
		 - Unmount the SD card using fs_unmount()
		 - Power OFF the SD card 
		 - Power ON the SD card
		 - Tried to mount the SD card using fs_mount() but got the error fs mount error (-5).

**To Reproduce**
Steps to reproduce the behavior:
 - hardware which allows to power off the actual SD Card
	 1. enable SDMMC
	 2. Unmount file system and power off the SD card during
	 3. Power ON the SD card and try to execute fs_mount()

**Expected behavior**
Should be able to re-mount the SD card after a power cycle (SD power OFF - ON)

**Impact**
Unable to remount SD without the mentioned workaround

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain : nrf connect SDK V2.2.0
 - Commit SHA or Version used: v3.2.99-ncs1

Please note that mounting, unmounting and remounting works fine, as long as the SD-card power cycle (OFF and ON) is not done. I couldn't find any other way for re-mounting the SD card after power cycle and hence, I tried to implement it myself. I contacted Nordic support team but in response they said that the functionality is a limitation of zephyr's disk access subsys, and recommendded to register an issue on zephyr github, hence I am registering it here.

danieldegrasse, 2023-07-20:
Hi @shubham-mante, this is indeed a limitation of the disk access subsystem. Zephyr as a whole does not provide mature support for "de-initializing" devices We could provide a `disk_access_deinit` function, but this would only solve half the issue. For example, FATFS does not provide a path to deinitialize a disk, only to unregister the mounted filesystem.

I think that manually setting the SD status to `SD_UNINIT` is an appropriate workaround for the issue

@de-nordic, do you have any thoughts on the value of an API like `disk_access_deinit`?

de-nordic, 2023-07-20:
> Hi @shubham-mante, this is indeed a limitation of the disk access subsystem. Zephyr as a whole does not provide mature support for "de-initializing" devices We could provide a `disk_access_deinit` [...]

I think we should have it. I was wondering whether we could actually move init and uninit under IOCTL and deprecate disk_access_init, reducing number of callbacks.
The problem is that, as mentioned above, that would require changes to fs_mount/unmount to be able to pass flags that would require init or deinit (re-init?). Currently users are not expected to do device initialization, they just mount something and magic happens.
Nevertheless with some work being done towards allowing powering down Zephyr, the de-initialization in Disk Access would be needed at some point anyway, so we may just start looking into what options we have right now.

danieldegrasse, 2023-07-20:
> I think we should have it. I was wondering whether we could actually move init and uninit under IOCTL and deprecate disk_access_init, reducing number of callbacks.

Given that the disk access API is considered stable, I would prefer we maintain the interface as is. Removing `disk_access_init` would be cleaner, but I don't think it justifies a stable API change.

To make disk deinitialization transparent for users, we would need to make changes to upstream FATFS, correct? I don't see any callback path in the driver that would run when `fs_unmount` is called. I do think that a `disk_access_deinit` API would be a good intermediate step though

de-nordic, 2023-07-21:
> Given that the disk access API is considered stable, I would prefer we maintain the interface as is. Removing `disk_access_init` would be cleaner, but I don't think it justifies a stable API change.

`disk_access_init` may remain as is just the `disk_operations` would get changed to drop the `init` and replace the call within `disk_access_init` to `ioctl` with proper command.

> 
> To make disk deinitialization transparent for users, we would need to make changes to upstream FATFS, correct? 

To VFS generally, because LittleFS can also use this and there is incoming ext2 support that could also use the new feature.

> I don't see any callback path in the driver that would run when `fs_unmount` is called. I do think that a `disk_access_deinit` API would be a good intermediate step though

`fs_umount` would probably need API change when it would take some flag or bool for the de-init request.

shubham-mante, 2023-07-21:
> `disk_access_init` may remain as is just the `disk_operations` would get changed to drop the `init` and replace the call within `disk_access_init` to `ioctl` with proper command.

I have one doubt here, will having a separate `disk_sdmmc_access_deinit` operation as a part of `disk_operations` which can be called inside the previously suggested `disk_access_deinit` API help? Because in **sdmmc_stm32.c:297**, I found `stm32_sdmmc_access_deinit()` and thought if same can be followed in **sdmmc_subsys.c**.

danieldegrasse, 2023-07-21:
> I have one doubt here, will having a separate disk_sdmmc_access_deinit operation as a part of disk_operations which can be called inside the previously suggested disk_access_deinit API help? Because [...]

Yes, a deinit API within the SD layer would be also be required in order to implement this fully for the SDMMC subsystem. The advantage of extending the deinit layering to the disk API is that other disk devices (and filesystems) could leverage it.

gargar165, 2023-10-18:
Trying to learn Zephyr and ran into this issue since I want to hotswap SDs without rebooting... Is the SD layer in "zephyr/subsys/sd"? I'm considering giving this a shot.

danieldegrasse, 2023-12-05:
> Trying to learn Zephyr and ran into this issue since I want to hotswap SDs without rebooting... Is the SD layer in "zephyr/subsys/sd"? I'm considering giving this a shot.

Yes, the SD layer is implemented there. We would likely need to implement a "deinit" API within the SD card code, and extend this to the disk API layer. If you'd like to look into this I'd be happy to review any PRs

gargar165, 2023-12-05:
Sounds good! I'll probably break it up into two PRs for the SD card code and disk API individually, but we will see

Yaxit, 2023-12-13:
Stumbled on this issue this week as well.
@gargar165 looking forward to your PRs!

gargar165, 2023-12-22:
FINALLY got my microSD SPI/SDIO breakout board in the mail. Not sure if any of my boards support SDIO at this moment, so I'm going to start with just SPI for the time being.

danieldegrasse, 2024-01-03:
> FINALLY got my microSD SPI/SDIO breakout board in the mail. Not sure if any of my boards support SDIO at this moment, so I'm going to start with just SPI for the time being.

I'd worry more about SDMMC cards than SDIO devices- I'm not aware of cases where SDIO or eMMC devices are generally hotswapped, but SD cards are of course expected to be removable during device operation. Thank you for taking a look at this!

Yaxit, 2024-03-04:
Not stale, still an issue

commit d18cbb60 (2024-06-07): drivers: disk: add DISK_IOCTL_CTRL_DEINIT command to supported IOCTLs

Add DISK_IOCTL_CTRL_DEINIT ioctl command to disk subsystem. When
disk_access_ioctl() is called with this command, the disk will be
de-initialized. After this IOCTL completes, the disk can safely be
reinitialized.

Fixes #60628