zephyrproject-rtos/zephyr #60629: http_client_req throws a FATAL EXCEPTION error whenever the wifi connection is unstable

mariucker, 2023-07-20:
**Describe the bug**
I'm performing tests with the example available at zephyr/samples/net/cloud/tagoIo_http_post and I'm having problems whenever I perform tests with wifi. I compiled the same example to perform the connection through a modem and everything worked. However, when I tested the example with wifi, some issues appeared. The errors only appear when the internet connection is lost. Sometimes the system hangs at this point in the sockets.c file:
  if (connect(ctx->sock, addr->ai_addr, addr->ai_addrlen) < 0) {
and in some tests a FATAL EXCEPTION error is generated when the http_client_req function is executed:
ret = http_client_req(ctx->sock, &req, ....

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p laways -b esp_wrover_kit samples/net/cloud/tagoio_http_post -- -DEXTRA_CONF_FILE=overlay-wifi.conf

**Expected behavior**
Internet connection failures should not crash code or generate FATAL EXCEPTION errors.

**Impact**
I can't create an application that crashes whenever the connection is lost.

**Logs and console output**
[00:00:11.320,000] <err> os:  ** FATAL EXCEPTION
[00:00:11.321,000] <err> os:  ** CPU 0 EXCCAUSE 28 (load prohibited)
[00:00:11.321,000] <err> os:  **  PC 0x40085638 VADDR 0x8c
[00:00:11.321,000] <err> os:  **  PS 0x60420
[00:00:11.321,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:4 CALLINC:2)
[00:00:11.321,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:" **  A0 %p  SP %p  A2 %p  A3 %p" argument:1
[00:00:11.321,000] <err> os:  **  A0 0x80085924  SP 0x3ffe7680  A2 (nil)  A3 0x60420
[00:00:11.321,000] <err> os:  **  A4 0x3ffb1a70  A5 0x3ffb7cfc  A6 0x60820  A7 (nil)
[00:00:11.321,000] <err> os:  **  A8 0x80085630  A9 0x3ffe7670 A10 0x3ffe69cc A11 0x3ffcc898
[00:00:11.321,000] <err> os:  ** A12 (nil) A13 (nil) A14 0x3ffce448 A15 (nil)
[00:00:11.321,000] <err> os:  ** LBEG 0x400eb0c8 LEND 0x400eb126 LCOUNT (nil)
[00:00:11.321,000] <err> os:  ** SAR 0x3

Ba0x40085635:0x30x40085921:0x30x40084fa8:0x30x400ec849:0x30x400d7be3:0x3

[00:00:11.321,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:11.321,000] <err> os: Current thread: 0x3ffcc808 (rx_q[0])
[00:00:11.519,000] <err> os: Halting system

rlubos, 2023-07-20:
> FATAL EXCEPTION error whenever the **wifi connection** is unstable

> The errors only appear when the **internet connection** is lost.

So which one is causing the issue? Do you need to disconnect at the Wi-Fi level, or just lose the internet connectivity?

Adding @sylvioalves as this looks more like a platform-related issue.

sylvioalves, 2023-07-20:
Hi @mariucker, which Zephyr version are you using?

@rlubos TBH, I can't even build that sample code:  
``` c
/home/sylvio/projects/espressif/zephyr/zephyr/samples/net/cloud/tagoio_http_post/src/wifi.c: In function 'handle_wifi_connect_result':
/home/sylvio/projects/espressif/zephyr/zephyr/samples/net/cloud/tagoio_http_post/src/wifi.c:18:46: error: 'struct net_mgmt_event_callback' has no member named 'info'
   18 |                                            cb->info;
      |                                              ^~
/home/sylvio/projects/espressif/zephyr/zephyr/samples/net/cloud/tagoio_http_post/src/wifi.c: At top level:
/home/sylvio/projects/espressif/zephyr/zephyr/samples/net/cloud/tagoio_http_post/src/wifi.c:28:13: warning: 'wifi_mgmt_event_handler' defined but not used [-Wunused-function]
   28 | static void wifi_mgmt_event_handler(struct net_mgmt_event_callback *cb,

```

I guess latest wifi_mgmt API changes might have caused this?

mariucker, 2023-07-20:
I am using version 3.3

rlubos, 2023-07-21:
@sylvioalves It looks that the configuration of the sample is incomplete for the board... I had to add DT overlay enabling wifi:
```
&wifi {
	status = "okay";
};
```
Also some adjustements in `overlay-wifi.conf`:
```
CONFIG_NET_DHCPV4=y
CONFIG_NET_L2_ETHERNET=y
```

Otherwise we're left with already known issue(enum collision, https://github.com/zephyrproject-rtos/zephyr/issues/59881) but that shouln't be the case for Zephyr v3.3.

carlescufi, 2023-07-26:
@sylvioalves could you try to reproduce? this does not look like a networking issue, but rather a platform-specific one like @rlubos pointed out.

rlubos, 2023-07-26:
> @rlubos, Most Wi-Fi sample codes for ESP32 requires `NET_RX_STACK_SIZE` and `NET_TCP_WORKQ_STACK_SIZE` to be higher than default level. I can add custom .conf for all samples or set default values in Kconfig file for each item. Any thoughts?

If it's always the case for the driver, that it needs more stacks, I guess it'd make sense to increase the default for that particular driver. I would consider however doing it in the driver's Kconfig file, we do it for `nRF700x`, you can redefine the Kconfig symbol and add a custom default, for example:
https://github.com/nrfconnect/sdk-nrf/blob/main/drivers/wifi/nrf700x/Kconfig.nrf700x#L67

mariucker, 2023-07-27:
> @mariucker I think the issue is related to stack size. Would you add the below in your application and re-test? I also suggest you, whenever possible, to run `kernel stacks` in shell terminal to evaluate used stack size. This might give you some hint in those weird scenarios. Let's see if it solves.
> 
> ```c
> CONFIG_NET_TCP_WORKQ_STACK_SIZE=2048
> CONFIG_NET_RX_STACK_SIZE=2048
> ```
> 
> @rlubos, Most Wi-Fi sample codes for ESP32 requires `NET_RX_STACK_SIZE` and `NET_TCP_WORKQ_STACK_SIZE` to be higher than default level. I can add custom .conf for all samples or set default values in Kconfig file for each item. Any thoughts?

@sylvioalves the addition of CONFIG_NET_TCP_WORKQ_STACK_SIZE=2048 and CONFIG_NET_RX_STACK_SIZE=2048 solved the problem. Thank you!

closed 2023-07-27 by sylvioalves