zephyrproject-rtos/zephyr #60631: LE Audio: Controller: Failed to establish CIS when IUT is Peripheral

mkasenberg, 2023-07-20:
At BAP/USR/SCC/BV-135-C PTS test case, where the IUT is in Peripheral role, I encounter an issue at HCI_LE_Create_CIS. The PTS controller rejects the command with 'Unsupported LMP Parameter Value' (0x20) status.
After searching commits related to CIS, I found out this CONFIG_BT_CTLR_PERIPHERAL_ISO_EARLY_CIG_START option, which fixed the issue. Is this a legit solution? Can I use it in the tester application freely?

Protocol Viewer logs:
- with CONFIG_BT_CTLR_PERIPHERAL_ISO_EARLY_CIG_START=y
[BAP_USR_SCC_BV_135_C_success.zip](https://github.com/zephyrproject-rtos/zephyr/files/12107517/BAP_USR_SCC_BV_135_C_success.zip)
- without CONFIG_BT_CTLR_PERIPHERAL_ISO_EARLY_CIG_START
[BAP_USR_SCC_BV_135_C_fail.zip](https://github.com/zephyrproject-rtos/zephyr/files/12107523/BAP_USR_SCC_BV_135_C_fail.zip)

henrikbrixandersen, 2023-07-25:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Thalley, 2023-08-14:
@mkasenberg please update this issue using the Bug template

Thalley, 2023-08-14:
@cvinayak please have a look at this issue as it sounds to be related to the controller

Thalley, 2023-10-16:
@mkasenberg is this still an issue? The Controlled has had several fixes and optimizations since this issue was created, so can you please retest?

mkasenberg, 2023-10-17:
I've tested again and it seems that the `CONFIG_BT_CTLR_PERIPHERAL_ISO_EARLY_CIG_START=y` is still needed to pass the BAP/USR/SCC/BV-135-C test case.

cvinayak, 2024-01-24:
> I've tested again and it seems that the `CONFIG_BT_CTLR_PERIPHERAL_ISO_EARLY_CIG_START=y` is still needed to pass the BAP/USR/SCC/BV-135-C test case.

I have reviewed the implementation and confirm that `CONFIG_BT_CTLR_PERIPHERAL_ISO_EARLY_CIG_START=y` is needed to successfully establish CIS with peer central controllers that use low cis_offset value compared to local zephyr controller's minimum cis_offset that it can support without have overlapping ACL plus CIS that is close to each other.

@mtpr-ot can confirm my understanding.

Closing this issue, as `CONFIG_BT_CTLR_PERIPHERAL_ISO_EARLY_CIG_START=y` is the fix. I will try to remember to send a PR to any `overlay-bt_ll_sw_split.conf` that upstream samples are using for CIS support in a subsequent PR soon.

closed 2024-01-24 by cvinayak