zephyrproject-rtos/zephyr #60638: define and document on how to add cases with pytest or robot

hakehuang, 2023-07-20:
**Is your enhancement proposal related to a problem? Please describe.**
Now zephyr has supported robot and pytest, and now the robot script is alone with samples code folder, and pytest cases are put to a testsuites/pytest/... folder. we need clear define on how to adding new cases for it.

**Describe the solution you'd like**
some proposal:

1. keep the existing paradigm as is, and let robot script go with source code, and pytest scripts goes to the testsuits/pytest folder with casename as subfolder.
2.  force all test scripts to go with src code.
3. force all inteactive test scripts goes to a testsuits/pytest folder, or even a separated repo.

**Additional context**

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/shell/shell_module/shell_module.robot

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/testsuite/pytest/shell/pytest/test_shell.py

PerMac, 2023-07-20:
We can only say that we are -1 on moving all pytests/robot test to special place, since the policy was always to keep tests directly in a folder with source code of the test app they relate to. We just proposed an extra subfolder called `pytest`. We also think that moving pytest files directly to the same folder where the testcase.yaml or to src will cause a mess. There can be more than one file used by pytest for such tests and keeping all the files mixed together will just cause mess. 
We don't mind the location for robot files.

hakehuang, 2023-07-27:
according to TWG discussion the agreements are:

```
Prefer : Add scripts along with source code src folder.
Not prefer: add a new directory to host the test script, unless author can justify 
```

PerMac, 2023-07-28:
Let me rephrase:
It is recommended to put tests in the same "top" directory the sources are. However, there can be some extra structure introduced by the tests. The reasoning is that when test are in some completely different place it is harder to figure out what is failing and where to find the code.