zephyrproject-rtos/zephyr #60662: rom_report is unaware of logging strings

JordanYates, 2023-07-21:
**Describe the bug**

The `rom_report` tool, which is very useful for tracking down memory usage, is normally unaware of logging strings, heaping them all together under the `(hidden)` path.

**To Reproduce**

`west build -b nrf52840dk_nrf52840 zephyr/samples/subsys/logging/logger -t rom_report`

```
Path                                                                                             Size    %
==============================================================================================================
Root                                                                                            37756 100.00%
├── (hidden)                                                                                     3634   9.62%
├── (no paths)                                                                                   1266   3.35%
```
`west build -b nrf52840dk_nrf52840 zephyr/samples/subsys/logging/logger -t rom_report -- -DCONFIG_LOG_FMT_SECTION=y`
```
Path                                                                                             Size    %
==============================================================================================================
Root                                                                                            39455 100.00%
├── (hidden)                                                                                     2014   5.10%
├── (no paths)                                                                                   4585  11.62%
```

With the addition of `CONFIG_LOG_FMT_SECTION`, which forces the strings into a defined memory region, `rom_report` can now remove them from `(hidden)`.

**Expected behavior**

Logging strings should be correctly assigned to their source file.

**Environment (please complete the following information):**
 - 278563ed45e5a0724d469e079348cbdd416bd1ad (Zephyr v3.5.99)

JordanYates, 2023-07-21:
Using the python `bitarray` library I have hacked together some code in `size_report` that displays which memory addresses are not being 'found' by the script.
```
            addr_min = min([x['start'] for x in addr_ranges['rom']])
            addr_max = max([x['end'] for x in addr_ranges['rom']])
            b = bitarray.bitarray(addr_max - addr_min)
            b.setall(0)
            for symlist in symbols['rom'].values():
                for onesym in symlist:
                    s = onesym['symbol']
                    start = s['st_value'] - addr_min
                    b[start:start+s['st_size']] = True

            total_missing = 0
            for i in bitarray.util.intervals(b):
                if (i[0] == 0):
                    size = i[2] - i[1]
                    print("{:d} empty bytes @ {:08X}".format(size, i[1] + addr_min))
                    if (size < 10e3):
                        total_missing += size
            print("Total Missing:", total_missing)
```
Running `west build -b nrf52840dk_nrf52840 zephyr/samples/hello_world/ -t rom_report` outputs the following (with manually annotated sources):
```
64 empty bytes @ 00000000 (_vector_table)
49 empty bytes @ 00000100 (__aeabi_uldivmod)
2 empty bytes @ 00000133 (padding?)
4 empty bytes @ 00000D99 (z_arm_platform_init  [.S file])
60 empty bytes @ 0000100D (z_arm_cpu_idle_init [.S file])
8 empty bytes @ 000010B1 (z_SysNmiOnReset  [.S file])
116 empty bytes @ 00001135 (z_arm_pendsv [.S file])
92 empty bytes @ 00001239 (_isr_wrapper [.S file])
20 empty bytes @ 00001521 (z_arm_bus_fault and friends [.S file])
48 empty bytes @ 0000154D (z_arm_reset [.S file])
7 empty bytes @ 000047CD (padding between .text and init structures? refers to _udivmoddi4 for some reason)
4 empty bytes @ 00004A5C (padding between .log_const_ and .rodata)
2 empty bytes @ 00004A66 (*fill*)
82 empty bytes @ 00004B74 (.rodata.main.str1.1 read only data from various files)
91 empty bytes @ 00004BCC (.rodata.main.str1.1 read only data from various files)
536851405 empty bytes @ 00004C33 (artifact of my hacky script and .rodata)
Total Missing: 649
```
Which roughly lines up with the 766 hidden bytes reported by the rest of the tool. (The difference is larger for more complex applications though, which indicates I'm missing something [29233 vs 39544])

dcpleung, 2023-08-03:
The report relies on ELF debug information to gather symbols' data and their size. If the compiler does not emit symbols for a piece of data, it is grouped under `(hidden)`.

Moreover, counting strings are complicated as the compiler would combine partial strings together, which would result in double counting.

There is not much we can do if we only look at the ELF file for symbols after the fact where the linker has already discarded those symbols. Maybe someone can write a linker plugin?

closed 2024-01-13 by github-actions[bot]

macgyverek, 2025-01-09:
> Using the python `bitarray` library I have hacked together some code in `size_report` that displays which memory addresses are not being 'found' by the script.
>
> ```
>             addr_min = [...]

Could you attach modified size_raport file?? I have problem with add a path to the original code.