zephyrproject-rtos/zephyr #60668: Driver: Goodix, GT911 touch controller stops working/responding if there are multiple touch inputs

mariopaja, 2023-07-21:
Touch controller stops working/responding if there are multiple touches. I know that multi-touch is ignored (although support would be great). However, the driver should better handle when the end-user violates this condition. A simple solution would be to replace **return 0** with **points = 1**. 

https://github.com/zephyrproject-rtos/zephyr/blob/b69cd3cbc101cae3558a4730fa320e25a43b1543/drivers/input/input_gt911.c#L119-L122

fabiobaltieri, 2023-07-25:
@danieldegrasse you have hw to test this right? Any chance you could look into it?

danieldegrasse, 2023-08-15:
> @danieldegrasse you have hw to test this right? Any chance you could look into it?

I do- just tried this with an RT1170 and RK055HDMIPI4MA0 module, which uses the GT911 touch IC. I wasn't able to reproduce the issue- even when using several fingers to touch the display still recognized and responded to one of my inputs.

@mariopaja- I do see how this could be an issue, although I can't reproduce it on my display. How are you able to reproduce this?

mariopaja, 2023-08-17:
@danieldegrasse I am using a custom esp32 board that which uses GT911 touch IC that I bought out of aliexpress (See below). I don't exclude the possibility that it could be  a HW problem. I know it is bit difficult to reproduce it because it is a very specific case

[image]
[image]

This is the code I am using:

**main.c**
```
#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(main, 0);
#define RET_STR(ret) ((ret) == 0 ? "OK" : "ERROR")
#define LOG(func, ret) \
	LOG_INF("%s: %d <%s>", (func), ret, RET_STR(ret))

#include <stdio.h>
#include <errno.h>
#include <zephyr/kernel.h>
#include <zephyr/devicetree.h>
#include <zephyr/drivers/gpio.h>
#include <zephyr/drivers/i2c.h>
#include <zephyr/drivers/kscan.h>

#define I2C_1 DT_NODELABEL(i2c1)
#if !DT_NODE_HAS_STATUS(I2C_1, okay)
#error "I2C_1 is disabled"
#endif

#define SPI_2 DT_NODELABEL(spi2)
#if !DT_NODE_HAS_STATUS(SPI_2, okay)
#error "SPI_2 is disabled"
#endif

const struct device *gt911_dev = DEVICE_DT_GET(DT_CHOSEN(zephyr_keyboard_scan));

static void gt911_cb(const struct device *dev, uint32_t row, uint32_t col, bool pressed)
{
	ARG_UNUSED(dev);
	if (pressed)
	{
		LOG_INF("row = %u col = %u", row, col);
	}
}

void gt911_enable(void)
{
	if (!device_is_ready(gt911_dev))
	{
		LOG_ERR("%s is not ready", gt911_dev->name);
		return;
	}

	kscan_config(gt911_dev, gt911_cb);
	kscan_enable_callback(gt911_dev);
}

void main(void)
{
	int ret;

	gt911_enable();

	while (true)
	{
		k_msleep(1000);
	}
}
```

pro.conf
```
# LOG Configuration
CONFIG_LOG=y
CONFIG_NET_LOG=y
CONFIG_LOG_BACKEND_SHOW_COLOR=y
CONFIG_LOG_INFO_COLOR_GREEN=y
CONFIG_NET_CONFIG_LOG_LEVEL_INF=y
CONFIG_LOG_BUFFER_SIZE=8192

# Touch Panel Configuration
CONFIG_I2C=y
CONFIG_KSCAN=y
CONFIG_KSCAN_GT911=y   

```

mariopaja, 2023-08-26:
@danieldegrasse I just created a [PR 61922](https://github.com/zephyrproject-rtos/zephyr/pull/61922) for this matter

faxe1008, 2023-08-29:
I can replicate the issue on a RK043FN66HS-CTG display, performing a multi touch leads to unresponsiveness of the gt911.

danieldegrasse, 2023-08-30:
> I can replicate the issue on a RK043FN66HS-CTG display, performing a multi touch leads to unresponsiveness of the gt911.

thanks for the tip, I can replicate the issue with the same display

fabiobaltieri, 2023-08-30:
https://github.com/zephyrproject-rtos/zephyr/pull/61922 is merged, closing, thanks for the contribution.

closed 2023-08-30 by fabiobaltieri