zephyrproject-rtos/zephyr #60678: `portability.posix.eventfd` is failing on `qemu_cortex_a53_smp`

carlocaione, 2023-07-21:
On these PRs:

https://github.com/zephyrproject-rtos/zephyr/pull/60626/checks?check_run_id=15229833588
https://github.com/zephyrproject-rtos/zephyr/actions/runs/5620196109/job/15228779043

the test is failing with:

```
START - test_stress
BOARD: qemu_cortex_a53
TEST_DURATION_S: 5
UPDATE_INTERVAL_S: 1
223209 writes
223209 reads
451461 writes
451461 reads
682912 reads
682940 writes
914342 writes
914308 reads
avg: 228872 writes/s
avg: 228865 reads/s
E: ELR_ELn: 0x0000000000000000
E: ESR_ELn: 0x0000000086000006
E:   EC:  0x21 (Instruction Abort taken without a change in Exception level.)
E:   IL:  0x1
E:   ISS: 0x6
E: FAR_ELn: 0x0000000000000000
E: TPIDRRO: 0x0100000040034ca8
E: x0:  0x0000000000000000  x1:  0x0000000040031000
E: x2:  0x0000000000000000  x3:  0x0000000000000001
E: x4:  0x000000004007f3f0  x5:  0x000000000000000a
E: x6:  0x00000000ffffffd8  x7:  0x0000000000000073
E: x8:  0x000000000000000a  x9:  0x00000000ffffffff
E: x10: 0x0000000000000000  x11: 0x0000000000000000
E: x12: 0x0000000000000000  x13: 0x0000000000000000
E: x14: 0x0000000000000000  x15: 0x0000000000000000
E: x16: 0x0000000000000000  x17: 0x0000000000000000
E: x18: 0x0000000000000000  lr:  0x0000000000000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 1
E: Current thread: 0x400327b0 (test_stress)
E: Halting system
```

The problem was bisected down to https://github.com/zephyrproject-rtos/zephyr/commit/2bc980f83190b22f4aa3f6b1560b40ece649ed95 but the root cause must be still found.

carlocaione, 2023-07-21:
Test was disabled in https://github.com/zephyrproject-rtos/zephyr/pull/60675

fabiobaltieri, 2023-07-21:
cc @mniestroj @jgl-meta @cfriedt

gramsay0, 2023-08-12:
I was having a play with this trying to make a minimal example.
Can someone verify that I'm not doing something silly...

This crashes for me, even without SMP:

source:

	#include <zephyr/kernel.h>
	
	static K_THREAD_STACK_DEFINE(thread_stack, 4096);
	
	static void thread_func(void *arg1, void *arg2, void *arg3)
	{
		printk("");
	}
	
	static void run_thread(void)
	{
		struct k_thread thread; // <-- make static to prevent crash
		k_thread_create(&thread, thread_stack, K_THREAD_STACK_SIZEOF(thread_stack),
				thread_func, NULL, NULL, NULL, 0, 0, K_NO_WAIT);
		k_thread_join(&thread, K_FOREVER);
	}
	
	int main(void)
	{
		printk("Start\n");
		run_thread();
		printk("done\n"); // <-- crash
		k_sleep(K_FOREVER);
		return 0;
	}

conf:

	CONFIG_LOG=y
	
Build and run:

	west build -b qemu_cortex_a53 path/to/test -t run

cfriedt, 2023-08-12:
> 		struct k_thread thread; // <-- make static to prevent crash

The k_thread should probably not be in a separate thread's stack area space. If the MMU is enabled, it would likely fault there. That explanation would likely rationalize the crash, in any case.

gramsay0, 2023-08-12:
> > ```
> >   struct k_thread thread; // <-- make static to prevent crash
> > ```
> 
> The k_thread should probably not be in a separate thread's stack area. If the MMU is enabled, it would likely fault there. That explanation would likely rationalize the crash, in any case.

That would make sense, although moving the thread up to `main` and passing a pointer to `run_thread` also prevents the crash.
Maybe something is accessing the thread variable after it goes out of scope?

gramsay0, 2023-08-13:
I think I might have found the issue...

`CONFIG_FPU_SHARING` is enabled by default, but I don't think it clears ownership when the thread ends.
Clearing that ownership before leaving the k_thread scope seems to fix it:

	static void run_thread(void)
	{
		struct k_thread thread;
		k_thread_create(&thread, thread_stack, K_THREAD_STACK_SIZEOF(thread_stack),
				thread_func, NULL, NULL, NULL, 0, 0, K_NO_WAIT);
		k_thread_join(&thread, K_FOREVER);
	
		struct k_thread *owner = atomic_ptr_get(&_current_cpu->arch.fpu_owner);
		if (owner == &thread) {
			atomic_ptr_clear(&_current_cpu->arch.fpu_owner);
		}
	}

Otherwise the k_thread's `saved_fp_context` is accessed after it goes out of scope

cfriedt, 2023-08-13:
> I think I might have found the issue...
> 
> `CONFIG_FPU_SHARING` is enabled by default, but I don't think it clears ownership when the thread ends.
> Clearing that ownership before leaving the k_thread scope seems to fix it:

Oh wow - good find!!

closed 2023-08-16 by carlescufi

gramsay0, 2023-08-17:
Test is re-enabled in https://github.com/zephyrproject-rtos/zephyr/pull/61582 now that fix is merged