zephyrproject-rtos/zephyr #60683: Mcuboot app.overlay is being dropped if another overlay is specified

marcoster, 2023-07-21:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

Could be a regression to: #15083

**Describe the bug**
The issue happnes on zephyr-3.4.0.

While building *bootloader/mcuboot/boot/zephyr* i.e. for the *nrf52840dk_nrf52840* target includes the app.overlay by default, this overlay is being dropped when either 1) an overlay file is passed through the *west build* command (with --DDTC_OVERLAY_FILE=test.overlay) or if an overlay file is present for the specified board.
Since *app.overlay* specifies that boot_partition should be chosen for mcuboot as application partition/linking the application is begin linked to the wrong memory address in my case.

**To Reproduce**
Variant1: build bootloader with additional overlay
`cd bootloader/mcuboot/boot/zephyr
touch test.overlay
west build -p -b nrf52840dk_nrf52840 -- -DDTC_OVERLAY_FILE=test.overlay
`

Variant 2: add overlay for board and try to build:
The latter I've currently tested by creating an empty overlay for the nrf52840dk_nrf52840 in the boards folder of the bootloader (path mentioned above) through:
`cd bootloader/mcuboot/boot/zephyr
touch boards/nrf52840dk_nrf52840
west build -p -b nrf52840dk_nrf52840
`

**Expected behavior**
Don't drop app.overlay if another overlay is being specified in addition to it.

**Impact**
Bootloader is being linked to slot0_partition instead to boot_partition as specified in app.overlay

**Logs and console output**
```
Without additional overlay:
mcuboot/boot/zephyr (74c4d1c) ❯ west build -p -b nrf52840dk_nrf52840 -- -DDTC_OVERLAY_FILE=test.overlay
-- west build: making build dir /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.26.4
-- Found Python3: /usr/bin/python3.11 (found suitable exact version "3.11.3") found components: Interpreter
-- Cache files will be written to: /home/marco/.cache/zephyr
-- Zephyr version: 3.4.0 (/home/marco/dev/zephyr/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: nrf52840dk_nrf52840
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.1 (/opt/zephyr-sdk)
-- Found Dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/marco/dev/zephyr/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: test.overlay
-- Generated zephyr.dts: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/dts.cmake
Parsing /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/marco/dev/zephyr/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/boards/nrf52840dk_nrf52840.conf'
Configuration saved to '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
MCUBoot bootloader key file: /home/marco/dev/zephyr/bootloader/mcuboot/root-rsa-2048.pem
-- Configuring done (1.6s)
-- Generating done (0.0s)
-- Build files have been written to: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build
-- west build: building application
[1/276] Preparing syscall dependency handling

[2/276] Generating include/generated/version.h
-- Zephyr version: 3.4.0 (/home/marco/dev/zephyr/zephyr), build: zephyr-v3.4.0
[266/276] Linking C executable zephyr/zephyr_pre0.elf

[270/276] Linking C executable zephyr/zephyr_pre1.elf

[276/276] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       37858 B       472 KB      7.83%
             RAM:       23808 B       256 KB      9.08%
        IDT_LIST:          0 GB         2 KB      0.00%

```
With an overlay:`
```
mcuboot/boot/zephyr (74c4d1c) ❯ west build -p -b nrf52840dk_nrf52840
-- west build: making build dir /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.26.4
-- Found Python3: /usr/bin/python3.11 (found suitable exact version "3.11.3") found components: Interpreter
-- Cache files will be written to: /home/marco/.cache/zephyr
-- Zephyr version: 3.4.0 (/home/marco/dev/zephyr/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: nrf52840dk_nrf52840
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.1 (/opt/zephyr-sdk)
-- Found Dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/marco/dev/zephyr/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/boards/nrf52840dk_nrf52840.overlay
-- Generated zephyr.dts: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/dts.cmake
Parsing /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/marco/dev/zephyr/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/boards/nrf52840dk_nrf52840.conf'
Configuration saved to '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
MCUBoot bootloader key file: /home/marco/dev/zephyr/bootloader/mcuboot/root-rsa-2048.pem
-- Configuring done (1.7s)
-- Generating done (0.0s)
-- Build files have been written to: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build
-- west build: building application
[1/276] Preparing syscall dependency handling

[2/276] Generating include/generated/version.h
-- Zephyr version: 3.4.0 (/home/marco/dev/zephyr/zephyr), build: zephyr-v3.4.0
[266/276] Linking C executable zephyr/zephyr_pre0.elf

[270/276] Linking C executable zephyr/zephyr_pre1.elf

[276/276] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       37858 B       472 KB      7.83%
             RAM:       23808 B       256 KB      9.08%
        IDT_LIST:          0 GB         2 KB      0.00%

```

**Environment (please complete the following information):**
 - OS: Arch Linux x86_64
 - zephyr-sdk 0.16.1-1
 - Commit SHA or Version used
    - zephyr: 356c8cb
    - mcuboot: 74c4d1c

**Additional context**
Fresh init of zephry 3.4.0; no changes applied.

jfischer-no, 2023-07-21:
Please read https://docs.zephyrproject.org/latest/build/dts/howtos.html#set-devicetree-overlays

nordicjm, 2023-07-21:
Not a bug, this is expected behaviour. The bug was actually in previous versions of mcuboot whereby it forcefully appended to the list (you would get an ever expanding list each time cmake re-ran, if you go to an old version of zephyr/mcuboot and configure once then just run `cmake .` over and over you will see the number of times app.overlay is appended keeps growing and growing)

closed 2023-07-21 by nordicjm

marcoster, 2023-07-24:
Oh sorry, my fault - but at least the behaviour of dropping the *app.overlay* was kind of really unexpected.