zephyrproject-rtos/zephyr #60684: Mcuboot: adding jedec,spi-nor to dts breaks build

marcoster, 2023-07-21:
**Describe the bug**
When adding an *jedec,spi-nor* flash to an dts file of an board (in my testings nrf52840dk_nrf52840) the build of mcuboot fails with a linker error mentioning "undefined reference to '__device_dts_ord_132'.

If the spi-nor is set to *status="disabled"* the build works again without an error. But since I normally don't want to specify separate dts files for my application and my bootloader this isn't that nice.
Also I'm not able to get the bootlaoder working on external nor i.e. for the slot1_partition with this (found some examples but wasn't able to get it to build).

**To Reproduce**
1. Initialize new zephyr 3.4.0 env
2. Edit *zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts* (details see below)
3. goto bootloader/mcuboot/boot/zephyr
4. west build -p -b nrf52840dk_nrf52840
5. --> build fails

Build-Output:
```
mcuboot/boot/zephyr (74c4d1c) ❯ west build -p -b nrf52840dk_nrf52840
-- west build: making build dir /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.26.4
-- Found Python3: /usr/bin/python3.11 (found suitable exact version "3.11.3") found components: Interpreter
-- Cache files will be written to: /home/marco/.cache/zephyr
-- Zephyr version: 3.4.0 (/home/marco/dev/zephyr/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: nrf52840dk_nrf52840
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.1 (/opt/zephyr-sdk)
-- Found Dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/marco/dev/zephyr/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/dts.cmake
Parsing /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/marco/dev/zephyr/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/boards/nrf52840dk_nrf52840.conf'
Configuration saved to '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at /home/marco/dev/zephyr/zephyr/CMakeLists.txt:870 (message):
  No SOURCES given to Zephyr library: drivers__spi

  Excluding target from build.

MCUBoot bootloader key file: /home/marco/dev/zephyr/bootloader/mcuboot/root-rsa-2048.pem
-- Configuring done (1.6s)
-- Generating done (0.0s)
-- Build files have been written to: /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build
-- west build: building application
[1/278] Preparing syscall dependency handling

[2/278] Generating include/generated/version.h
-- Zephyr version: 3.4.0 (/home/marco/dev/zephyr/zephyr), build: zephyr-v3.4.0
[268/278] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/zephyr_pre0.map
: && ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  modules/mbedtls/libmodules__mbedtls.a  modules/mcuboot/libmcuboot_util.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -L/home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cd /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build/zephyr && /usr/bin/cmake -E echo
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(spi_nor.c.obj):(.rodata.spi_nor_config_0+0x0): undefined reference to `__device_dts_ord_132'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/marco/dev/zephyr/bootloader/mcuboot/boot/zephyr/build
```

dts-file spi2 after changes:
```
&spi2 {
	compatible = "nordic,nrf-spi";
	status = "okay";
	pinctrl-0 = <&spi2_default>;
	pinctrl-1 = <&spi2_sleep>;
	pinctrl-names = "default", "sleep";

	cs-gpios = <&gpio1 12 GPIO_ACTIVE_LOW>;

	at25: at25sf081b@0 {
		compatible = "jedec,spi-nor";
        //status = "disabled";
		reg = <0>;
		spi-max-frequency = <1000000>;
		jedec-id = [1f 85 01];
		size = <0x800000>;
		has-lock = <0x00>;
		has-dpd;
		t-enter-dpd = <500>;
		t-exit-dpd = <20>;
	};
};
```

Adding `nordic,pm-ext-flash = &at25;` to *chosen* doesn't change anything (saw this on some examples).
I'm able to get rid of the warning *No SOURCES given to Zephyr library: drivers__spi* by adding *CONFIG_NRFX_SPI2* to the defconfig; but then I get some compiler warnings (unused spi functions) and the linking still fails with the same error.

**Expected behavior**
A build of the bootloader with the dts file, more or less ignoring the defined flash.

**Impact**
Separate dts files or at least a specific overlay needed for mcuboot which doesn't seem right.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Arch-Linux x84_64
 - zephyr-sdk 0.16.1-1
 - zephyr 356c8cb
 - mcuboot 74c4d1c

**Additional context**
Fresh init of zephry 3.4.0; no changes applied.

nordicjm, 2023-07-21:
You get the ord error because you have a device enabled but without corresponding Kconfig to include the driver, i.e. you have not enabled SPI or SPI NOR, so you need to enable it to use it.

As for disabled devices, this is a known longstanding issue and duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/52395

nordicjm, 2023-07-21:
> Adding nordic,pm-ext-flash = &at25; to chosen doesn't change anything (saw this on some examples).

That applies to NCS only, NCS uses partition manager which is unrelated to and not part of zephyr

marcoster, 2023-07-24:
Thanks for the help - the configurations you mentioned I already tried but digging a bit deeper (and having a weekend off) lead met to *CONFIG_MULTITHREADING* as the missing piece in my puzzle (which lead to a quite missleading linker-error imho...; somehow similar to #10028 i guess).

nordicjm, 2023-07-24:
> Thanks for the help - the configurations you mentioned I already tried but digging a bit deeper (and having a weekend off) lead met to _CONFIG_MULTITHREADING_ as the missing piece in my puzzle (which lead to a quite missleading linker-error imho...; somehow similar to #10028 i guess).

That doesn't really make sense, the error you got is specifically an error because a driver is being used without anything registering to it, thus the symbol for it is missing. If you had multithreading disabled, it also wouldn't build and would fail at the linking stage, but it would not fail with that error, it would fail with errors about being unable to find semaphore and mutex functions.

marcoster, 2023-07-24:
> That doesn't really make sense, the error you got is specifically an error because a driver is being used without anything registering to it, thus the symbol for it is missing. If you had [...]

Yeah, it doesn't really make sensor for me as well, but I've just retried the following:

- apply the dts-changes as shown above (no flash partition defined on at25)
- build mcuboot with `west build -p -b nrf52840dk_nrf52840` --> results in the ord error as above
- add CONFIG_MULTITHREADING to nrf52840dk_nrf52840_defconfig
- build with again with the same command --> results in a linker error because rodata will not fit in FLASH - which is easily fixable by modifying the flash0 layout in devicetree.

I also tried around now with the following config-options to get the build fixed without multithreading:
`CONFIG_NRFX_SPI2=y
CONFIG_SPI_NOR=y
CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_SPI=y
CONFIG_FLASH=y
` 
But none fixed the build unless I added MULTITHREADING.
And all the other options don't seem to make any difference to code-size - FLASH is always overflowed by exactly 2334 bytes.

At least for our use-case here we're fine and I think we'll even use the external flash in some future products for *slot1_partition* since it looks like some stuff with graphics on a display will come and then it's definitely good to have this option for a bigger app partition.

If I can investigate anything regarding this feel free to give me some hints or ideas what I can try.