zephyrproject-rtos/zephyr #60696: No output on SWO pin (EFR32BG22)

reportingsjr, 2023-07-21:
**Describe the bug**
I'm running test firmware on an efr32bg22 thunderboard dev board (SiLabs BRD4184B) and I'm trying to configure SWO logging so I don't need to add a debug UART to my hardware. I have a handful of LOG_WRN and LOG_INFO scattered in the event loop of my program. These logging calls work ok if I leave the default logging backend (UART output).

Enabling CONFIG_LOG_BACKEND_SWO does not seem to work. I confirmed that SWO works fine with some example code from Simplicity Studio. 
I did add a PIN_SWO define for PA03 to the efr32bg22 soc.h file to make sure that wasn't messing with anything.

Additionally I followed the SWO setup from simplicity studio and compared it to the setup in soc/arm/silabs_exx32/common/soc.c and it looks like things are set up correctly.

Lastly, I left the SWO frequency configuration as 0 so it uses the default, which is 2250kHz. J-link SWO viewer detects this and seems to be ok with it, but I haven't tried reducing the frequency.

I measured the SWO pin with a scope just to make sure there wasn't an issue with the hardware, and there is no activity.

**To Reproduce**
Enable logging and the SWO backend. Use J-link SWO viewer to see SWO log output.

**Expected behavior**
Logging output in the J-link SWO viewer.

**Impact**
Annoyance

**Logs and console output**
No logs since SWO output isn't working

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: latest head

kmeinhar, 2023-07-25:
Hi @reportingsjr I had the same problem with SWO.
But this was quite a while ago and we since switch to RTT logging because we needed the pin.
Back then I wrote down the following notes, I hope they help you at least create a local patch:
* In zephyr/soc/arm/silabs_exx32/common/soc.c add define PIN_SWO
  * #define PIN_SWO {gpioPortA, 3, gpioModePushPull, 1}
  * Also prevent swo_init() to be excluded with ifdef CONFIG_SOC_GECKO_DEV_INIT
* In GDB execute: mon SWO EnableTarget 0 0 1 0
* Open Telnet connection to JLink: telnet localhost 2332

kmeinhar, 2023-07-25:
Ans I found this patch I hope that helps a bit:
```
diff --git a/soc/arm/silabs_exx32/common/soc.c b/soc/arm/silabs_exx32/common/soc.c
index c2e152aec0..2b1e235400 100644
--- a/soc/arm/silabs_exx32/common/soc.c
+++ b/soc/arm/silabs_exx32/common/soc.c
@@ -174,9 +174,11 @@ static ALWAYS_INLINE void dcdc_init(void)
 #endif
 
 #ifdef CONFIG_LOG_BACKEND_SWO
+#define PIN_SWO {gpioPortA, 3, gpioModePushPull, 1}
 static void swo_init(void)
 {
        struct soc_gpio_pin pin_swo = PIN_SWO;
+       CMU_ClockEnable(cmuClock_GPIO, true);
 
 #if defined(_SILICON_LABS_32B_SERIES_2)
        GPIO->TRACEROUTEPEN = GPIO_TRACEROUTEPEN_SWVPEN;
@@ -235,12 +237,11 @@ static int silabs_exx32_init(void)
        /* Initialize system clock according to CONFIG_CMU settings */
        clock_init();
 
+#endif /* !CONFIG_SOC_GECKO_DEV_INIT */
 #ifdef CONFIG_LOG_BACKEND_SWO
        /* Configure SWO debug output */
        swo_init();
 #endif
-#endif /* !CONFIG_SOC_GECKO_DEV_INIT */
-
        return 0;
 }

```

fabiobaltieri, 2023-07-25:
Hi, Zephyr is an open-source project, and as such we welcome and encourage the community to submit patches directly. Would you be able to submit a pull request with the fix? This is the [contribution guideline documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines).

kmeinhar, 2023-07-26:
Hi @fabiobaltieri , my patch is not tested and needs more work (e.g. configurable SWO pin). Sadly currently I do not have a board to test changes.

fabiobaltieri, 2023-07-26:
> Hi @fabiobaltieri , my patch is not tested and needs more work (e.g. configurable SWO pin). Sadly currently I do not have a board to test changes.

Got it, @reportingsjr could you help with that? Up the maintainer if they have time to work on it otherwise.

closed 2023-10-09 by github-actions[bot]