zephyrproject-rtos/zephyr #60714: PWM lockup issue after update to nrfx 3.1.0

fabiobaltieri, 2023-07-22:
**Describe the bug**
Hi, bumped into another PWM issue after the 3.1.0 upgrade (https://github.com/zephyrproject-rtos/zephyr/pull/59838). This is a bit odd, somehow the driver locks up in some situation when calling `pwm_set_pulse_dt` multiple times with the same pulse value but with a delay in between (yeah I know, I had a pulse code that was repeating the set to 0 and triggered it, just my luck).

It can be reproduced with this patch on top of the stock `fade_led` sample:

```diff
diff --git a/samples/basic/fade_led/src/main.c b/samples/basic/fade_led/src/main.c
index c49f1a9936..7b5f53ad71 100644
--- a/samples/basic/fade_led/src/main.c
+++ b/samples/basic/fade_led/src/main.c
@@ -41,6 +41,13 @@ int main(void)
                        return 0;
                }
 
+               k_sleep(K_MSEC(SLEEP_MSEC));
+               ret = pwm_set_pulse_dt(&pwm_led0, pulse_width);
+               if (ret) {
+                       printk("Error %d: failed to set pulse width\n", ret);
+                       return 0;
+               }
+
                if (dir) {
                        pulse_width += step;
                        if (pulse_width >= pwm_led0.period) {
```

build with `west build -p -b nrf52dk_nrf52832 samples/basic/fade_led`

locks up somewhere around:

```
#0  __nrfy_internal_pwm_events_process (p_reg=0x4001c000, mask=2) at /home/fabiobaltieri/zephyrproject/modules/hal/nordic/nrfx/haly/nrfy_pwm.h:446
#1  0x00004e84 in nrfy_pwm_events_process (mask=2, p_reg=<optimized out>) at /home/fabiobaltieri/zephyrproject/modules/hal/nordic/nrfx/haly/nrfy_pwm.h:157
#2  pwm_stopped_check (p_instance=p_instance@entry=0x788c <pwm_nrfx_0_config>) at /home/fabiobaltieri/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_pwm.c:181
#3  0x000071f2 in nrfx_pwm_stopped_check (p_instance=p_instance@entry=0x788c <pwm_nrfx_0_config>) at /home/fabiobaltieri/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_pwm.c:454
#4  0x00003a86 in pwm_nrfx_set_cycles (dev=<optimized out>, channel=<optimized out>, period_cycles=<optimized out>, pulse_cycles=<optimized out>, flags=<optimized out>) at /home/fabiobaltieri/zephyrproject/zephyr/drivers/pwm/pwm_nrfx.c:201
#5  0x00000436 in z_impl_pwm_set_cycles (flags=1, pulse=<optimized out>, period=320000, channel=0, dev=0x7554 <__device_dts_ord_54>) at /home/fabiobaltieri/zephyrproject/zephyr/include/zephyr/drivers/pwm.h:492
#6  pwm_set_cycles (flags=1, pulse=<optimized out>, period=320000, channel=0, dev=0x7554 <__device_dts_ord_54>) at /home/fabiobaltieri/zephyrproject/zephyr/build/zephyr/include/generated/syscalls/pwm.h:37
#7  pwm_set (dev=0x7554 <__device_dts_ord_54>, channel=channel@entry=0, period=period@entry=20000000, pulse=pulse@entry=400000, flags=flags@entry=1) at /home/fabiobaltieri/zephyrproject/zephyr/include/zephyr/drivers/pwm.h:558
#8  0x00000496 in pwm_set_pulse_dt (spec=<optimized out>, pulse=400000) at /home/fabiobaltieri/zephyrproject/zephyr/include/zephyr/drivers/pwm.h:605
#9  main () at /home/fabiobaltieri/zephyrproject/zephyr/samples/basic/fade_led/src/main.c:38
```

Easy workaround for me but is this something you could look into? Thanks!

**Impact**
Spins and locks the thread.

**Environment (please complete the following information):**
 - OS: Linux, Zephyr SDK, 0bfe3cc2d0

fabiobaltieri, 2023-07-22:
cc @awojasinski-nordicsemi @nika-nordic @carlescufi

awojasinski-nordicsemi, 2023-07-26:
The lock you observed in your app is caused by the driver not changing the internal state of itself. The event is cleared when `nrfx_pwm_stopped_check()` is called for the first time and each next call  is returning `false` instead of `true`

closed 2023-07-27 by carlescufi