zephyrproject-rtos/zephyr #60730: RTIO: Update dead dependencies to RTIO simple executor

kmeinhar, 2023-07-24:
**Is your enhancement proposal related to a problem? Please describe.**

There are still references in code and documentation to RTIO simple executor.
These were removed with PR #57436

**Describe the solution you'd like**

Find and remove all references similar to the following:
* https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_sam_twihs_rtio.c#L425
* https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/services/rtio/index.rst?plain=1#L150

kmeinhar, 2023-07-24:
@teburd Am I correct that in the assumption that `RTIO_EXECUTOR_SIMPLE_DEFINE` was replaced with `RTIO_DEFINE`?

teburd, 2023-07-24:
@kmeinhar In effect yes, there were two executors with different memory footprints at one point. There is only one now with a very low footprint. Thanks for finding these, will send a patch up.

PR #60748 (2023-08-16): rtio: Remove references to simple executor
The simple executor was removed with the usage of the spsc queue but some stray references remained. Remove those.

Fixes #60730
closed 2023-08-16 by carlescufi