zephyrproject-rtos/zephyr #60731: MAINTAINERS: Proposal to introduce new status: unattended

de-nordic, 2023-07-24:
## Introduction

Currently neither state, in subsystem description, addresses an issue where subsystem is still actively used and may be receiving features but there is no-one, specifically an expert, to take care of it.
New status named, "unattended",  "abandoned' or "orphaned", could be used to indicate that worrying state of subsystem.
Such status should also emphasize that currently the subsystem lacks architectural oversight and vision.

Settings subsystem is an example of such problematic state:
 - it is in use, even by other subsystems
 - there are fixes and enhancements needed; for example improvement in removal of inactive keys;
 - there have been some features that have died due to lack of reviews.
Neither of currently available statuses correctly describes it: it is not "maintained", "old fixes" means that somebody could review and accept them in the context of the established general subsystem architecture, and sure it is not "obsolete".

### Problem description
We need a status that allows us to indicate that some subsystem is in limbo.

### Proposed change
Adding new keyword like "unattended" to remind that help is needed with subsystem that is still in active use but lacks caretaker.

### Concerns and Unresolved Questions

Using keyword like" unattended" may discourage people from using the subsystem, effectively making them rethink usability of the system as a whole.

## Alternatives
I do not know any at this point.

nordicjm, 2023-07-24:
Could `unmaintained` be used? If so, it is already a valid status

de-nordic, 2023-07-24:
> Could `unmaintained` be used? If so, it is already a valid status

Is there? If yest, then ok it may be "unmaintained". I have only looked at the MAINTAINERS.yml, and there only the three statuses I have mentioned.

nordicjm, 2023-07-24:
See https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/get_maintainer.py#L484

de-nordic, 2023-07-24:
Seems to be missing here:
https://github.com/zephyrproject-rtos/zephyr/blob/fe15f7736ca5c0e1a47f5d9b5797f4973f6734a9/MAINTAINERS.yml#L9-L25

de-nordic, 2024-05-31:
> isnt this covered by `odd fixes' already?

Yes, I think so. Closing the issue.

closed 2024-05-31 by de-nordic