zephyrproject-rtos/zephyr #60741: stdin always returning data even when there is none when bound to ST-LINK serial uart

codewitch-honey-crisis, 2023-07-24:
Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...
I can properly printf to stdout, but stdin is always returning data, even when there should be none.

while(getc(stdin)!=-1); // hangs

This is unexpected behavior.

Is there some kind of workaround where I can check stdin for available data?

**To Reproduce**

Steps to reproduce the behavior:
1. make a main.c with printf("start\n");while(-1!=serial_getch(stdin)); printf("done\n");
2. west flash -b=nucleo_h723gz
3. observe the st-link output. "done" never prints.

**Expected behavior**
When there is no serial data, getch() and getchar() should return -1 like the C spec indicates.

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (e.g Zephyr SDK, west)
 - latest main (I checked out two days ago) according to VERSION it's 3.4.99

codewitch-honey-crisis, 2023-07-24:
Yes, that is the issue I'm getting at. Hoping for a workaround at the very least.

What do you mean by routing this to the correct subsystem? sorry I'm a noob. How do I route this to the console subsys section?

erwango, 2023-07-24:
We're triaging issues with labels and assigning to people that will be in charge of the point depending of the area of expertise.
If your point ends up in the wrong subsystem, you likely won't get the good support or some time with be lost.

carlescufi, 2023-07-25:
Something is off here, I can't find anywhere `__stdin_hook_install` being called, so how is `getc()` supposed to work at all? cc @cfriedt @stephanosio @keith-packard

carlescufi, 2023-07-25:
> I reproduced the issue with the following:

@erwango while this is useful, @codewitch-honey-crisis is actually using the C library's `getc()`, which requires a stdin hook to be installed. You are using the console subsystem's `console_getchar()` which does not use the C standard library and may have different semantics.

codewitch-honey-crisis, 2023-07-25:
carlsecufi, i tried both.

carlescufi, 2023-07-25:
> carlsecufi, i tried both. getchar was my primary initially until it wasn't working. Then I moved to getc. I thought getchar() was part of the same layer. my mistake.

Both `getchar()` and `getc()` are part of the C standard library and require a `stdin` hook to be installed. `console_getchar()` is a Zephyr-specific function that does not require any hooks to be installed, and instead uses the Devicetree `zephyr-console` chosen node.

cfriedt, 2023-07-25:
> Something is off here, I can't find anywhere `__stdin_hook_install` being called, so how is `getc()` supposed to work at all? cc @cfriedt @stephanosio @keith-packard 
> Note: @codewitch-honey-crisis is using newlib, but the same applies to all the other C standard libraries

Would be great to have a Ztest in a draft PR so that there is a single source of truth for the issue, and if it could be reproduced on a qemu platform, even better.

Wrt the hook, it might be necessary to do a cmdline search through git history as to where it was used in the past. GitHub does not always return useful results, especially if files have moved.

It might be used in the SDK but that's hard for me to say ATM, on mobile.

keith-packard, 2023-07-25:
I just tried getchar on `qemu_cortex_m0`. Results:

 * minimal C library. getchar is not defined, resulting in compile warnings and a link error
 * picolibc. getchar returns -1
 * newlib. getchar hangs.

Here's my exciting sample program:

```
#include <zephyr/kernel.h>
#include <stdio.h>

int main(void)
{
	printk("hello world\n");
	printk("getchar returns %d\n", getchar());
	return 0;
}
```
I duplicated samples/hello_world to get the rest of the build configuration. I didn't spend more than a few minutes attempting to debug newlib.

keith-packard, 2023-07-25:
>     1. make a main.c with printf("start\n");while(-1!=serial_getch(stdin)); printf("done\n");

Which C library are you using here? The default minimal C library doesn't provide getchar or getc at all, so this shouldn't even build. Tests with picolibc show that it works as expected. Tests with newlib never return from getc at all.

Given this, I'm thinking you're using newlib. I'd suggest simply moving to picolibc where things should work as expected. I'm not sure we can fix newlib's behavior here as it is packaged with the toolchain.

codewitch-honey-crisis, 2023-07-25:
sorry, I was using newlib. I tried using picolib but I couldn't get it working after googling for info.

keith-packard, 2023-07-25:
> sorry, I was using newlib. I tried using picolib but I couldn't get it working after googling for info.

Odd. That "should" have been a simple matter of selecting that in your project configuration with `CONFIG_PICOLIBC=y`. For example, I just built samples/hello-world for the `nucleo_h723zg` with:
```
$ west build -b nucleo_h723zg samples/hello_world/ -- -DCONFIG_PICOLIBC=y
```
In any case, it sounds like there's a bug in newlib here, not Zephyr. I'm unsure of how to proceed though; if you manage to make picolibc work, then there's no need to even attempt to debug that -- it's pretty tricky as it's built as part of the SDK, not as a part of Zephyr itself.

codewitch-honey-crisis, 2023-07-25:
It sounds like you tested this with picolib. If newlib doesn't work, I guess worst case if it were my project I'd put a note somewhere in the documentation? Aside from that, I'd say there's not much to be done, as it's not a Zephyr project. I have a workaround so I'm satisfied, personally. If you want to close this issue as "won't fix" or whatever, you won't see me complain. :)

keith-packard, 2023-07-25:
Yeah, I don't think it's worth digging into what newlib does in this case. Debugging on qemu led into the bowels of newlib from which the code never returned though.

closed 2023-07-25 by keith-packard

carlescufi, 2023-08-22:
See https://github.com/zephyrproject-rtos/zephyr/pull/61045

cfriedt, 2023-09-22:
I'm seeing similar behaviour, but I've isolated it to the uart_console driver. I.e. it is independent of the libc.

The fix is relatively trivial. I think it addresses this issue as well, but I might just open a new one to start fresh.