zephyrproject-rtos/zephyr #60750: cbprintf_cxx.h error with void*

ElectricalPaul, 2023-07-24:
There are two specializations  of `z_cbprintf_cxx_arg_size` for `float` and `void *`. When using a `void *`, the linker seems to randomly pick the [specialization](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/cbprintf_cxx.h#L99) or the [generic version](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/cbprintf_cxx.h#L107). The generic version causes a compiler error because of pointer arithmetic.

```
In file included from /usr/local/google/home/pfagerburg/zephyrproject/zephyr/include/zephyr/sys/cbprintf_internal.h:62,
                 from /usr/local/google/home/pfagerburg/zephyrproject/zephyr/include/zephyr/sys/cbprintf.h:124,
                 from /usr/local/google/home/pfagerburg/zephyrproject/zephyr/include/zephyr/logging/log_msg.h:11,
                 from /usr/local/google/home/pfagerburg/zephyrproject/zephyr/include/zephyr/logging/log_core.h:9,
                 from /usr/local/google/home/pfagerburg/zephyrproject/zephyr/include/zephyr/logging/log.h:11,
                 from /usr/local/google/home/pfagerburg/chromiumos/src/third_party/pigweed/pw_log_zephyr/public/pw_log_zephyr/log_zephyr.h:16,
                 from /usr/local/google/home/pfagerburg/chromiumos/src/third_party/pigweed/pw_log_zephyr/public_overrides/pw_log_backend/log_backend.h:16,
                 from /usr/local/google/home/pfagerburg/chromiumos/src/third_party/pigweed/pw_log/public/pw_log/log.h:62,
                 from /usr/local/google/home/pfagerburg/chromiumos/src/third_party/pigweed/pw_rpc/call.cc:18:
/usr/local/google/home/pfagerburg/zephyrproject/zephyr/include/zephyr/sys/cbprintf_cxx.h: In instantiation of 'size_t z_cbprintf_cxx_arg_size(T) [with T = const void*; size_t = unsigned int]':
/usr/local/google/home/pfagerburg/chromiumos/src/third_party/pigweed/pw_rpc/call.cc:306:3:   required from here
/usr/local/google/home/pfagerburg/zephyrproject/zephyr/include/zephyr/sys/cbprintf_cxx.h:109:27: error: pointer of type 'void *' used in arithmetic [-Werror=pointer-arith]
  109 |         return sizeof(arg + 0);
      |                      ~~~~~^~~~
```

My build environment is:
* Linux
* Zephyr SDK 0.16.0
* Zephyr hash 47102de474cb055804e8f786e74c72d8248fd7b0
* Pigweed hash [97c6a1079c65b9c4ad61c83b806c362e268ee5ea](https://pigweed.googlesource.com/pigweed/pigweed/+/97c6a1079c65b9c4ad61c83b806c362e268ee5ea)

Changing `sizeof(arg + 0)` to `sizeof(T)` and deleting the `void *` specialization seems to fix it.

We should also consider deleting the `float` specialization.

ElectricalPaul, 2023-07-24:
Created PR https://github.com/zephyrproject-rtos/zephyr/pull/60751 to fix

ElectricalPaul, 2023-07-25:
PR has merged, and my code builds correctly. Should we also remove the specialization for `float`?

yperess, 2023-07-26:
@dcpleung I'm not sure we want to close this. I believe that the same issue will happen with `float`

jhedberg, 2023-08-01:
@yperess seems like a separate issue might have been better for the `float` case so that we can map which PR closes which bug

yperess, 2023-08-01:
> @yperess seems like a separate issue might have been better for the `float` case so that we can map which PR closes which bug

Your call, both issues are from the same cause, the specialization isn't done right and we get 2 copies of the same function. We're at the mercy of the linker after that.

ElectricalPaul, 2023-08-03:
I think CI didn't complain because it wasn't also building with the Pigweed libraries. Also I realize I forgot to include a link to the source for the application I was building - [Chameleon v3 EC](https://chromium.googlesource.com/chromiumos/platform/chameleon/+/cf50534b5dddec8dfb16cd2466cd5fdba4d5eabd).

it seemed to be a random thing, because we have been building this code for a while, but perhaps the linker was always choosing the "right" copy of the function until one day it didn.t

dcpleung, 2023-08-03:
Maybe certain compiler/linker flags are passed which alter the behavior?

Though, without a way to reproduce using the main zephyr tree, it would be impossible to verify any fixes.

ElectricalPaul, 2023-08-03:
> Though, without a way to reproduce using the main zephyr tree, it would be impossible to verify any fixes.

That's a difficult problem. The PR that merged definitely fixed the issue I was seeing, but without the Pigweed libraries and a Google Chameleon v3 target to build in Zephyr CI, it's nearly impossible to make sure this doesn't come back later.

I will spend some time thinking about whether I can make a much smaller test case that fails on `void *` before that PR, and passes afterwards.

closed 2023-12-18 by github-actions[bot]